realsenseai/librealsense #13112: RuntimeError: No device connected

rr922018, 2024-07-02:
I have been troubled by this problem for a long time. I hope someone can help solve it. thank you!
Images are in the picture.pdf file.
[picture.pdf](https://github.com/user-attachments/files/16065539/picture.pdf)

The Intel Realsense D405 camera can be used normally, and the realsense viewer can display it, as shown in Figure1.

I created a virtual environment with Python 3.8 and installed pyrealsene2 in the environment.The package is shown in Figure 2,3,4.

The code is as follows:
import pyrealsense2 as rs
pipe =rs.pipeline()
profile = pipe.start()
try :
  for i in range(0,100):
     frames = pipe.wait for frames()
     for f in frames :
        print(f.profile)
finally:
      Pipe.stop()

Error as shown in Figure 5.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { Intel Realsense D405 } | 
| Operating System & Version |  Linux (Ubuntu 18) ROS1 | 
| Platform                                 |NVIDIA Jetson TX2 NX  |
| SDK Version                            | librealsense SDK 2.0 }                          | 
| Language                            |python }                          |

MartyG-RealSense, 2024-07-02:
Hi @rr922018  Is the camera detected if you launch your pyrealsense2 script in sudo admin permissions mode with the instruction below, please?

`sudo python3 detect1.py`

rr922018, 2024-07-02:
okay, thank you.
I just tried and now the error is shown in the picture.
[picture1.pdf](https://github.com/user-attachments/files/16066787/picture1.pdf)

MartyG-RealSense, 2024-07-02:
There was a past case of this rare **glbic_2.29** error at https://github.com/IntelRealSense/librealsense/issues/12137

Please try changing your **import pyrealsense2 as rs** line to the one below.

`import pyrealsense2.pyrealsense2 as rs`

rr922018, 2024-07-02:
Hi.
After attempting to install the above method, an error occurred as follows:
[picture1.pdf](https://github.com/user-attachments/files/16066995/picture1.pdf)

rr922018, 2024-07-02:
Hi.
After trying the above method, the error is as follows:
[picture1.pdf](https://github.com/user-attachments/files/16067004/picture1.pdf)

MartyG-RealSense, 2024-07-02:
You are using Python 3.8 in your virtual environment but the glibc error message states that it is required by Python 3.9.  Is it possible that you might have more than one Python 3 version installed on your computer?

rr922018, 2024-07-02:
Yes, the Python environment is 3.9. In this environment, I used Conda create -name envs_yolov8 Python=3.8 to create a 3.8 virtual environment and ran this code in the envs_yolov8 environment.

MartyG-RealSense, 2024-07-03:
A RealSense user at https://github.com/IntelRealSense/librealsense/issues/6922#issuecomment-663976346 who used Conda with the wrapper and also experienced **RuntimeError: no device connected** shared the solution that worked for them in their particular case.

rr922018, 2024-07-04:
Hi,I tried the above method.https://github.com/IntelRealSense/librealsense/issues/6922#issuecomment-663976346
According to https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python, I reinstalled Lirealsense 2.55.1 and Pyrealsense 2, but now realsense viewer no longer displays the device.

MartyG-RealSense, 2024-07-04:
Did you remove your previous installation of librealsense before installing 2.55.1 please?

rr922018, 2024-07-05:
Sorry, the previous version was not uninstalled. Can I just uninstall the previous version of Lirealsense now? Or do I need to uninstall and reinstall both versions?

MartyG-RealSense, 2024-07-05:
If you installed librealsense from Debian packages with apt install then you can uninstall all packages with 'realsense' in their name using the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

If you installed librealsense from source code with CMake then you can uninstall the previous version only by going to the 'build' directory of the source code folder and using this build command:

`sudo make uninstall && make clean`

rr922018, 2024-07-06:
Hello, I have uninstalled both versions and reinstalled Lirealsense 2.55.1. However, the realsense-viewer still cannot display the device.

rr922018, 2024-07-09:
sorry,D405 can't be detected.
[image]
No device detected.Is it plugged in?

MartyG-RealSense, 2024-07-09:
So the RealSense Viewer worked until you installed librealsense2 and pyrealsense2 - as mentioned at https://github.com/IntelRealSense/librealsense/issues/13112#issuecomment-2209167939 - but after that the Viewer no longer detected the D405.

You performed the installation using the information at https://github.com/IntelRealSense/librealsense/issues/6922#issuecomment-663976346

I do not know why an installation of the pyrealsense2 wrapper would prevent the Viewer or rs-enumerate-devices from detecting the camera.

The source code installation guide for Jetson boards at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 may be worth investigating as it works with a range of different Jetson boards and installs pyrealsense2 at the same time as the librealsense SDK.

closed 2024-07-24 by MartyG-RealSense