realsenseai/librealsense #13114: Nano Orin

djaniel, 2024-07-02:
Hello 

I am having some difficulties when setting up librealsense and camera D435i on a Jetson Nano Orin. Here is my setup:

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i and D435 | 
| Firmware Version                   | 5.16.01 | 
| Operating System & Version |   Linux Ubuntu 22  | 
| Kernel Version (Linux Only)    |  5.15.136-tegra                                         | 
| Platform                                 | NVIDIA Nano Orin, Jetson Pack 6.0 |
| SDK Version                            |  2.55.1                          | 
| Language                            |  C/C++/python                          | 
| Segment			|  Robotics Humble ROS2       | 

### Issue Description

I recently reinstalled everything on the Jetson Nano Orin (Fresh start, Ubuntu 22, jetson pack 6.0). I installed from source `librealsense` and tested `realsense-viewer`. The only camera that worked with no particular changes to the CMake configuration was D435. Unfortunately , the D435i was listed when typing `lsusb` but not shown when launching `rs-enumerate-devices`. 

I changed the `RSUSB_BACKEND` flag to true, and the `rs-enumerate-devices` started recognizing the D435i camera. Unfortunately the connection is not reliable, `realsense-viewer` crashes when starting all three streams (RGB, depth, IMU). A lot of error messages appear on the console of `realsense-viewer`. Here is an screenshot:
[image]

Do you have any suggestions?

MartyG-RealSense, 2024-07-02:
Hi @djaniel  There is a known issue for some RealSense users when starting three streams (RGB, depth, IMU) in the RealSense Viewer tool.  The problem, which typically manifests as No Frames Received on one of the streams, does not have a fix.  It is therefore recommendable to only enable two streams at a time if you need to use the IMU (depth + IMU, or RGB + IMU).

MartyG-RealSense, 2024-07-09:
Hi @djaniel  Do you require further assistance with this case, please?  Thanks!

nlitz88, 2024-07-14:
I have also run into this problem with my Jetson Orin Nano running Jetpack 6.0. @MartyG-RealSense do you know if this is a librealsense issue or a Jetpack 6.0 issue?

In either case, while there may not be a fix at the moment, is a fix in the pipeline or planned for the future?

MartyG-RealSense, 2024-07-14:
Hi @nlitz88  The RealSense Viewer 'enabling three streams' issue with the Motion Module is not related to JetPack 6.  A fix is not currently planned.

It is possible to display depth, RGB and IMU streams in the **rs-capture** pre-made example program as it is not affected by the issue.  If you installed librealsense with the examples included then you should be able to find a pre-built executable version of rs-capture in the **usr/local/bin** folder of Linux.

[image]

As mentioned above, when writing your own program script you can workaround the issue to avoid problems occurring by creating two separate pipelines in Python or using callbacks in C++.

nlitz88, 2024-07-16:
Thanks for getting back @MartyG-RealSense 

Maybe this deserves its own issue in [realsense-ros](https://github.com/IntelRealSense/realsense-ros), but I am wondering if this has anything to do with a similar issue realsense ros has when attempting to launch the realsense camera node with the gyroscope and accelerometer enabled.

When I run the following launch command:
```
ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true
```

I get a very similar output compared to the output Alexander is getting in his post: https://github.com/IntelRealSense/realsense-ros/issues/3162#issuecomment-2225081341

My output is as follows:

> [INFO] [launch]: All log files can be found below /home/admin/.ros/log/2024-07-16-01-25-32-097798-f1tarmo-721784
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [721809]
[realsense2_camera_node-1] [0m[INFO] [1721107532.553202078] [camera.camera]: RealSense ROS v4.51.1[0m
[realsense2_camera_node-1] [0m[INFO] [1721107532.553485413] [camera.camera]: Built with LibRealSense v2.55.1[0m
[realsense2_camera_node-1] [0m[INFO] [1721107532.553525286] [camera.camera]: Running with LibRealSense v2.55.1[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.476560625] [camera.camera]: Device with serial number 032622072910 was found.
[realsense2_camera_node-1] [0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.476768982] [camera.camera]: Device with physical ID 2-1.1-3 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.476805687] [camera.camera]: Device with name Intel RealSense D435I was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.477230529] [camera.camera]: Device with port number 2-1.1 was found.[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.477274050] [camera.camera]: Device USB type: 3.2[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.483157388] [camera.camera]: getParameters...[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.489649092] [camera.camera]: JSON file is not provided[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.492735500] [camera.camera]: Device Name: Intel RealSense D435I[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.505793887] [camera.camera]: Device Serial No: 032622072910[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.509419988] [camera.camera]: Device physical port: 2-1.1-3[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.512172692] [camera.camera]: Device FW version: 5.16.0.1[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.517942363] [camera.camera]: Device Product ID: 0x0B3A[0m
[realsense2_camera_node-1] [0m[INFO] [1721107533.519191545] [camera.camera]: Sync Mode: Off[0m
[realsense2_camera_node-1] [0m[INFO] [1721107538.342939504] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30[0m
[realsense2_camera_node-1] [0m[INFO] [1721107539.990298696] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30[0m
[realsense2_camera_node-1] [0m[INFO] [1721107539.991906702] [camera.camera]: Set ROS param gyro_fps to default: 200[0m
[realsense2_camera_node-1] [0m[INFO] [1721107539.992069201] [camera.camera]: Set ROS param accel_fps to default: 63[0m
[realsense2_camera_node-1] [0m[INFO] [1721107540.015772732] [camera.camera]: Stopping Sensor: Stereo Module[0m
[realsense2_camera_node-1] [0m[INFO] [1721107540.268961861] [camera.camera]: Starting Sensor: Stereo Module[0m
[realsense2_camera_node-1] [0m[INFO] [1721107540.440405748] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30[0m
[realsense2_camera_node-1] [0m[INFO] [1721107540.440716091] [camera.camera]: Stopping Sensor: RGB Camera[0m
[realsense2_camera_node-1] [0m[INFO] [1721107540.450258138] [camera.camera]: Starting Sensor: RGB Camera[0m
[realsense2_camera_node-1] [0m[INFO] [1721107540.516928595] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30[0m
[realsense2_camera_node-1] [0m[INFO] [1721107540.517158361] [camera.camera]: Stopping Sensor: Motion Module[0m
[realsense2_camera_node-1]  16/07 01:25:40,534 WARNING [281472820766944] (ds-calib-parsers.cpp:35) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[realsense2_camera_node-1] [0m[INFO] [1721107540.537118572] [camera.camera]: Starting Sensor: Motion Module[0m
[realsense2_camera_node-1] [0m[INFO] [1721107540.566035313] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 63[0m
[realsense2_camera_node-1] [0m[INFO] [1721107540.566233526] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200[0m
[realsense2_camera_node-1] [0m[INFO] [1721107540.587328164] [camera.camera]: RealSense Node Is Up![0m
[realsense2_camera_node-1]  16/07 01:25:40,683 WARNING [281472795404512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/07 01:25:40,697 WARNING [281472795404512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  16/07 01:25:40,716 WARNING [281472795404512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  16/07 01:25:41,701 WARNING [281472795404512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

And so on.

Just for some context, I have:

- A realsense D435i with firmware version 5.16.0.1
- librealsense version 2.55.1 (built as a part of [NVIDIA's Isaac ROS realsense Dockerfile](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/Dockerfile.realsense), if that matters).
- Realsense ROS version 4.51.1
- NVIDIA Jetson Orin Nano 8GB running Jetpack 6.0

nlitz88, 2024-07-16:
Quick update: I just downgraded the D435i's firmware to version 5.13.0.50 [per NVIDIA's Isaac ROS RealSense setup guide](https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/sensors/realsense_setup.html) (mainly because that is my target application). NOW, when I run
```
ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true
```
I only get a few of those "control_transfer returned error" messages, but ultimately I'm getting messages published to both `/camera/accel/sample` and `/camera/gyro/sample` at 60 Hz and 200 Hz, respectively.

Soooo...maybe downgrading the firmware solved the problem for this scenario? Sorry if this isn't relevant here, just wanted to reported what I found!

MartyG-RealSense, 2024-07-16:
Hi @nlitz88  I'm very pleased to hear that you have made positive progress! in your particular case!

@djaniel  Have you been able to make any progress since your previous comment, please?

MartyG-RealSense, 2024-07-24:
Hi @nlitz88  Do you require further assistance with this case, please?  Thanks!

nlitz88, 2024-07-24:
I believe my problem is solved for now--thank you for steering me in the right direction @MartyG-RealSense !

MartyG-RealSense, 2024-07-24:
You are very welcome, @nlitz88 - I'm pleased that you had a successful outcome.  Thanks very much for the update!

MartyG-RealSense, 2024-07-30:
Case closed due to solution achieved and no further comments received.

closed 2024-07-30 by MartyG-RealSense

djaniel, 2024-08-29:
Thank you for your interest. After your reply, I abandoned the idea of launching all streams. I am more cautious, now.