realsenseai/librealsense #13115: Some area of the color Image mismatched with the depth Image

yijionglin, 2024-07-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    | 6.5.0-41-generic     | 
| Platform                                 | PC  |
| SDK Version                            |    SDK 2.0 (maybe...)            | 
| Language                            | python                      | 
| Segment			|  CV                       | 

### Issue Description
Hello there,

I have tried to map the segmentation area in the color image to the depth image. However, I found there is some area shifted in the depth image. 

It looks like a problem with the integers of the depth pixels (the `depth_pixel` generated from `project_color_pixel_to_depth_pixel` is a list with two float values), here is the snippet of my code:
 
```
    for i, color_pixel in enumerate(seg_inds):
        depth_pixel = rs_camera.project_color_pixel_to_depth_pixel(
                    depth_frame.get_data(), depth_scale,
                    depth_min, depth_max,
                    depth_intrin, color_intrin,  color_to_depth_extrin,depth_to_color_extrin, color_pixel)
        
        # depth_pixel_round = np.round(depth_pixel).astype(int)
        # depth_pixels_seg.append(depth_pixel_round)
        # cv2.circle(depth_colormap_c, (depth_pixel_round[0], depth_pixel_round[1]), 1, color_shaded_dep, thickness)

        depth_pixel_small = np.asarray(depth_pixel).astype(int)
        depth_pixels_seg.append(depth_pixel_small)
        cv2.circle(depth_colormap_c, (depth_pixel_small[0], depth_pixel_small[1]), 1, color_shaded_dep, thickness)
    cv2.imwrite(depth_colormap_save_path, depth_colormap)
    cv2.imwrite(depth_colormap_seg_save_path, depth_colormap_c)
    return depth_pixels_seg
```
>

And here are the original color/depth images:
[image]
[image]

And here are the segmented color/depth images:

[image]
[image]

Could you please give me some suggestions on this?

Many thanks,
Bourne

yijionglin, 2024-07-02:
A side question:

I have seen a few people post their snips of code ([example 1](https://github.com/IntelRealSense/librealsense/issues/10484#issuecomment-1136316275), [example 2](https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008)) in the issues which use:

```
        depth_point_ = rs.rs2_project_color_pixel_to_depth_pixel(
            depth_frame.get_data(), depth_scale,
            depth_min, depth_max,
            depth_intrin, color_intrin, depth_to_color_extrin, color_to_depth_extrin, color_point)
```

However, according to [the API document](https://intelrealsense.github.io/librealsense/doxygen/rsutil_8h.html#af04b7b1ba50b1d9e03089d1e5186d2f8), the order of the depth_to_color_extrin and color_to_depth_extrin is swapped. 

I stick with the input argument order as indicated in the document and it works better than the inverted one. But I am not sure if this is the cause of my aforementioned problem.

yijionglin, 2024-07-02:
Here is another example figure of my problem, but it is not that notable as the segmented area is larger.
[image]

yijionglin, 2024-07-03:
Hi there,

If possible, could I get any suggestions on this bit?

Best,

MartyG-RealSense, 2024-07-03:
Hi @yijionglin  The depth colorization on your images appears to be reversed.  Usually depth is colored from blue (near range depth values) to red (far range depth values).  In your images though, the near range is represented by red and the far range by blue.  The link below has resources about setting the colorizer **color scheme** with Python code.

https://support.intelrealsense.com/hc/en-us/community/posts/1500000949421-D435i-color-scheme-problem

If you do not mind your depth image being colored this way though then you do not need to adjust it.

The depth image of the white machine does seem to be very distorted though and the black component with a vertical shaft has a black shadow behind it.

It is a general physics principle (not specific to RealSense) that dark grey or black absorbs light and so makes it more difficult for depth cameras to read depth information from such surfaces.  The darker the color shade, the more light that is absorbed and so the less depth detail that the camera can obtain.  Increasing the strength of the external light illumination cast onto the black area can help to bring out depth information from it.

If you have access to the RealSense Viewer tool then you could try resetting your camera to its default factory-new calibration using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to see whether that makes a difference.

MartyG-RealSense, 2024-07-03:
You could test whether there is a problem in your code by running the RealSense SDK's **rs-align** depth-color alignment program.  If you installed the SDK with examples and tools included then it should be in the Ubuntu directory **usr/local/bin**

Regarding the black shadow, I mean the one behind the orange shaft.

[image]

yijionglin, 2024-07-03:
> You could test whether there is a problem in your code by running the RealSense SDK's **rs-align** depth-color alignment example program. If you installed the SDK with examples and tools included then a pre-built executable version of rs-align should be in the Ubuntu directory **usr/local/bin**
> 
> Regarding the black shadow, I mean the one behind the orange shaft.
> 
> ![image](https://private-user-images.githubusercontent.com/41145062/345455761-9b8fb027-e7c3-4ee9-8331-d423fba0c931.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAwMjkwNjYsIm5iZiI6MTcyMDAyODc2NiwicGF0aCI6Ii80MTE0NTA2Mi8zNDU0NTU3NjEtOWI4ZmIwMjctZTdjMy00ZWU5LTgzMzEtZDQyM2ZiYTBjOTMxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzAzVDE3NDYwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRhZGE3NmU3NjAxYWVlYTA2MmM1NDFhOGYyZWNlNzY5OTEyOWUwY2I5NTU0MGUwMzZhMjllNTkxOTk2NjI2MGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0Gr4qbSLCGDgLuVMJJku3_uMK7Kd878HAdE0un3--MY)

Hi @MartyG-RealSense , thanks for the suggestions!

I have just tried the RealSense SDK's rs-align depth-colour alignment example program and here is the screenshot, how could we know if it is well aligned or not? Here are the two images with color and depth references (by clicking the box down below)

[image]
[image]

yijionglin, 2024-07-03:
After the calibration reset process following the indication you shared, I have also played with the 3D view option in the realsense-viewer, where this phenomenon still exists. Here is a short video for it.

[Screencast from 03-07-24 19:11:05.webm](https://github.com/IntelRealSense/librealsense/assets/26537492/5662da4d-6145-42db-b7e3-990794279b78)

MartyG-RealSense, 2024-07-04:
Thanks very much for the testing and images.  They do seem to indicate that the problem is not in your code.  

Does the Viewer image improve if you change the 'Depth Units' setting at **Stereo Module > Controls > Depth Units** from its default of 0.001 to **0.0001** please?

If it does make a difference, you can change the depth units in your program script with the **rs.option.depth_units** instruction.

```
profile = pipeline.start()
depth_sensor = profile.get_device().first_depth_sensor()
if depth_sensor.supports(rs.option.depth_units):
depth_sensor.set_option(rs.option.depth_units, 0.001)
```

yijionglin, 2024-07-04:
Hi @MartyG-RealSense ,

Thanks for your reply! I have tried the value you suggested and it seems it doesn't make any difference for this issue. By the way, what does the depth unit mean? Is it similar to the resolution?

MartyG-RealSense, 2024-07-04:
It is a difficult setting to explain, and one that does not really need an explanation in order to use it.  It relates to the number of meters per one increment of depth.  0.001 would indicate millimeter scale, while 0.01 would indicate centimeter scale.

The simplest way to think about it is that when observing an object at relatively close range, changing the depth scale to 0.0001 can sometimes reduce the size of the black shadow around an object.

MartyG-RealSense, 2024-07-18:
Hi @yijionglin  Do you require further assistance with this case, please?  Thanks!

yijionglin, 2024-07-18:
Hi @MartyG-RealSense 

Thanks for asking and sorry for not getting back to you.

I have a feeling that this small problem is unavoidable based on the current code provided by Realsense and I think we should keep this issue open until the problem is solved. But feel free to close it for any reason.

Best,

MartyG-RealSense, 2024-07-18:
It's no trouble at all.  I do not have anything further to add though, so I will close the issue.

closed 2024-07-18 by MartyG-RealSense