realsenseai/librealsense #13116: Installation on Jetson Orin Nano

ClaudioCarbone, 2024-07-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |     D456                                                   | 
| Operating System & Version |  Ubuntu 22.04                                         | 
| Kernel Version (Linux Only)    |  5.15.136-tegra                                       | 
| Platform                                 | NVIDIA Jetson Orin Nano                       |

I'm currently trying to use the Intel Realsense d456 on my Jetson Orin Nano board, i'm following this guide: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md
I am not able to use realsense-viewer and pipe.start(), they both say there are no devices connected.

The first problem arises with this part which says:
"For Jetson Orin™ with JetPack 6.0 you will need to follow build for MIPI driver as NVIDIA released Kernel 5.15 default configuration with disabled HID: [Intel® RealSense™ camera driver for GMSL* interface](https://github.com/IntelRealSense/realsense_mipi_platform_driver)".
I have Jetson 6.0 but the guide is confusing for me, i dont understand what i have to change if i have a d456 instead of a d457. Also the instructions say that "Debian packages will be generated in _images_ folder." but i cant find this folder. Maybe this happened because this guide is "D457 MIPI on Jetson AGX Xavier", so i had to skip this part and hoped it was not needed.

Continuing with the fist guide i followed [Step 4. Install with Debian Packages](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages).
The installation finished with no issues but as i said, on realsense-viewer, no devices are found. I tried the viewer on windows and everything works there.

MartyG-RealSense, 2024-07-03:
Hi @ClaudioCarbone  The MIPI driver provides support for the **IMU** on a USB cable connection with RealSense cameras that are equipped with an IMU (including D456) when using JetPack 6.  Not using the MIPI driver should not prevent the camera from being detected.

Let's set aside the MIPI driver for the moment and focus on trying to get the camera detected.  Is it detected in realsense-viewer if you launch it in sudo admin permissions mode with the command **sudo realsense-viewer**

ClaudioCarbone, 2024-07-03:
Hi @MartyG-RealSense , thank you for your reply.

First of all i wanted to say that this warning appeared again after I reinstalled everything
[image]
I saw this problem in other issues and opted to remove the /etc/udev/ruled.d/99.realsense-libusb.rules file and the problem is solved.

But again this is what realsense-viewer printed:
```
16:51:28 [Info]  - librealsense version: 2.55.1

16:51:28 [Info] ./src/rs.cpp:3103 - Framebuffer size changed to 1850 x 1016
16:51:28 [Info] ./src/rs.cpp:3103 - Window size changed to 1850 x 1016
16:51:28 [Info] ./src/rs.cpp:3103 - Scale Factor is now 1
16:51:28 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video2
16:51:28 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video3
16:51:28 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video4
16:51:28 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video5
16:51:28 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.3/video4linux/video6
16:51:28 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.3/video4linux/video7
16:51:28 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video2
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video3
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video4
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video5
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.3/video4linux/video6
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.3/video4linux/video7
16:51:29 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video2
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video3
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video4
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video5
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.3/video4linux/video6
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.3/video4linux/video7
16:51:29 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video2
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video3
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video4
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video5
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.3/video4linux/video6
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.3/video4linux/video7
16:51:29 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video2
16:51:29 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video3
[... cut ...]
16:51:30 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.3/video4linux/video6
16:51:30 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.3/video4linux/video7
16:51:30 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
```

If I try to add a new source there is no option for the realsense d456 that i have.

After, I tried to run the command `sudo realsense-viewer` but nothing changes.

ClaudioCarbone, 2024-07-03:
I want to add that ubuntu is detecting the various UVC videoX cameras but nothing is shown as source in the viewer.
(they start from video2 because i had another USB camera for tests connected, that i disconnected before opening the viewer)

[image]

MartyG-RealSense, 2024-07-03:
Another JetPack 6 user had similar camera detection problems to yours yesterday at https://github.com/IntelRealSense/librealsense/issues/12566#issuecomment-2202761827

Do you have the option to downgrade to JetPack 5.1 on your Jetson?  RealSense users with a Jetson who have had problems with JetPack 6 have been able to achieve a working setup after a JetPack downgrade.

ClaudioCarbone, 2024-07-03:
@MartyG-RealSense This could not be a viable option since this is not the only task I'm working on with the board and also I'm not the only user to operate on it.
I will still consider it as a last option but I hope there's something else I can try before that.

jeezrick, 2024-07-04:
> @MartyG-RealSense This could not be a viable option since this is not the only task I'm working on with the board and also I'm not the only user to operate on it. I will still consider it as a last option but I hope there's something else I can try before that.

try change line 47 in ./libuvc_installation.sh file to
cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=ON, and build it.

That's how I let realsense-viewer find the camera. If imu doesn't working, downgrade the firmware of camera.

MartyG-RealSense, 2024-07-04:
SDK 2.55.1 requires camera firmware driver version **5.16.0.1**.  Do you have this firmware installed in your D456 camera, please?

MartyG-RealSense, 2024-07-05:
Okay, thanks very much.  If you updated the camera in the RealSense Viewer in Windows then it will have installed the recommended firmware for that particular Viewer version.  So if it was not the 2.55.1 version then an older firmware than 5.16.0.1 will have been installed.

You should be able to check the firmware version with `rs-enumerate-devices -s`

ClaudioCarbone, 2024-07-05:
> Okay, thanks very much. If you updated the camera in the RealSense Viewer in Windows then it will have installed the recommended firmware for that particular Viewer version. So if it was not the [...]

I tried `rs-enumerate-devices -s` but it still tells me that there are no devices connected.
I have tried to change USB and also flip the usb type c connected to the camera but nothing changes. I also saw that some cameras have a switch if you open the little door with a pin, but mine does not have the switch inside.

[image]

I also checked and on windows I have the same viewer version that I installed on ubuntu which is 2.55.1 and the firmware version of the camera is 5.16.0.1.

[image]

Could it be that on the Jetson I don't have USB 3.0?

MartyG-RealSense, 2024-07-05:
Only the D457 camera model has a moveable mode switch under the little door on top of the camera.

The RealSense SDK supports USB 2.1, so the camera should be detected even if it is detected as USB 2.1.  The ports on a Jetson Orin Nano are USB 3.2 though.

For Nano boards, Intel strongly recommend enabling the **barrel jack** power connector for extra power using the instructions at the link below.  Orin Nano has a barrel jack.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

ClaudioCarbone, 2024-07-05:
> Only the D457 camera model has a moveable mode switch under the little door on top of the camera.
>
> The RealSense SDK supports USB 2.1, so the camera should be detected even if it is detected as USB [...]

Thanks for the suggestion, I am already using the DC power jack though!

MartyG-RealSense, 2024-07-06:
The best way forward would be to wipe the Jetson and install everything from the beginning, including Ubuntu 22.04.  I note that this would not be an option for you though as "it is not the only task I'm working on with the board and also I'm not the only user to operate on it".

The next best option would be to uninstall the librealsense Jetson packages with the command below and then perform the package install instructions again.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Another approach would be a source code build with the **RSUSB backend** method using the instructions for Jetson at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

ClaudioCarbone, 2024-07-06:
Hi @MartyG-RealSense thank you for your support, after uninstalling librealsense with `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge` I built from source with the RSUSB backend method that you mentioned and finally the realsense camera was detected!
By running `rs-enumerate-devices` I can now see the camera and by running `sudo realsense-viewer` I am able to open the camera streams.

I can't declare victory yet because I think i'm having some issues with the motion module. Should I open a new issue or can I continue here?

Anyway these are the messages when i open the viewer:

[image]

When i try to turn on the motion module, a lot of errors and hundreds of warnings show up:

[image]
[image]

I tried disabling the auto exposure but it does not fix it.

MartyG-RealSense, 2024-07-06:
The IMU problem could be because of the need to install the MIPI driver for USB support when using JetPack 6.  If you cannot downgrade to Jetpack 5.1 then there may not be a way around that issue, unfortunately.

Regarding the Python problem, does it make a difference if you unplug the camera and then re-insert it before running your script?

ClaudioCarbone, 2024-07-06:
> Regarding the Python problem, does it make a difference if you unplug the camera and then re-insert it before running your script?

I just tried to unplug it and re-insert it but the output is the same.

MartyG-RealSense, 2024-07-07:
A couple of months ago, another RealSense user at https://github.com/IntelRealSense/librealsense/issues/11368#issuecomment-2095667879 who was experiencing **RuntimeError: no device connected** with their Python script found though that the SDK's Python example script **python-tutorial-1-depth.py** could run successfully.  Does this script also work for you if you run it?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

That user's camera detection problem was solved by a libuvc backend installation of librealsense, though I note that you tried it earlier at https://github.com/IntelRealSense/librealsense/issues/13116#issuecomment-2208699941 and it did not resolve your problem.

ClaudioCarbone, 2024-07-07:
I tried to run `python-tutorial-1-depth.py` but also with this script i'm having the same issue of no device connected.

[image]

I've also tried to do the libuvc backend installation of librealsense but still, the python script does not detect the camera.

MartyG-RealSense, 2024-07-07:
It looks as though we have not mentioned in this discussion yet which method you used to install the pyrealsense2 wrapper.  The options are:

1.  Install it with the command **pip install pyrealsense2**.  Although pip install on Arm devices such as Jetson is supported for Python 3.7 to 3.9, it does not work with Jetson boards.

2.  Build the pyrealsense2 wrapper at the same time as the librealsense SDK by building librealsense from source code with CMake and adding the flag **-DBUILD_PYTHON_BINDINGS:bool=true** to the CMake build instruction.

3.  Building the pyrealsense2 wrapper from source code separately after librealsense has already been installed.

Methods 2 and 3 should work with any version of Python 3 that are newer than 3.7.

ClaudioCarbone, 2024-07-07:
Honestly i think i should install the pyrealsense2 wrapper again, because i uninstalled the library and reinstalled it.
Now i dont know in which folder I should this anymore, i used so many ways to install this that I have 3 folders in /home:
1. librealsense
2. librealsense-2.55.1
3. librealsense_build (which inside it has librealsense-master)

I think i should do this in `/librealsense-2.55.1`, i'm gonna build from source and i'm following [this guide](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md#building-from-source).

ClaudioCarbone, 2024-07-07:
Honestly i'm not sure because i have uninstalled it and reinstalled it. I should do i again but i'm not sure how, first thing is that i have now different folders in `/home` for librealsense and i dont know which one in the right one now (librealsense, librealsense-2.55.1 or librealsense_build).

I can try to build from source by following [this guide](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md#building-from-source) but I dont even know where to create the `/build` folder.

MartyG-RealSense, 2024-07-08:
The librealsense source code folder will have the version number in its name.  So it would be **librealsense-2.55.1**

The /build folder should be created with **mkdir build** whilst in the root directory of the librealsense source code folder, which is the one with the list of folders and files in the image below.

[image]

ClaudioCarbone, 2024-07-09:
@MartyG-RealSense I now followed [this guide](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md#building-from-source) and did it in the folder you showed me, but still no device connected.

I can try to do step 3 (Building the pyrealsense2 wrapper from source code separately after librealsense has already been installed) but I'm not sure how to proceed for this.

MartyG-RealSense, 2024-07-09:
Building the pyrealsense2 wrapper from source code separately after librealsense has already been installed uses the guide that you have already tried, unfortunately.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

You have already tried building with RSUSB = true and realsense-viewer was able to work but the camera was still not detected in the Python wrapper.  Can you confirm please whether or not you used the instructions at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 to build librealsense and pyrealsense2 together at the same time with RSUSB = true please?

ClaudioCarbone, 2024-07-09:
> It may be better to try a new approach. You could build from source on your Jetson using the JetsonHacks website's **buildLibrealsense.sh** build script.
> 
> https://github.com/JetsonHacksNano/installLibrealsense?tab=readme-ov-file#buildlibrealsensesh
> 
> This script includes the **-DBUILD_PYTHON_BINDINGS=bool:true** flag in its CMake build instruction, so it should build librealsense and the pyrealsense2 wrapper at the same time.
> 
> https://github.com/JetsonHacksNano/installLibrealsense/blob/master/buildLibrealsense.sh#L135
> 
> You should also add the flag **-DBUILD_GRAPHICAL_EXAMPLES=true** to the buildLibrealsense.sh script if you want to build it with realsense-viewer included in the installation.

@MartyG-RealSense I tried the JetsonHacks scripts like you have suggested. This is what i did:
1. `git clone https://github.com/JetsonHacksNano/installLibrealsense.git`
2. `cd installLibrealsense`
3. `./installLibrealsense.sh`
4. modified line 135 of buildLibrealsense.sh script by adding `-DBUILD_GRAPHICAL_EXAMPLES=true`
5. `./buildLibrealsense.sh -v 2.55.1`

But I had to stop here because i got this error:

[image]

As you can see from the image i tried to upgrade librealsense with pip but, as expected, that didn't work.

G-S-Rodrigues, 2024-07-09:
Hi. 
I'm also struggling to install realsense on jetpack 6 in a jetson orin.

I have follow the instructions and build from source using JetsonHacks website's buildLibrealsense.sh build script, where i changed in line 135 to:

`/usr/bin/cmake ../ -DBUILD_EXAMPLES=true -DFORCE_LIBUVC=ON -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_CUDA="$USE_CUDA" -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS=bool:true`

However when i run the realsense-viewer my color stream and IMU don't work. I guess that the IMU (libusb error) should be from the MIPI driver missing, but from the issues that i i'm reading the RGB camera should already be working. My fps are ok and the depth stream is also fine. 

any advice that you can give me.

MartyG-RealSense, 2024-07-09:
Hi @G-S-Rodrigues  You do not need to add **-DFORCE_RSUSB_BACKEND=ON** to the build script as the **-DFORCE_LIBUVC=ON** flag already in the script performs the same function.

@ClaudioCarbone  What happened when you tried the JetSonHacks script.  Did you experience the same problems?

ClaudioCarbone, 2024-07-09:
@MartyG-RealSense after running the JetsonHack script now i dont have any device detected even on the viewer. Should i build from source to fix like i did [here](https://github.com/IntelRealSense/librealsense/issues/13116#issuecomment-2211708573)?

G-S-Rodrigues, 2024-07-09:
> Hi @G-S-Rodrigues You do not need to add **-DFORCE_RSUSB_BACKEND=ON** to the build script as the **-DFORCE_LIBUVC=ON** flag already in the script performs the same function.
> 
> @ClaudioCarbone What happened when you tried the modified JetsonHacks script. Did you experience the same problems?

thanks for the quick reply. I still don't have image from the rgb camera in `realsense-viewer`. What should i try to do?

MartyG-RealSense, 2024-07-10:
@ClaudioCarbone  Sure, if you have had some success with https://github.com/IntelRealSense/librealsense/issues/13116#issuecomment-2211708573 before then it would be worthwhile to try it again to see if the positive outcome is repeated.

@G-S-Rodrigues  Please try using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to reset the camera to its factory-new default calibrartion settings.

ClaudioCarbone, 2024-07-10:
@MartyG-RealSense i did it and, like before, I have an output on the viewer but not with the python library

MartyG-RealSense, 2024-07-10:
@ClaudioCarbone  I believe that you have tried almost everything possible to get the Python library working.

Sometimes there are cases where the Python wrapper works fine on a PC but on a Jetson the wrapper will not detect a camera no matter what installation method is used.  This may be such a case, unfortunately.

ClaudioCarbone, 2024-07-10:
> @ClaudioCarbone I believe that you have tried almost everything possible to get the Python library working.
>
> Sometimes there are cases where the Python wrapper works fine on a PC but on a Jetson the [...]

I guess my only option now is to downgrade to Jetpack 5.1, will that probably fix my problems?

MartyG-RealSense, 2024-07-10:
A downgrade to 5.1 has worked for other users of JetPack 6 who experienced problems with camera models equipped with an IMU.

ClaudioCarbone, 2024-07-11:
> A downgrade to 5.1 has worked for other users of JetPack 6 who experienced problems with camera models equipped with an IMU.

Ok, I am working on Jetpack 5.1 now. What is the best way to install all all the libraries?

Should i build from source or use JetsonHacks repo?

ClaudioCarbone, 2024-07-11:
@MartyG-RealSense I was wondering also if there's a way to get raw depht data not using pyrealsense2, because I can get some frames using opencv using `cv2.VideoCapture(/dev/video2)`

MartyG-RealSense, 2024-07-12:
When building from source code, the .so files should be placed in the **librealsense > build > wrappers > Python** directory of the source code folder if the flag **-DBUILD_PYTHON_BINDINGS:bool=true** is included in the CMake build instruction.

The JetsonHacks **buildLibrealsense** source code build script includes the Python bindings flag, so you should be able to conveniently install librealsense and the pyrealsense2 wrapper at the same time using this script.

https://github.com/JetsonHacksNano/installLibrealsense/blob/master/buildLibrealsense.sh

In regard to capturing raw data without using pyrealsense2, if librealsense is built from source with CMake with the flag **-DFORCE_LIBUVC=OFF** included in the CMake build instruction then you should be able to capture raw data from the camera using standard Linux tools, though it would not be nicely colorized.

In regard to PYTHONPATH, this tells the pyrealsense2 wrapper where to find your installation of Python on your computer.  It can sometimes be awkward to get working.  You do not need to set up a PYTHONPATH though if you instead copy the .so files **librealsense2.so** and **pyrealsense2.so** to the same folder that your pyrealsense2 program script is n.

When using Python 3, the pyrealsense2.so file will have a more complex auto-generated filename based on the Python 3 version number you are using.  For example, if someone was using Python 3.5 then the filename might be something like **pyrealsense2.cpython-35m-arm-linux-gnueabihf.so**, with '35m' denoting that Python 3.5 has been detected on the computer.

ClaudioCarbone, 2024-07-12:
> When using Python 3, the pyrealsense2.so file will have a more complex auto-generated filename based on the Python 3 version number you are using. For example, if someone was using Python 3.5 then [...]

At `/home/librealsense-2.55.1/build/release` I found some files named:
1. `pyrealsense2.cpython-310-aarch64-linux-gnu.so.2.55.1`
2. `librealsense2.so.2.55.1`
3. `librealsense2-gl.so.2.55.1`

I tried copying them in my home folder where I also have the test python script (python-tutorial-1-depth.py), but it still tells me that no device are connected.

Are those the correct files? Why are them in a different folder as expected?

MartyG-RealSense, 2024-07-12:
I believe that **librealsense2.so.2.55.1** is the librealsense2.so file that you need but that you should look in the **/usr/local/lib** folder for it.

I would normally expect the pyrealsense2 .so file to be located in the **librealsense > build > wrappers > Python ** directory of the librealsense source code folder after the wrapper has been built from source code.

ClaudioCarbone, 2024-07-12:
That's correct, I found **librealsense2.so.2.55.1** at **/usr/local/lib** so I copied it in my home folder where i have the python script.

As for the **pyrealsense2 .so** file, there is no .so file at **librealsense > build > wrappers > Python** like you suggested.
So I deleted the whole folder and tried to build again using [this method](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049) which i followed untill step 6 and ignored the rest for now.

I don't know if it was a good idea but I also removed the last 3 lines of .bashrc that he told to add in step 7, because I'm worried that they might create some errors.

The problem is that I still can't find the .so file at **librealsense > build > wrappers > Python**, what am I doing wrong here?

[image]

MartyG-RealSense, 2024-07-12:
There was a past case of someone building the wrapper on a Jetson Nano where their librealsense2.so file was in **usr/local/lib**, but they also had the pyrealsense2.so files generated in their **/build/wrappers/python** folder.

[image]

The link below lists the steps that they took that led to those files being generated.

https://support.intelrealsense.com/hc/en-us/community/posts/4405767306003/comments/4405944882963

ClaudioCarbone, 2024-07-12:
@MartyG-RealSense
I have seen this post before, he's getting the same problem as me which is not finding the librealsense2.so file at **librealsense > build > wrappers > Python** but only in **usr/local/lib**.
Unless I'm not seeing some other answers, the conversation ended with:
"Let's put aside the .so files for now and see if we can improve on the installation process by going back to the beginning. [...]"

Which was never answered by the user and, since the instructions are the same as [this comment](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049) which I followed again today, I'm not sure this can fix the issue.

Am I perhaps missing something?

MartyG-RealSense, 2024-07-13:
As far as I can tell you have not missed anything.  You have files in your build > Python folder (though not the ones that you need), so this shows that some kind of build process has taken place.

Can your Python script run correctly if you input the command below into the terminal before you run the script in order to state where to find Python 3.10 without needing to put the two files into the project folder?

`export PYTHONPATH=$PYTHONPATH:/usr/local/lib`

ClaudioCarbone, 2024-07-13:
Thank you for your suggestion, I tried to run `export PYTHONPATH=$PYTHONPATH:/usr/local/lib` directly from terminal and then tried the python script but it was not successful.
So i used `which python3` to check the python 3 path and this is what I got:

[image]

I also tried `export PYTHONPATH=$PYTHONPATH:/usr/bin` but it didn't work either.

MartyG-RealSense, 2024-07-13:
What about **/usr/bin/python3** as the path?  That is the path that another Jetson user at https://github.com/IntelRealSense/librealsense/issues/12377 got and their build included the .so files.

ClaudioCarbone, 2024-07-15:
> What about **/usr/bin/python3** as the path? That is the path that another Jetson user at #12377 got and their build included the .so files.

I also tried `export PYTHONPATH=$PYTHONPATH:/usr/bin/python3` but it didn't work.

I noticed the user in the mentioned post had a differen't issue, which is "No Module named pyrealsense2", in my case my python script returns "No device connected".
Nonetheless, I noticed he tried to build using [this repo](https://github.com/35selim/RealSense-Jetson) which I will try too, with his changes. I will also modify this script in line 17 and add `-DFORCE_RSUSB_BACKEND=true`.

@MartyG-RealSense I still think there is something that I did wrong while building the pyrealsense2 library and I wanted to ask you how would you check if pyrealsense2 was correctly installed in your workstation, so I can try it myself.
Unfortunately I have until this thursday to work on this, otherwise I'm gonna have to give up, abandon the camera and use something else for my project. This really surprises me considering how much I paid for it, how much time I've spent trying to fix this and all the support you gave me.

MartyG-RealSense, 2024-07-15:
If you create a test script with only the line **import pyrealsense2 as rs** in it then that will confirm whether the pyrealsense2 wrapper and and Python 3 can find each other or whether instead you will receive **No Module named pyrealsense2**.

**No device connected** errors are significantly more difficult to solve though, unfortunately.  Usually the RealSense Viewer will be working fine but the camera is not detected by Python and the reason for this is not clear.  Multiple approaches may be tried and the outcome is the same even when it seems that procedures have been followed correctly each time.  I very much understand how frustrating it can be and appreciate your patience.

ClaudioCarbone, 2024-07-15:
Yeah this problem of no device connected seems really impossible to fix and I undestand that this is happening because I'm working on a Jetson board.

Anyway I was looking at [this post](https://github.com/IntelRealSense/librealsense/issues/12377#issuecomment-1804184916) again and I was trying to follow what he did but when he show these commands I don't know how to correct them for my case.

```
sudo mkdir /usr/local/lib/python3.6/pyrealsense2
sudo ln -s ~/librealsense/build/wrappers/python/pybackend2.cpython-36-aarch64-linux-gnu.so /usr/local/lib/python3.6/pyrealsense2/
sudo ln -s ~/librealsense/build/wrappers/python/pyrealsense2.cpython-36-aarch64-linux-gnu.so /usr/local/lib/python3.6/pyrealsense2/
```

I looked into the **/librealsense/build/wrappers/python/pybackend2** folder but I dont have the **pybackend2.cpython-310...** or **pyrealsense2.cpython-310...** files

[image]

For **pybackend2.cpython-310...** I found it in none of the folders of the system but I found **pyrealsense2.cpython-310...** at **/home/librealsense/build/release** like last week

MartyG-RealSense, 2024-07-16:
In regard to the `sudo mkdir` and `sudo ln` lines, you would of course change the '3.6' references to '3.10' for your particular Python 3 version.  And Python 3.6 is no longer supported by pyrealsense2.  Python 3.7 support is also being removed in the next librealsense version as Python 3.12 support is officially introduced.

ClaudioCarbone, 2024-07-16:
> In regard to the `sudo mkdir` and `sudo ln` lines, you would of course change the '3.6' references to '3.10' for your particular Python 3 version. And Python 3.6 is no longer supported by [...]

@MartyG-RealSense Exactly, I typed `sudo mkdir /usr/local/lib/python3.10/pyrealsense2` and there is no problem there, but I looked into the **/librealsense/build/wrappers/python/pybackend2** folder but I dont have the **pybackend2.cpython-310...** or **pyrealsense2.cpython-310...** files

nlitz88, 2024-07-17:
@ClaudioCarbone I saw this issue earlier when trying to fix my own camera and wanted to share what worked for me. As a disclaimer, however, I have not read this entire thread and may be missing very key details that make my experience completely irrelevant! (just bear that in mind).

If you follow the rather confusing web of posts that stem from a [post I made on NVIDIA's Isaac ROS forums](https://forums.developer.nvidia.com/t/realsense-d435i-not-working-with-jetson-orin-nano-8gb-jetpack-6-0/298870), maybe it is possible to get your camera working in a similar way.

A summary of the steps I took:

1. Installed Jetpack 6.0 on my Jetson Orin Nano 8GB
2. Used NVIDIA's Isaac ROS Dev Docker container with their provided librealsense "add-on" Docker Image. Followed [these instructions from the Isaac ROS wiki](https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/sensors/realsense_setup.html#setup-instructions). This basically leaves you with a Docker container with built and installed inside of it.
3. (Only necessary for getting IMU working in parallel with RGB + depth capture, with realsense-ros in my experience): Downgraded my camera's firmware to version 5.13.0.50 via RealSense viewer on my windows laptop.

Again, I didn't read the entire history of the steps you've tried, so this might not be useful at all. I just wanted to put this here as a last-ditch possibility as it has worked for me so far. @MartyG-RealSense is probably your best bet either way, though!!

ClaudioCarbone, 2024-07-24:
> Hi @ClaudioCarbone Do you require further assistance with this case, please? Thanks!

I'm going to resume working on this as soon as I can, i will try @nlitz88 suggestion again since i tried it and it didn't work.

MartyG-RealSense, 2024-08-09:
Case closed due to no further comments received.

closed 2024-08-09 by MartyG-RealSense

Andd54, 2024-10-10:
Hi @MartyG-RealSense had an D435i working on Jetson Orin Nano, what bothers me is I tired to use the libuvc script ./scripts/libuvc_installation.sh within the repo, and I make sure to add python executable and related cmake options: cmake ../ -DFORCE_LIBUVC=true -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/home/orinnano/anaconda3/envs/pose_est_env/bin/python -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true. However, after the installation, I was able to use the realsense-viewer while the python wrapper pyrealsense2 was unable to detect any devices.

MartyG-RealSense, 2024-10-11:
Hi @Andd54  You would only need to use the `-DYPTHON_EXECUTABLE` parameter to install pyrealsense2 for a specific Python version if you have more than one version of Python installed on the computer (such as Python 2.7 and Python 3.10).  

If you do wish to use -DPYTHON_EXECUTABLE, the command **which python3** can be used to confirm which path to use with the parameter.

`-DFORCE_LIBUVC=true` and `-DFORCE_RSUSB_BACKEND=ON` perform the same function of bypassing the Linux kernel, so you only need to use one of them.  I would recommend removing **-DFORCE_LIBUVC=true**

It is worth bearing in mind though that it is a known issue with other RealSense users that sometimes realsense-viewer will work fine but the pyrealsense2 wrapper will not detect the camera.  There is not a known reason for this, unfortunately.