realsenseai/librealsense #13118: Realsense camera detected by the system but not by  realsense-viewer (Linux)

LeVraiBoris, 2024-07-03:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|

Realsense camera (sr305) is not detected by realsense-viewer : 
 - The camera is on an externally powered USB3 hub (plugin directly into the computer does not help).
 - The computer is a high end desktop workstation 
 - The camera is detected by the kernel:
 ```
#lsusb
Bus 002 Device 003: ID 0bda:0328 Realtek Semiconductor Corp. USB3.0-CRW
Bus 002 Device 005: ID 8086:0b48 Intel Corp. Intel(R) RealSense(TM) Depth Camera SR305
Bus 002 Device 004: ID 05e3:0612 Genesys Logic, Inc. Hub
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 006: ID 413c:2110 Dell Computer Corp. Dell Wired Multimedia Keyboard
Bus 001 Device 005: ID 413c:1010 Dell Computer Corp. USB 2.0 Hub [MTT]
Bus 001 Device 003: ID 046d:c01b Logitech, Inc. MX310 Optical Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
- The kernel module is up and running:
```
#modinfo uvcvideo |grep version
version:        1.1.1-realsense-1.3.26
srcversion:     A167E8DAE134FA179F7BFE3
vermagic:       6.5.0-41-generic SMP preempt mod_unload modversions 
```
- The camera can be used as a normal RGB Webcam
- Still Realsense-viewer does not find it :
```
03/07 18:34:04,417 INFO [123399045426368] (backend-v4l2.cpp:753) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.2/2-8.2.4/2-8.2.4:1.0/video4linux/video0
 03/07 18:34:04,418 INFO [123399045426368] (backend-v4l2.cpp:753) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.2/2-8.2.4/2-8.2.4:1.0/video4linux/video1
 03/07 18:34:04,418 INFO [123399045426368] (backend-v4l2.cpp:753) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.2/2-8.2.4/2-8.2.4:1.2/video4linux/video2
 03/07 18:34:04,418 INFO [123399045426368] (backend-v4l2.cpp:753) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.2/2-8.2.4/2-8.2.4:1.2/video4linux/video3
 03/07 18:34:04,419 INFO [123399045426368] (context.cpp:128) Found 0 RealSense devices (0xff requested & 0xff from device-mask in settings)
```
-Running the application as root leads to the same issue.

I don't know what else to try. Any suggestion would be highly appreciated. 
Thank you.

jeezrick, 2024-07-04:
build from source
```
mkdir build && cd build
cmake .. -DBUILD_PYTHON_BINDINGS=true -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true -DBUILD_WITH_CUDA=true  && make -j12 && sudo make install
```
follow [issue/13080](https://github.com/IntelRealSense/librealsense/issues/13080). 

**I don't know if you should do this on PC, but it certianly solve this problem for me on jetson orin.**

MartyG-RealSense, 2024-07-04:
Hi @LeVraiBoris  Official support for SR300-type cameras continued up until SDK 2.54.2.  Please try installing librealsense 2.54.2 instead of 2.55.1.

From the release notes for 2.55.1:

[image]

LeVraiBoris, 2024-07-05:
Downloaded version 2.50 from here: 
[https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0](url)
And compiled with: 
```
$ cd librealsense-2.50.0
$ mkdir build && cd build
$ cmake .. -DBUILD_PYTHON_BINDINGS=true -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release && make -j12 && sudo make install
```
This is working !
 Thank you very much.

closed 2024-07-05 by LeVraiBoris