realsenseai/librealsense #13119: what's the best way to setup librealsense2 and pyrealsense2 on jetpack5.1.3 (jetpack 5.1.3, l4t 35.5.0)

jeezrick, 2024-07-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.13.0 | 
| Operating System & Version |   ubuntu22  | 
| Kernel Version (Linux Only)    |  	5.15.136-tegra   | 
| Platform                                 | NVIDIA Jetson orin nx |

### Issue Description
I would like to share my status working with Realsense 455 on jetpack 6.0 after intense searching in exsiting issues, and hopefully we can make it right.

1. build from source
```
mkdir build && cd build
cmake .. -DBUILD_PYTHON_BINDINGS=true -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true -DBUILD_WITH_CUDA=true  && make -j12 && sudo make install
```
follow [issue/13080](https://github.com/IntelRealSense/librealsense/issues/13080). 

I think the `DFORCE_RSUSB_BACKEND=true` option fix the problem that camera can't be detected by realsense-viewer when user install sdk from debian. But although by doing this, camera can be detected, rgb stream is extreamly laggy, and viewer crashes after open motion module, exactly like [issues/13020](https://github.com/IntelRealSense/librealsense/issues/13020). 

2. download firmware version

The camera firmware version is 5.16 by default, so follow this [issues/12989](https://github.com/IntelRealSense/librealsense/issues/12989#:~:text=I%20have%20another%20D455%20with%20firmware%205.13.%20That%20one%20can%20work.%20So%20I%20downgrade%20this%20one%27s%20firmware%20to%205.13%20and%20it%20works.%20Thank%20you%20for%20your%20help.%20I%20think%20there%20may%20be%20some%20issue%20between%20the%20latest%20firmware%20and%20Jetson.), downgrade firmware to 5.13. Now motion module is working.

MartyG-RealSense, 2024-07-04:
Hi @jeezrick  Option 2 (a source code build) with `-DFORCE_RSUSB_BACKEND=ON` and `-DBUILD_WITH_CUDA=ON` flags included in the CMake build instruction typically works best on Jetson Orin.

**-DBUILD_WITH_CUDA=ON** enables the librealsense SDK's CUDA support, which accelerates point clouds, depth-color alignment and YUY to RGB color conversion.  It does not enhance any other processes except those three.  It accelerates them by offloading work from the Jetson's CPU onto its graphics GPU.

CUDA support is included and enabled by default in Option 1's debian packages for Jetson.  

The **installLibrealsense.sh** JetsonHacks build script installs the same debian packages (librealsense2-utils and librealsense2-dev) that Option 1 does, so CUDA support should be provided by that method.

jeezrick, 2024-07-04:
> Hi @jeezrick Option 2 (a source code build) with `-DFORCE_RSUSB_BACKEND=ON` and `-DBUILD_WITH_CUDA=ON` flags included in the CMake build instruction typically works best on Jetson Orin.
> 
> **-DBUILD_WITH_CUDA=ON** enables the librealsense SDK's CUDA support, which automatically accelerates point clouds, depth-color alignment and YUY to RGB color conversion. It does not enhance any other processes except those three. It accelerates them by offloading work from the Jetson's CPU onto its graphics GPU.
> 
> CUDA support is included and enabled by default in Option 1's debian packages for Jetson.
> 
> The **installLibrealsense.sh** JetsonHacks build script installs the same debian packages (librealsense2-utils and librealsense2-dev) that Option 1 does, so CUDA support should be provided by that method.

Thanks for the reply.
So the only diffirence between option 2 and the others is `-DFORCE_RSUSB_BACKEND=ON` since all method enabled cuda support, right? What diffirence does `-DFORCE_RSUSB_BACKEND=ON` make?

MartyG-RealSense, 2024-07-04:
Correct, the debian package builds will automatically include CUDA support whilst on a source code build does not and requires -DBUILD_WITH_CUDA to enable the support.

RSUSB = On builds of librealsense bypass the kernel and so reduce the likelihood that there will be errors because of a conflict between librealsense and the kernel version that is being used.

closed 2024-07-24 by MartyG-RealSense