realsenseai/librealsense #13120: UNKNOWN in rs2_get_stream_profiles(sensor:00000219E1CD0980): MFGetAttributeRatio(pMediaType, MF_MT_FRAME_RATE_RANGE_MIN, &frameRateMin.numerator, &frameRateMin.denominator) returned: HResult 0xc00d36e6: "无法找到请求的属性。

LeaMaps, 2024-07-04:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|
|**Name**|Platform Camera|
|**Serial Number**|10400b6|
|**Usb Type Descriptor**|2.01|

When I try to use our camera with realsense viewer, some errors occured. And I can view the RGB image with Window's camera setting. It  proves our camera is connected and the computer recognizes the camera. I want to know how to solve this problem.

**UNKNOWN in rs2_get_stream_profiles(sensor:00000219E1CD0980):
MFGetAttributeRatio(pMediaType, MF_MT_FRAME_RATE_RANGE_MIN,
&frameRateMin.numerator, &frameRateMin.denominator) returned: HResult
0xc00d36e6: "无法找到请求的属性。"
[image]**

[image]

MartyG-RealSense, 2024-07-04:
Hi @LeaMaps  The **Platform Camera** is a non-RealSense RGB camera, typically a USB webcam or the built-in webcam of a laptop.  Such a camera may be viewed in the 'Camera' app of Windows but the Platform Camera mode of the RealSense Viewer will not work with all cameras. 

For example, the built-in webcam of one laptop might work fine with it but the webcam on another laptop of a different make / model will not.  There is not a solution available if a particular webcam cannot be accessed by the Viewer as a Platform Camera, unfortunately.

********

I note that the bottom of your image says '515'.  It looks as though you are attempting to access a RealSense L515 camera in the Viewer but it is not being detected.  This is because support for L515 was removed from the current latest 2.55.1 version because it is a retired camera model.  If you use the previous **2.54.2** version then your L515 should be able to work with it.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

LeaMaps, 2024-07-05:
> Hi @LeaMaps The **Platform Camera** is a non-RealSense RGB camera, typically a USB webcam or the built-in webcam of a laptop. Such a camera may be viewed in the 'Camera' app of Windows but the Platform Camera mode of the RealSense Viewer will not work with all cameras.嗨，平台摄像头是非实感 RGB 摄像头，通常是 USB 网络摄像头或笔记本电脑的内置网络摄像头。可以在 Windows 的“相机”应用程序中查看此类相机，但实感查看器的平台相机模式不适用于所有相机。
> 
> For example, the built-in webcam of one laptop might work fine with it but the webcam on another laptop of a different make / model will not. There is not a solution available if a particular webcam cannot be accessed by the Viewer as a Platform Camera, unfortunately.例如，一台笔记本电脑的内置网络摄像头可能工作正常，但另一台不同品牌/型号的笔记本电脑上的网络摄像头则不行。不幸的是，如果查看器无法将特定网络摄像头作为平台摄像机访问，则没有可用的解决方案。
> 
> I note that the bottom of your image says '515'. It looks as though you are attempting to access a RealSense L515 camera in the Viewer but it is not being detected. This is because support for L515 was removed from the current latest 2.55.1 version because it is a retired camera model. If you use the previous **2.54.2** version then your L515 should be able to work with it.我注意到您图像的底部写着“515”。您似乎正在尝试访问查看器中的实感 L515 摄像头，但未检测到该摄像头。这是因为对 L515 的支持已从当前最新的 2.55.1 版本中删除，因为它是已停用的相机型号。如果您使用以前的 2.54.2 版本，那么您的 L515 应该能够使用它。
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

I have solved my problem, thanks for your help.

MartyG-RealSense, 2024-07-05:
You are very welcome.  :)

closed 2024-07-18 by MartyG-RealSense