realsenseai/librealsense #13122: Incorrect frame timestamps from the future

Egor18, 2024-07-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15                                         | 
| Platform                                 | PC |
| SDK Version                            |  v2.50.0                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
Hi!

We use D435 with librealsense v2.50.0 and have noticed that sometimes the incoming frames have timestamps from the future, meaning the frame time is greater than the system time upon arrival.

Here’s the plot of this happening created from the frame metadata:
[image]

The above is an extreme example, sometimes it’s only 10-100ms from the future.

The clock domain we use is `RS2_TIMESTAMP_DOMAIN_GLOBAL_TIME`, meaning that [global_timestamp_reader](https://github.com/IntelRealSense/librealsense/blob/v2.50.0/src/global_timestamp_reader.cpp) calculates the frame timestamp using linear regression.
```
RS2_TIMESTAMP_DOMAIN_GLOBAL_TIME,    /**< Frame timestamp was measured in relation to the camera clock and converted to OS system clock by constantly measure the difference*/
```

I found [this PR](https://github.com/IntelRealSense/librealsense/pull/12401) that reports a similar issue and suggests that the timestamps from the future are caused by occasional increases in the duration of getting the hw time from the device (`_device->get_device_time_ms()` call) because it skews the linear regression. [Another PR](https://github.com/IntelRealSense/librealsense/pull/6043) also seems to confirm this theory.

So I added debug logs to measure the duration of the `_device->get_device_time_ms()` call and the difference between system time and frame time.
```
diff --git a/src/global_timestamp_reader.cpp b/src/global_timestamp_reader.cpp
index be8f5c24a..a4eaf77b5 100644
--- a/src/global_timestamp_reader.cpp
+++ b/src/global_timestamp_reader.cpp
@@ -208,6 +208,7 @@ namespace librealsense
             double system_time_start = duration<double, std::milli>(system_clock::now().time_since_epoch()).count();
             double sample_hw_time = _device->get_device_time_ms();
             double system_time_finish = duration<double, std::milli>(system_clock::now().time_since_epoch()).count();
+            LOG_WARNING("Duration of getting the hardware clock: " << system_time_finish - system_time_start << "(ms)");
             double command_delay = (system_time_finish-system_time_start)/2;
 
             std::lock_guard<std::recursive_mutex> lock(_read_mtx);
```
[image]
[image]
Here’s the distribution of the `_device->get_device_time_ms()` durations:
[image]
```
0.0 - 1.0: 93.35%
1.0 - 2.0: 6.11%
2.0 - 3.0: 0.23%
3.0 - 4.0: 0.17%
4.0 - 5.0: 0.02%
...
100.0 - 101.0: 0.00%
101.0 - 102.0: 0.01%
102.0 - 103.0: 0.03%
```
As you can see 99.7% of all calls take 0 - 2 ms, however, there are a few taking 100ms, and on the other machine I tried it goes up to 1000ms!

It’s pretty easy to reproduce, you just need to run the camera for 12-24hrs.

So I did a simple experiment and excluded the samples where the duration is more than 10ms:
```
            if ((system_time_finish - system_time_start) < 10.0) {
                _coefs.add_value(crnt_sample);
            } else {
                LOG_WARNING("Got unreliable measurement, ignoring");
            }
```
I let it run for about 12hrs, and here's the result:
[image]
As you can see it improved the situation a lot (although not a proper fix).

It is not clear what causes the occasional increases in the duration of the `_device->get_device_time_ms()` calls up to 100-1000ms from 1-2ms since it’s something in the firmware, however, I hope that at least the global timestamp reader could be fixed to handle this situation better. We could either try to ignore the outliers or ensure that the frame timestamp calculated by regression is at least not more than system time.

Frame timestamps from the future are especially bad for us because we use ROS and there’s obviously no [tfs](http://wiki.ros.org/tf) in the tf buffer for the frames from the future, however, I imagine this could be causing all sorts of trouble for other users as well.

Egor18, 2024-07-04:
This is how I build librealsense, although originally we noticed the issue in librealsense from deb:
```
cmake -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true ..
make -j8
```
Here's the patch you can apply:
```
diff --git a/src/global_timestamp_reader.cpp b/src/global_timestamp_reader.cpp
index be8f5c24a..9dfe5e406 100644
--- a/src/global_timestamp_reader.cpp
+++ b/src/global_timestamp_reader.cpp
@@ -222,7 +222,11 @@ namespace librealsense
                 _coefs.update_samples_base(sample_hw_time);
             }
             CSample crnt_sample(sample_hw_time, system_time);
-            _coefs.add_value(crnt_sample);
+            if ((system_time_finish - system_time_start) < 2.0) {
+                _coefs.add_value(crnt_sample);
+            } else {
+                LOG_WARNING("Got unreliable measurement, ignoring. Duration of getting the hardware clock: " <<  system_time_finish - system_time_start << "(ms)");
+            }
             _is_ready = true;
             return true;
         }
@@ -281,13 +285,20 @@ namespace librealsense
 
     double global_timestamp_reader::get_frame_timestamp(const std::shared_ptr<frame_interface>& frame)
     {
+        double system_time = duration<double, std::milli>(system_clock::now().time_since_epoch()).count();
         double frame_time = _device_timestamp_reader->get_frame_timestamp(frame);
         rs2_timestamp_domain ts_domain = _device_timestamp_reader->get_frame_timestamp_domain(frame);
         if (_option_is_enabled->is_true() && ts_domain == RS2_TIMESTAMP_DOMAIN_HARDWARE_CLOCK)
         {
             auto sp = _time_diff_keeper.lock();
             if (sp)
+            {
                 frame_time = sp->get_system_hw_time(frame_time, _ts_is_ready);
+                if (frame_time > system_time)
+                {
+                    LOG_ERROR("Got future timestamp. Future=" << frame_time - system_time << "(ms)");
+                }
+            }
             else
                 LOG_DEBUG("Notification: global_timestamp_reader - time_diff_keeper is being shut-down");
         }
```
Let D435 run for 12-24h and it's pretty much guaranteed that you'll get several "Got unreliable measurement" and durations of getting hardware clock in 100s ms.

Egor18, 2024-07-04:
I would appreciate if someone could take a look at this.
Thanks.

Egor18, 2024-07-05:
I also noticed that every time when the duration of getting the hw clock is huge there's always this message as well:
```
(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```
[image]
I reproduced it on 2 different machines as well.

MartyG-RealSense, 2024-07-06:
Hi @Egor18  There have been a few previous cases (though not recently) of timestamps that are in the future, such as https://github.com/IntelRealSense/librealsense/issues/10068 which suggests disabling Global Time to see whether it corrects the future timestamp problem.  

Another suggestion is using **ros::Time::now()** to get transforms at the current time.

https://answers.ros.org/question/331027/realsense-d435-lookup-would-require-extrapolation-into-the-future/

If the **control_transfer returned** warning is generating continuously in the log then it indicates that there may be a communication problem between the camera and the computer.  For example, the computer's CPU may be struggling to keep up with processing the frames that are arriving at the USB port from the camera.

Egor18, 2024-07-09:
Hi @MartyG-RealSense 
> Another suggestion is using ros::Time::now() to get transforms at the current time.

Yeah, that's exactly what we had to do to deal with this issue, however, I wanted to report the root cause because this means that global timestamp reader and global time are unreliable. My understanding is that the calculated global time was supposed to be more accurate than just using ros::Time::now().

Anyway, it actually might be a good idea to add a flag to https://github.com/IntelRealSense/realsense-ros to stamp frames using just the host time.

> If the control_transfer returned warning is generating continuously in the log then it indicates that there may be a communication problem between the camera and the computer. For example, the [...]

It does not happen all the time (just a bunch of times in 12h), but every time it happens there's a spike in the duration of the `device->get_device_time_ms()` call.

MartyG-RealSense, 2024-07-09:
Thanks very much for the suggestion.  It is actually okay to disable global time when using ROS because unlike librealsense, ROS has its own timing mechanism to help to avoid time drift, as advised at https://github.com/IntelRealSense/realsense-ros/issues/796#issuecomment-497997643 by the Intel RealSense developer who created the ROS1 wrapper.

closed 2025-01-06 by MartyG-RealSense