realsenseai/librealsense #13125: Pyrealsense2 Runtime Error: No Decive Connected

desq28, 2024-07-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   {Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  5.15.136-tegra                                         | 
| Platform                                 | Jetson Orin nano |
| SDK Version                            |  2.55.1 }                          | 
| Language                            |  Python }                          | 

### Issue Description
Hi.

I want to connect the camera through pyrealsense and so far I have been unsuccesful. Basically, I get `pipe.start()
RuntimeError: No device connected` everytime I run a Python file. I also tried running the examples from wrappers/python/examples and all of them get stuck in the "No device connected" error. Realsense-viewer works fine, `rs-ennumerate-devices` always detects the camera, but running the examples from the SKD is inconsistent. Hello-Realsense does not work unless I downgrade the FW to 5.13.0.50. However, other examples such as RS-distance properly detect the camera with the recent FW and run with no issues. The fact that some examples work makes me think this is not a hardware issue.

I followed the [Jetson Hacks](https://github.com/JetsonHacksNano/installLibrealsense) tutorial for installing Librealsense from source using ./buildLibrealsense.sh -j 2. 

Any advice will be greatly appreciated, thanks in advanced.

MartyG-RealSense, 2024-07-06:
Hi @desq28  Is the camera detected if you launch your Python script in **sudo** admin permissions mode?  For example:

`sudo python3 test.py `

desq28, 2024-07-08:
Hi Marty. 

I tried using a brief example from the align depth to color example.
```
import pyrealsense2 as rs  

pipeline = rs.pipeline()

config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipeline.start(config)
```

I still get the same error     `profile = pipeline.start(config), RuntimeError: No device connected`

I did the build process trying with the master branch and the development branch from Librealsense (after uninstalling previous build), using the following command, but I still get stuck with the RuntimeError. Realsense-viewer still correctly detects the camera.

 `mkdir build
cd build
cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true
make -j2
sudo make install`

I will attempt to downgrade back to Jetpack 5.1.3 and check if the library works, but I would appreciate your help to get it running in JP 6.0.

Thanks!

desq28, 2024-07-08:
Hey @MartyG-RealSense. Couldn't figure it out, so I installed the development branch of Librealsense and then tried running examples such as align-depth2color.py or opencv-viewer_example.py and, indeed, it works and the camera is detected. I can't believe the solution was so simple. Why do I need to use sudo? Did I miss any steps which grant permissions?

Thanks for your help.

I followed this commands to install, just in case it helps anyone.
```
cd librealsense
mkdir build
cd build
cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true
make -j2
sudo make install
```

MartyG-RealSense, 2024-07-09:
You are very welcome. I'm pleased to hear that you achieved a solution.  Thanks very much for sharing the details of it!

When performing a build of librealsense from the source code folder, the **udev rules** that handle device access permissions are installed by running the `./scripts/setup_udev_rules.sh` udev permissions script in the root directory of the source code folder (the same directory that the **mkdir build** command is used in).  

If there is a problem with device access that is caused by a problem with the udev rules (such as them being missing) then sudo admin power can be used to run a program.

desq28, 2024-07-10:
Hi @MartyG-RealSense. 

Just checked and the file 99-realsense-libusb.rules is indeed located at /etc/udev/rules.d/. Do you have any other ideas about what may be causing issues with the permissions? Also, marking this as closed since main issue was solved.

Thanks!

closed 2024-07-10 by desq28