realsenseai/librealsense #13130: D455 Errors when activating imu stream in rs-viewer

timbojones91, 2024-07-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |  Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  5.10.0-1009-rockchip                                         | 
| Platform                                 | Rock 5A SBC  |
| SDK Version                            |  {2.55.1}                          | 
| Language                            |  none                          | 
| Segment			| robot     | 

### Issue Description

Hello 

ive installed a fresh 22.04 [https://github.com/Joshua-Riek/ubuntu-rockchip] for Rock 5a sbc. 
Camera works fine under win10.

ive build the latest sdk from source using `./scripts/setup_udev_rules.sh` and the following cmake argument:

`cmake ../ -DBUILD_EXAMPLES=true -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_GRAPHICAL_EXAMPLES=TRUE`

`sudo make uninstall && make clean && make && make -j$(($(nproc)-1)) && sudo make install`

<
<
<

realsense-viewer works with rgb and depth stream. 

if i start up realsense-viewer ive got this errors/warning:

```
18:13:37 [Info]  - librealsense version: 2.55.1

18:13:37 [Info] /home/nvu/librealsense/src/rs.cpp:3103 - Framebuffer size changed to 1850 x 1016
18:13:37 [Info] /home/nvu/librealsense/src/rs.cpp:3103 - Window size changed to 925 x 508
18:13:37 [Info] /home/nvu/librealsense/src/rs.cpp:3103 - Scale Factor is now 2
18:13:37 [Error] /home/nvu/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_IO
18:13:37 [Error] /home/nvu/librealsense/src/uvc-sensor.cpp:412 - acquire_power failed: failed to set power state
18:13:37 [Error] /home/nvu/librealsense/src/rs.cpp:237 - [rs2_create_device( info_list:0xaaaabf9aee90, index:0 ) UNKNOWN] failed to set power state
18:13:37 [Error] /home/nvu/librealsense/src/rs.cpp:237 - [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
18:13:37 [Warn] /home/nvu/librealsense/src/rs.cpp:373 - null pointer passed for argument "device"
18:13:37 [Warn] /home/nvu/librealsense/src/rs.cpp:3106 - Couldn't refresh devices - failed to set power state
18:13:38 [Info] /home/nvu/librealsense/src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
18:13:38 [Info] /home/nvu/librealsense/src/context.cpp:116 - ... 6-1-6
18:13:38 [Info] /home/nvu/librealsense/src/context.cpp:128 - Found 1 RealSense devices (0xfe requested & 0xff from device-mask in settings)
18:13:39 [Info] /home/nvu/librealsense/src/environment.cpp:136 - Found 3 unreachable streams, 3 extrinsics deleted
18:13:46 [Info]  - Intel RealSense D455 was selected as a default device
18:13:46 [Info] /home/nvu/librealsense/common/notifications.cpp:511 - Intel RealSense D455      Connected

18:13:46 [Info] /home/nvu/librealsense/common/notifications.cpp:538 - Current SW version: 2.55.1.0
Recommended SW version: 2.55.1.6486
18:13:46 [Debug]  - Current FW >= Bundled FW for: Intel RealSense D455      (S/N 031422250402)
Current Version: 5.16.0.1
Recommended Version: 5.16.0.1
18:13:47 [Info] /home/nvu/librealsense/src/core/notification.cpp:22 - Motion Module force pause
18:13:47 [Error] /home/nvu/librealsense/common/notifications.cpp:511 - Motion Module force pause

```

As soon i start the imu stream stereo and rgb streams stop and ive got the following errors:

```
18:11:18 [Error] /home/nvu/librealsense/src/error-handling.cpp:99 - Error during polling error handler: get_xu(id=7) failed! Last Error: Success
18:11:21 [Error] /home/nvu/librealsense/src/error-handling.cpp:99 - Error during polling error handler: get_xu(id=7) failed! Last Error: Success
18:11:24 [Error] /home/nvu/librealsense/src/global_timestamp_reader.cpp:239 - Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
18:11:24 [Error] /home/nvu/librealsense/src/error-handling.cpp:99 - Error during polling error handler: get_xu(id=7) failed! Last Error: Success
18:11:27 [Error] /home/nvu/librealsense/src/error-handling.cpp:99 - Error during polling error handler: get_xu(id=7) failed! Last Error: Success
18:11:30 [Error] /home/nvu/librealsense/src/global_timestamp_reader.cpp:239 - Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
18:11:30 [Error] /home/nvu/librealsense/src/error-handling.cpp:99 - Error during polling error handler: get_xu(id=7) failed! Last Error: Success
```

<
<
<
<

this wonders me also

ive installed it today via cloning the librealsense.git 
why is he saying im not on the newest version?

```

18:13:46 [Info] /home/nvu/librealsense/common/notifications.cpp:538 - Current SW version: 2.55.1.0
Recommended SW version: 2.55.1.6486
```

2.55.1.6486 cant also found in releases on github?

MartyG-RealSense, 2024-07-09:
Hi @timbojones91  There is typically a slight difference in version number but it is not important and your 2.55.1.0 version is fine.  

When starting the RealSense Viewer, there are sometimes momentary hiccups in accessing streams that generate a warning but the problem auto-corrects.  In the case of the **Motion Module force pause** error, enabling the IMU streams after the error has occurred typically results in the IMU working normally.

There is a known issue with the Viewer for some RealSense users where, when three streams are enabled, if one of those stream types is the Motion Module then one of the streams will fail.  There is no fix in the Viewer for this issue except to only use two streams (depth + IMU or RGB + IMU) or only enable the Motion Module.

The RealSense example program **rs-capture**, which can also display depth, RGB and IMU, is not affected by this problem.  

[image]

timbojones91, 2024-07-09:
```
nvu@zorroNVU:~$ rs-capture
 09/07 20:11:11,503 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:12,674 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:13,296 ERROR [281473046933696] (d400-thermal-monitor.cpp:97) Error during thermal compensation handling: get_xu(id=15) failed! Last Error: Success
 09/07 20:11:13,850 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:15,041 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:16,229 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:16,391 ERROR [281473046933696] (d400-thermal-monitor.cpp:97) Error during thermal compensation handling: get_xu(id=15) failed! Last Error: Success
 09/07 20:11:17,418 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:18,588 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:19,476 ERROR [281473046933696] (d400-thermal-monitor.cpp:97) Error during thermal compensation handling: get_xu(id=15) failed! Last Error: Success
 09/07 20:11:19,766 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:20,927 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:22,101 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:22,540 ERROR [281473046933696] (d400-thermal-monitor.cpp:97) Error during thermal compensation handling: get_xu(id=15) failed! Last Error: Success
 09/07 20:11:23,264 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:24,416 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:25,346 ERROR [281473400486464] (rs.cpp:237) [rs2_pipeline_wait_for_frames( pipe:0xaaaaedf02120 ) UNKNOWN] Frame didn't arrive within 15000
^A 09/07 20:11:25,590 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 09/07 20:11:25,610 ERROR [281473046933696] (d400-thermal-monitor.cpp:97) Error during thermal compensation handling: get_xu(id=15) failed! Last Error: Success
 09/07 20:11:26,781 ERROR [281473181397184] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0xaaaaedf02120):
    Frame didn't arrive within 15000
nvu@zorroNVU:~$ 

```

unfortanely this doesnt work. 

in realsense viewer i can open depth and rgb stream and it works just fine. if i open imu stream, the connection to the other streams gets lost and it throws the above mentioned errors. if i enable only the imu this happens too. if i had imu enabled and disabled again, the camera still doesnt work. 

eg. opening rgb stream after imu stream on/off leads to following error:

```
Invalid Value in rs2_open_multiple(sensor:0xaaaad21c62d0,
profiles:0xaaaad24795e0, count:1):

Failed to resolve the request:
	Format: RGB8, width: 1280, height: 720

Into:
	Formats:
	 YUYV
```

timbojones91, 2024-07-09:
ros_realsense is also non working. 
```

ros2 run realsense2_camera realsense2_camera_node
# or, with parameters, for example - temporal and spatial filters are enabled:
ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_color:=false -p spatial_filter.enable:=true -p temporal_filter.enable:=true
```

timbojones91, 2024-07-10:
Installation of realsense-ros bricked the realsense sdk. 

now the sdk doesnt recognise the camera at all. 

-
-
event recording of plugging in:
-
`nvu@zorroNVU:~$ sudo udevadm monitor
[sudo] password for nvu: 
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[1032.401033] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1 (usb)
KERNEL[1032.412692] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0 (usb)
KERNEL[1032.413397] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0/video4linux/video0 (video4linux)
KERNEL[1032.413469] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0/video4linux/video1 (video4linux)
KERNEL[1032.414091] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0/video4linux/video2 (video4linux)
KERNEL[1032.414316] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0/video4linux/video3 (video4linux)
KERNEL[1032.414508] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0/media0 (media)
KERNEL[1032.414579] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0/input/input16 (input)
KERNEL[1032.414706] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0/input/input16/event5 (input)
KERNEL[1032.414904] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0 (usb)
KERNEL[1032.414979] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.1 (usb)
KERNEL[1032.415263] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.1 (usb)
KERNEL[1032.416364] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.2 (usb)
KERNEL[1032.416982] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.2 (usb)
KERNEL[1032.417193] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.3 (usb)
KERNEL[1032.417386] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.3/video4linux/video4 (video4linux)
KERNEL[1032.417578] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.3/video4linux/video5 (video4linux)
KERNEL[1032.417769] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.3/media1 (media)
KERNEL[1032.417974] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.3 (usb)
KERNEL[1032.418178] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.4 (usb)
KERNEL[1032.418381] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.4 (usb)
KERNEL[1032.418582] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.5 (usb)
KERNEL[1032.418782] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.5/0003:8086:0B5C.0009 (hid)
KERNEL[1032.418976] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.5/usbmisc/hiddev0 (usbmisc)
KERNEL[1032.419085] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.5/0003:8086:0B5C.0009/hidraw/hidraw0 (hidraw)
KERNEL[1032.419650] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.5/0003:8086:0B5C.0009 (hid)
KERNEL[1032.419857] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.5 (usb)
KERNEL[1032.420087] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1 (usb)
UDEV  [1032.433335] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1 (usb)
UDEV  [1032.438886] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.2 (usb)
UDEV  [1032.439049] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.1 (usb)
UDEV  [1032.439803] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.3 (usb)
UDEV  [1032.441683] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.2 (usb)
UDEV  [1032.442501] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0 (usb)
UDEV  [1032.442786] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.1 (usb)
UDEV  [1032.448733] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.5 (usb)
UDEV  [1032.449685] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.3/media1 (media)
UDEV  [1032.456505] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0/input/input16 (input)
UDEV  [1032.457933] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.4 (usb)
UDEV  [1032.464809] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.0/media0 (media)
UDEV  [1032.464907] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.5/0003:8086:0B5C.0009 (hid)
UDEV  [1032.468218] bind     /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.4 (usb)
UDEV  [1032.468951] add      /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.5/usbmisc/hiddev0 (usbmisc)
[... cut ...]
10:06:41 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.3/video4linux/video4
10:06:41 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.7.auto/usb6/6-1/6-1:1.3/video4linux/video5
10:06:41 [Info] ./src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
10:06:41 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)

```

thank you for helping

MartyG-RealSense, 2024-07-10:
The error **Failed to read busnum/devnum** is very rare.  It always happens with single board computers and usually means that the USB port that the camera is plugged into was unable to supply the camera with sufficient power to operate.  Attaching the camera to a mains electricity powered USB hub instead of plugging the camera directly into a port on the board can help with this problem, because then the camera is drawing power from the mains plug socket instead of the computing board.

In regard to the SDK no longer working after installing the ROS wrapper: did you install the ROS1 or ROS2 wrapper?  

If you did install the ROS1 wrapper, did you do so with the command below?

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

This command installs both the librealsense SDK and the ROS1 wrapper at the same time.  So if there was a librealsense installation already on the computer then you could end up with two SDK installations that conflict with each other.

lyfadvance, 2024-07-12:
I have the same problem.but my platform is ubunt22.04 and rk3588 board.
 When I open the Realsense Viewer, RGB and stereo data can be acquired normally, but IMU data acquisition fails.
the error information of realsense viewer is :
Error during polling error handler: get_xu(id=7) failed! Last Error: Success
and
uvc-sensor.cpp 412 - acquire_power failed: failed to set power state
and 
failed to open usb interface:0, error: RS2_USB_STATUS_NO_DEVICE

lyfadvance, 2024-07-12:
@MartyG-RealSense

lyfadvance, 2024-07-12:
but, when i run realsense-viewer on vmware ubuntu20.04, it runs all fine.

MartyG-RealSense, 2024-07-12:
Hi @lyfadvance  When installing the librealsense SDK on an RK board, it should be built from source code with CMake with the flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction.  Is this the method that you used, please?

Also, when enabling the IMU in the RealSense Viewer, are you enabling only the IMU or are you enabling depth, RGB and Motion Module simultaneously?

lyfadvance, 2024-07-12:
I used the flag -DFORCE_RSUSB_BACKEND=TRUE.
And, to prevent it from being a power issue, I used a USB hub to pass through.
I only enabled IMU。When I only activate the IMU, the realsense viewer shows abnormalities. Additionally, after activating the IMU, if I activate the RGB camera, the RGB camera also malfunctions.

lyfadvance, 2024-07-12:
I was using SDK version 2.55 for the D455, but the D455 did not work properly. I have also used SDK 1.53 on an RK board to run the T265, and the T265 worked fine. 
 the firmware of d455 I used is newest。

lyfadvance, 2024-07-12:
can I rollback the version of  firmware of D455 for test.

lyfadvance, 2024-07-12:
@MartyG-RealSense

MartyG-RealSense, 2024-07-12:
You do not need to use my tag name each time.  I get automatically notified of new comments.  :)

Yes, you can downgrade the firmware, though there is a limit of 20 firmware downgrades during the lifetime of the camera and then downgrades are permanently locked and you can then only upgrade to a newer version than the currently installed one.

You can change your firmware version by first going to the firmware downloads page and downloading a .bin file of a firmware.

https://dev.intelrealsense.com/docs/firmware-releases-d400

**I cannot emphasize strongly enough** that if you have a D455 then you should not install a firmware version older than 5.13.0.50, otherwise the camera may become unusable.  It is okay to use 5.13.0.50.

Then go to the 'More' option near the top of the Viewer's options side-panel and select from its drop-down menu **Update Firmware**. A file selector box will pop up. Navigating to the location that you downloaded the .bin file to and selecting the file will automatically begin the firmware update.

timbojones91, 2024-07-12:
**Marty you are a genius!** 
Flashing 5.13.0.50 was the right hint for me. 

@lyfadvance test that 5.13.0.50 ill bet it works for you too. 

too bad ive sat the whole day on a power injector usb c cable to externally power the d455 without loosing the host vcc reset capability... 
a lot of effort without success...

MartyG-RealSense, 2024-07-12:
@timbojones91  You are very welcome!  I'm pleased to hear that a firmware change worked well for you.  :)

The **control_transfer returned** warning, if it is generating continuously in the log, can indicate a communication problem between the camera and the computer / computing board, typically a USB related problem such as the cable or the computer's port.  It could also occur if the computer's processing is unable to keep up with the rate at which new frames are arriving at the USB port.

If only a small number of the warnings appear in the log and then stop then you can safely ignore them.  Even when the warnings are generating continuously, the topics may sometimes still be published normally in RViz, in which case you can also ignore the warnings if RViz is behaving fine.

closed 2024-07-12 by timbojones91

lyfadvance, 2024-07-15:
you are right. There is some bug in newest firmware.