realsenseai/librealsense #13131: 455 detected as USB2.1

nataliesettles, 2024-07-08:
|---------------------------------|------------------------------------------- |
| Camera Model                       |  455  | 
| Firmware Version                   | current  | 
| Operating System & Version |   Win 10  |                                  | 
| Platform                                 | PC |
| SDK Version                            |  2.55.1.6486 }                          | 

I know this problem has been mentioned before, but unfortunately none of the fixes I came across in those threads have helpe me so far. It’s possible I may have missed other relevant threads.

After spending some time away from the set-up for my 455d, I came back to it to discover on boot up that it was now being detected as a USB 2.1 device in the RealSense viewer. The only change I made was that before I booted up the computer I swapped out the old powered USB hub I’d been using for a new Newnex FireNex uHub (a faster device I was hoping would address latency). I did shut everything down again and use the old hub but the problem persists. My set up uses a 20m active Newnex cable and a FireNex uHub. Any sense how I can fix this so it’s finally recognized as a USB3 device again?

Is there a preferred plugin order if I unplug and replug in all the components? Is it better to do while the computer is shut down or running?

MartyG-RealSense, 2024-07-09:
Hi @nataliesettles  Please first try unplugging the micro-sized end of the USB cable from the base of the camera, turning the connector around the other way and re-inserting it (USB-C cables are two-way insertion at the micro-sized end).

If flipping the connector direction does not make a difference, next try unplugging the full-sized end of the USB cable from the hub, then re-inserting it with a firm quick motion and not a slow insertion.  Slow insertion can cause a RealSense camera to be mis-detected as a USB 2.1 connection.

nataliesettles, 2024-07-09:
Hi @MartyG-RealSense - Thanks for your reply. Unfortunately neither of those approaches fixed the problem. I have noticed that my antivirus software is blocking an Intel driver called iqvw64e.sys — could this be related to the problem?

MartyG-RealSense, 2024-07-10:
Apparently iqvw64e is a software component of the Intel Network Adapter Diagnostic Driver, so it seems unlikely that a blocking of it is the cause of your USB 2.1 problem.

Are you using a USB hub device or your own choice of USB cable?  If either a hub or a cable are USB2 instead of USB3 then a RealSense camera will be detected as 2.1 even if the cable or hub is attached to a USB3 port on the computer.

nataliesettles, 2024-07-10:
As you suspected, it was one of the cables. Thanks for your help!

MartyG-RealSense, 2024-07-10:
You are very welcome.  I'm pleased to hear that you confirmed what the problem was.  Thanks very much for the update!

closed 2024-07-18 by MartyG-RealSense