realsenseai/librealsense #13137: Depth Accuracy

linahourieh, 2024-07-09:
Hello there,
I have some recorded RGB-D videos using D435 and real sense viewer, is there a way to get a kind of a metric that tells us how good the depth data was for each video? 
Best Regards,

MartyG-RealSense, 2024-07-09:
Hi @linahourieh  Whilst there is not a tool for providing feedback about the depth quality of recorded files, Intel's **Camera Depth Testing Methodology** PDF guide at the link below does outline a formal procedure for evaluating depth quality and discusses metrics.

https://dev.intelrealsense.com/docs/camera-depth-testing-methodology

MartyG-RealSense, 2024-07-18:
Hi @linahourieh  Do you require further assistance with this case, please?  Thanks!

linahourieh, 2024-07-18:
No that was helpful, Thank you for your assistance.

MartyG-RealSense, 2024-07-18:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks very much for the update!

closed 2024-07-18 by MartyG-RealSense