realsenseai/librealsense #13139: D455 imu and color/depth sync in python

jeezrick, 2024-07-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

```
Accel: x: -0.755112, y: -9.5811, z: -0.519752 at 1720582802.4813268, Gyro: x: -0.00159698, y: -0.00106465, z: 0 at 1720582802.4836738, fps=427.20554084334896
Accel: x: -0.755112, y: -9.60071, z: -0.509946 at 1720582802.4861624, Gyro: x: -0.00159698, y: -0.00159698, z: 0.000532325 at 1720582802.4861088, fps=290.86712898751733
Accel: x: -0.755112, y: -9.60071, z: -0.509946 at 1720582802.4861624, Gyro: x: -0.00106465, y: -0.000532325, z: 0.00159698 at 1720582802.488544, fps=425.213300892133
Accel: x: -0.764919, y: -9.5909, z: -0.539366 at 1720582802.490998, Gyro: x: -0.00106465, y: 0.00159698, z: 0.0021293 at 1720582802.4909785, fps=425.213300892133
Accel: x: -0.764919, y: -9.5909, z: -0.539366 at 1720582802.490998, Gyro: x: -0.00159698, y: 0.0031957, z: 0.00266163 at 1720582802.493413, fps=424.99787212483534
Accel: x: -0.774725, y: -9.56148, z: -0.539366 at 1720582802.4958327, Gyro: x: -0.00159698, y: 0.0031957, z: 0.00266163 at 1720582802.493413, fps=425.60162354134957
Accel: x: -0.774725, y: -9.56148, z: -0.539366 at 1720582802.4958327, Gyro: x: -0.0021293, y: 0.00372802, z: 0.00266163 at 1720582802.4958475, fps=791.0795926065636
Accel: x: -0.774725, y: -9.56148, z: -0.539366 at 1720582802.4958327, Gyro: x: -0.00159698, y: 0.0031957, z: 0.00266163 at 1720582802.4982815, fps=777.8753709198813
Accel: x: -0.764919, y: -9.55168, z: -0.529559 at 1720582802.5006657, Gyro: x: -0.00159698, y: 0.0031957, z: 0.00266163 at 1720582802.4982815, fps=424.7396455696203
Accel: x: -0.764919, y: -9.55168, z: -0.529559 at 1720582802.5006657, Gyro: x: -0.00159698, y: 0.00266163, z: 0.0031957 at 1720582802.5007153, fps=5706.536054421768
Accel: x: -0.764919, y: -9.55168, z: -0.529559 at 1720582802.5006657, Gyro: x: -0.0021293, y: 0.00266163, z: 0.00266163 at 1720582802.503148, fps=427.1620327935635
Processed depth at 1720582802.539346 and color frames at 1720582802.5393534, fps=30.144704216646662
Accel: x: -0.764919, y: -9.5811, z: -0.549172 at 1720582802.505498, Gyro: x: -0.0021293, y: 0.00266163, z: 0.00266163 at 1720582802.503148, fps=425.7744391432342
Accel: x: -0.764919, y: -9.5811, z: -0.549172 at 1720582802.505498, Gyro: x: -0.00266163, y: 0.00266163, z: 0.00159698 at 1720582802.5055816, fps=5667.978378378379
Accel: x: -0.764919, y: -9.5811, z: -0.549172 at 1720582802.505498, Gyro: x: -0.0021293, y: 0.00159698, z: 0.00106465 at 1720582802.5080159, fps=427.2925835370823
Accel: x: -0.764919, y: -9.60071, z: -0.519752 at 1720582802.5103292, Gyro: x: -0.0021293, y: 0.00159698, z: 0.00106465 at 1720582802.5080159, fps=425.3426630159213
Accel: x: -0.764919, y: -9.60071, z: -0.519752 at 1720582802.5103292, Gyro: x: 0, y: 0.00159698, z: 0.000532325 at 1720582802.510448, fps=788.9962377727614
Accel: x: -0.764919, y: -9.60071, z: -0.519752 at 1720582802.5103292, Gyro: x: 0.000532325, y: 0.00159698, z: 0.000532325 at 1720582802.5128808, fps=425.2995335631718
Accel: x: -0.764919, y: -9.5811, z: -0.509946 at 1720582802.5151606, Gyro: x: 0, y: 0.00266163, z: 0.000532325 at 1720582802.5153134, fps=425.64481428861376
Accel: x: -0.764919, y: -9.5811, z: -0.509946 at 1720582802.5151606, Gyro: x: -0.0021293, y: 0.00159698, z: 0.00106465 at 1720582802.5177457, fps=425.47210387502537
Accel: x: -0.764919, y: -9.57129, z: -0.549172 at 1720582802.5199904, Gyro: x: -0.0031957, y: 0.00266163, z: 0.00159698 at 1720582802.520178, fps=425.5584415584416
Accel: x: -0.764919, y: -9.57129, z: -0.549172 at 1720582802.5199904, Gyro: x: -0.00372802, y: 0.00372802, z: 0.0021293 at 1720582802.5226102, fps=419.3046086174148
Accel: x: -0.764919, y: -9.55168, z: -0.529559 at 1720582802.5248203, Gyro: x: -0.00426035, y: 0.005325, z: 0.00106465 at 1720582802.525042, fps=425.256412856129
Accel: x: -0.764919, y: -9.55168, z: -0.529559 at 1720582802.5248203, Gyro: x: -0.00479267, y: 0.005325, z: 0.000532325 at 1720582802.5274734, fps=425.1701976685251
Accel: x: -0.784532, y: -9.57129, z: -0.549172 at 1720582802.5296485, Gyro: x: -0.00479267, y: 0.00426035, z: 0.00106465 at 1720582802.5299048, fps=425.04094041345763
Accel: x: -0.784532, y: -9.57129, z: -0.549172 at 1720582802.5296485, Gyro: x: -0.00372802, y: 0.0031957, z: 0.00159698 at 1720582802.532336, fps=414.3340906845797
Accel: x: -0.784532, y: -9.60071, z: -0.509946 at 1720582802.5344765, Gyro: x: -0.00372802, y: 0.00159698, z: 0.00266163 at 1720582802.5347672, fps=400.83180428134557
Processed depth at 1720582802.569421 and color frames at 1720582802.5694284, fps=30.645331930501367
Accel: x: -0.784532, y: -9.60071, z: -0.509946 at 1720582802.5344765, Gyro: x: -0.0021293, y: 0.00106465, z: 0.0021293 at 1720582802.5371978, fps=287.9121361889072
Accel: x: -0.745305, y: -9.57129, z: -0.529559 at 1720582802.539304, Gyro: x: -0.0021293, y: 0.00106465, z: 0.0021293 at 1720582802.5371978, fps=782.5194029850746
Accel: x: -0.745305, y: -9.57129, z: -0.529559 at 1720582802.539304, Gyro: x: -0.0021293, y: 0.0021293, z: 0.00106465 at 1720582802.5396285, fps=5607.358288770053
Accel: x: -0.745305, y: -9.57129, z: -0.529559 at 1720582802.539304, Gyro: x: -0.00159698, y: 0.00266163, z: 0 at 1720582802.5420587, fps=291.49377997081103
Accel: x: -0.745305, y: -9.5909, z: -0.529559 at 1720582802.5441294, Gyro: x: -0.00159698, y: 0.00266163, z: 0 at 1720582802.5420587, fps=788.8478465299982
Accel: x: -0.745305, y: -9.5909, z: -0.529559 at 1720582802.5441294, Gyro: x: -0.0021293, y: 0.00106465, z: -0.000532325 at 1720582802.5444887, fps=775.8608953015168
Accel: x: -0.745305, y: -9.5909, z: -0.529559 at 1720582802.5441294, Gyro: x: -0.00159698, y: -0.00106465, z: 0 at 1720582802.5469186, fps=425.38580121703853
Accel: x: -0.764919, y: -9.5909, z: -0.519752 at 1720582802.548954, Gyro: x: -0.000532325, y: -0.0021293, z: 0.0021293 at 1720582802.5493484, fps=424.78266153534537
Accel: x: -0.764919, y: -9.5909, z: -0.519752 at 1720582802.548954, Gyro: x: 0, y: -0.00159698, z: 0.0021293 at 1720582802.5517778, fps=424.5246963562753
Accel: x: -0.764919, y: -9.57129, z: -0.529559 at 1720582802.5537791, Gyro: x: -0.00106465, y: -0.000532325, z: 0.00159698 at 1720582802.5542073, fps=425.3426630159213
Accel: x: -0.764919, y: -9.57129, z: -0.529559 at 1720582802.5537791, Gyro: x: -0.00159698, y: 0, z: 0.00106465 at 1720582802.5566363, fps=424.9548125633232
Accel: x: -0.774725, y: -9.5811, z: -0.539366 at 1720582802.558602, Gyro: x: -0.00266163, y: 0.000532325, z: 0.000532325 at 1720582802.559065, fps=424.9548125633232
Accel: x: -0.774725, y: -9.5811, z: -0.539366 at 1720582802.558602, Gyro: x: -0.0031957, y: 0.000532325, z: 0.00106465 at 1720582802.5614939, fps=290.9478357380688
Accel: x: -0.764919, y: -9.55168, z: -0.519752 at 1720582802.5634253, Gyro: x: -0.0031957, y: 0.000532325, z: 0.00106465 at 1720582802.5614939, fps=778.3084060122471
Accel: x: -0.764919, y: -9.55168, z: -0.519752 at 1720582802.5634253, Gyro: x: -0.0031957, y: 0.000532325, z: 0.000532325 at 1720582802.5639224, fps=789.441746659138
Accel: x: -0.764919, y: -9.55168, z: -0.519752 at 1720582802.5634253, Gyro: x: -0.00266163, y: 0.000532325, z: 0.00106465 at 1720582802.5663507, fps=425.51526833722227
Processed depth at 1720582802.5993268 and color frames at 1720582802.5993342, fps=30.43386519805249

```

Hi, I am running a python code, I create 2 pipeline to get color/depth stream and imu stream seperately. And I print the log right after I get the data. As you can see, depth/color timestamp is ahead 50 ms of accel/gyro timestamp. 

So what should I do syncing the data? I want to use the current image data, how can I assign the correct imu data.

I read [issues/11330](https://github.com/IntelRealSense/librealsense/issues/11330), [issues/2188](https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803), [issues/4525](https://github.com/IntelRealSense/librealsense/issues/4525), [issues/3205](https://github.com/IntelRealSense/librealsense/issues/3205), not really sure about the solution.

MartyG-RealSense, 2024-07-10:
Hi @jeezrick  It is normal for there to be an offset between IMU and RGB, as advised at the https://github.com/IntelRealSense/librealsense/issues/3205#issuecomment-460878526 case that you linked to.

In regard to IMU and depth, each IMU data packet is timestamped using the depth sensor hardware clock to allow temporal synchronization between gyro, accel and depth frames, as stated in the Core Capabilities section of Intel's guide for getting IMU data.

https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/

In order for data to be to be timestamped with the depth sensor hardware clock though, you need to have support for **hardware metadata** enabled in the librealsense SDK.  Hardware metadata support will be enabled if you do one of the following:

1.  Install librealsense from Debian packages.

2.  Build librealsense from source code with CMake with the **-DFORCE_RSUSB_BACKEND=TRUE** flag included in the CMake build instruction.

3.  Build librealsense from source code with CMake and apply a **kernel patch script** to patch the kernel for hardware metadata support.

jeezrick, 2024-07-11:
Thanks, I'm aware that depth and imu has some offset, imu data suppose to be ahead of depth data about 40 ms, which is not the case when I run this python code. 

But after experiment, it turns out, you just need to run this code for 4 ~ 5 seconds before it reachs the above pattern. So, all fine. 
Also, I am interested that what does this code do? what's the effect on final data? 
```
    for (rs2::sensor sensor : sensors)
        if (sensor.supports(RS2_CAMERA_INFO_NAME)) {
            ++index;
            if (index == 1) {
                sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
                // sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_LIMIT,50000);
                sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 1); // emitter on for depth information
            }
            // std::cout << "  " << index << " : " << sensor.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
            get_sensor_option(sensor);
            if (index == 2){
                // RGB camera
                sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE,1);

                // sensor.set_option(RS2_OPTION_EXPOSURE,80.f);
            }

            if (index == 3){
                sensor.set_option(RS2_OPTION_ENABLE_MOTION_CORRECTION,0);
            }

        }
```

MartyG-RealSense, 2024-07-11:
It looks as though if the Index number is set to 1 then auto-exposure and the camera's IR Emitter component are enabled.  Auto-exposure enables the camera to automatically adjust exposure as lighting conditions in the real-world scene change, whilst the IR emitter when enabled projects an invisible pattern of dots onto the surface of objects in the scene to aid depth analysis of those surfaces.

Index 2 enables auto-exposure only.

Index 3 disables IMU motion correction, meaning that raw IMU data will be used and the SDK will not attempt to 'fix' the data to reduce inaccuracies.

closed 2024-07-24 by MartyG-RealSense