realsenseai/librealsense #13140: D455, D435i cannot be detected by realsense-viewer, but D435 can be detected

HappySamuel, 2024-07-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i & D455 } | 
| Firmware Version                   | (5.16.0.1) | 
| Operating System & Version |   {Linux (Ubuntu 22.04) / Jetpack 6.0}  | 
| Kernel Version (Linux Only)    |  (5.15.136-tegra)                                         | 
| Platform                                 | NVIDIA Jetson AGX Orin  |
| SDK Version                            |  { legacy / 2.55.1.0 }                          | 
| Language                            |  {realsense-viewer }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<D455, D435i cannot be detected by realsense-viewer, but D435 can be detected>

Hi

I have newly installed Jetpack 6.0, and realsense-viewer. But found that D455 and D435i cannot be detected at all. While D435 can be detected. Any idea why is this happening? And how to resolve this issue?

Best,
Samuel

MartyG-RealSense, 2024-07-10:
Hi @HappySamuel  When using JetPack 6, the RealSense **MIPI driver** at the link below needs to be installed in order for HID type cameras (ones with an IMU) to be supported.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

I would recommend downgrading your JetPack version to 5.1 if possible to achieve HID support without the driver.

HappySamuel, 2024-07-18:
Hi @MartyG-RealSense 

Thanks for your response. I saw a post that stated downgrading the d435i firmware version to 5.13.0.50 can work on Jetpack 6.0. What's the main difference between two version?

Currently, i am going to test downgrading the d455 firmware verison to 5.13.0.55 and see if it can work on Jetpack 6.0 or not.

Best,
Samuel.

MartyG-RealSense, 2024-07-18:
The key changes in firmware versions in-between 5.13.0.50 and 5.16.0.1 are listed in the table below, taken from the firmware release notes document.

[image]

closed 2024-07-25 by HappySamuel