realsenseai/librealsense #13141: Unable to recover through system restart when D455 deep stream is abnormal

diplomatist, 2024-07-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |    How to solve deep stream anomalies     |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Ubuntu 20.04.6 LTS focal  | 
| Kernel Version (Linux Only)    |  5.10.160                                     | 
| Platform                                 | RK3588S |
| SDK Version                            |  2.55.1                       | 
| Language                            |  c++                      | 
| Segment			| others                        | 

### Issue Description
I compiled the SDK from source, circumventing the incompatibility with the kernel version 5.10.
I connected the D455 camera to the USB 3.0 port of the RK3588s and kept the RK3588s powered on. After running the SDK every few days. This process continued for 2-3 months.

Recently, I discovered an anomaly in the camera's depth stream, with the depth map displaying extensive areas of zero depth. Despite attempting a restart of the RK3588s, the camera's depth stream failed to recover. Ultimately, I resolved the issue by physically plugging and unplugging the USB 3.0 interfaces of D455 and RK3588s.

Could you please advise on how to address this issue in scenarios where continuous power operation is necessary?

MartyG-RealSense, 2024-07-10:
Hi @diplomatist  An alternative to USB unplug-replug is to perform a **hardware reset** of the camera.  An example of C++ code for doing so is below.

```
rs2::context ctx;
rs2::device dev = ctx.query_devices().front(); // Reset the first device
dev.hardware_reset();
rs2::device_hub hub(ctx);
dev = hub.wait_for_device();
```

diplomatist, 2024-07-10:
> Hi @diplomatistUSB拔出-重新插入的替代方法是执行相机**的硬件重置**。执行此操作的 C++ 代码示例如下。
> 
> ```
> rs2::context ctx;
> rs2::device dev = ctx.query_devices().front(); // Reset the first device
> dev.hardware_reset();
> rs2::device_hub hub(ctx);
> dev = hub.wait_for_device();
> ```

Thank you for your prompt advice! Once I've replicated the issue, I will test your suggestion!

diplomatist, 2024-07-10:
> Hi @diplomatist An alternative to USB unplug-replug is to perform a **hardware reset** of the camera. An example of C++ code for doing so is below.
>
> ```
> rs2::context ctx;
> rs2::device dev = [...]

Thank you for your prompt advice! Once I've replicated the issue, I will test your suggestion!

MartyG-RealSense, 2024-07-12:
The only alternative C++ hardware reset method that I have is a script at https://github.com/IntelRealSense/librealsense/issues/9287#issuecomment-867826974 that uses callbacks to iterate through all attached cameras and reset them. 

You should ensure that the RealSense Viewer is closed before running your script, as the **Multi-Streaming Model** rules of the RealSense SDK dictate that once a particlar stream (such as depth) has been enabled in one program, a subsequently launched program cannot access that same stream until the first program releases its claim on the stream by stopping the stream or closing the program.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

MartyG-RealSense, 2024-07-18:
Hi @diplomatist  Do you require further assistance with this case, please?  Thanks!

diplomatist, 2024-07-23:
> Hi @diplomatist Do you require further assistance with this case, please? Thanks!

I have tried the advice[#9287(comment)](https://github.com/IntelRealSense/librealsense/issues/9287#issuecomment-867826974) you provided.
`
void Camera::hardwareReset(void)
{

	rs2::context ctx;
	bool reconnected = false;
	ctx.set_devices_changed_callback(
		[&reconnected](rs2::event_information& info) {
			LOG(INFO) << "camera device change triggered!";
			for (auto&& dev_new : info.get_new_devices()) {
				string new_id = dev_new.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
				LOG(INFO) << "camera device connected: " << new_id;
				reconnected = true;
			}
		});
	this->device = ctx.query_devices().front();
	this->device.hardware_reset();
	//sleep(10);
	//rs2::device_hub hub(ctx);
	//this->device = hub.wait_for_device();

	auto wait_until = std::chrono::system_clock::now() + std::chrono::seconds(30);
	while (std::chrono::system_clock::now() < wait_until)
	{
		std::this_thread::sleep_for(std::chrono::milliseconds(100));
		if (reconnected) {
			LOG(INFO) << "camera reset complete!";
			break;
		}
	}
	if (!reconnected) {
		LOG(ERROR) << "camera no connected";
	}
}
`

After trying to restart the camera multiple times, although there were no segmentation faults, there was a probability that some depth anomalies would occur after restarting the camera. This issue can be resolved by covering the camera for a short period of time.

Thank you very much for your help, it has indeed solved many problems.

closed 2024-07-23 by diplomatist

MartyG-RealSense, 2024-07-23:
That's great to hear, @diplomatist - I'm pleased that I could help.  Thanks very much for the update!