realsenseai/librealsense #13142: L515 not working

DarthBogdo, 2024-07-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.8.1  | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.8.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  v2.50.0                          | 
| Language                            |  ROS/python                          | 
| Segment			|               Robot             | 

### Issue Description
The camera doesn't display a depth image in the realsense viewer or does so shortly only to report unreachable streams afterwards and show a black screen with occasional colorful dots. After unplugging the camera, the pointer submits a value of null and the device doesn't get recognized.

I am trying to use the 5.8 kernel and SDK v2.50.0 for compatibility, but I can't get past the patched kernel modules step: 

[image]

DarthBogdo, 2024-07-10:
Hello @MartyG-RealSense
I tried this previously on other versions of the SDK - didn't bear any results. Mostly doing so would initiate the "null pointer" issue as soon as I reconnected the device.

MartyG-RealSense, 2024-07-10:
Have you tried building the 2.50.0 SDK from source code and including the flag **-DFORCE_RSUSB_BACKEND=TRUE** in the CMake build instruction, please?  An RSUSB = true build of librealsense is not dependent on kernel versions or Linux versions and does not require a kernel patch script to be applied.

DarthBogdo, 2024-07-10:
I have added RSUSB to the CMake step of the process described here: https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide

However, when I switched back, the issues with the depth stream:
[image]
[image]

and connecting:
[image]

 persisted.

MartyG-RealSense, 2024-07-11:
The instructions at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692 for an RSUSB source code build are a little clearer than the guide that you linked to.

These are the CMake build flags that I recommend to use.

`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

DarthBogdo, 2024-07-11:
I've built 2.50.0 SDK according to your instructions. Unfortunately, the depth stream issue didn't go away. The feed is visible when I start the the SDK after booting and then it becomes unreachable as soon as I turn it off or it disappears by itself after a short period of time:
[image]

MartyG-RealSense, 2024-07-11:
A quickly disappearing depth image can be caused by **GLSL** settings in the Viewer's settings interface.  https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023 has instructions on how to disable GLSL.

DarthBogdo, 2024-07-11:
After looking into it, the GLSL options were already turned off.

MartyG-RealSense, 2024-07-11:
The Viewer images that you have provided look like a hardware fault in the camera that is not repairable.  If you purchased the camera less than a year ago then I would recommend enquiring to the retailer that you purchased it from about an **RMA return**.

MartyG-RealSense, 2024-07-11:
You could run the **rs-capture** example program to confirm whether the depth display problem is limited to the RealSense Viewer tool.

DarthBogdo, 2024-07-11:
I can't properly test it because of the irregular behaviour of the camera in terms of device detection (the null pointer case I described before). Could you give me any advice for that?

MartyG-RealSense, 2024-07-12:
It is highly likely that your camera has a hardware fault on its USB port on the base of the camera that causes the irregular behaviour, unfortunately.  It is not repairable.

DarthBogdo, 2024-07-17:
The camera now displays irregular behaviour: the depth stream actually works after launching rs-capture and waiting for a couple of minutes. Albeit, the device heats up considerably. 

Thank you for your support and efforts to help me!

closed 2024-07-17 by DarthBogdo

MartyG-RealSense, 2024-07-17:
You are very welcome!

I would add that if a camera heats up considerably after only a few minutes after starting the camera then it can indicate that there is a problem with the USB port or with the USB cable (such as a cable that has internal damage from having been bent a lot).