realsenseai/librealsense #13150: Frames didn't arrived within 5 seconds. && D435i Acc FPS limited?

HowieHepburn, 2024-07-12:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|042222071449|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|041323051495|
|**Firmware Update Id**|041323051495|
|**Dfu Device Path**||

**Description:** 
**1、D435i Acc FPS limited?**
I would like to know how to set the following parameters of the IMU to produce 200+FPS output in a combined imu topic, or is it impossible? Because I noticed that the default FPS of the accelerometer is 63. So what is the highest frequency of the entire IMU output?
```
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="true"/> *modified
  <arg name="enable_accel"        default="true"/>  *modified
  <arg name="enable_sync"         default="true"/> *modified
  <arg name="unite_imu_method"          default="copy"/> *modified
```

**2、Frames didn't arrived within 5 seconds.**
When I using realsense-viewer, the stereo and RGB camera module works well, but when I try to start Motion module, it warns 
```
Timestamp: 1720750402686.394531
Severity: Warn
Description: Frames didn't arrived within 5 seconds
```
like this， and the stereo and RGB camera module turns stuck.
[image]

When I try to use the  ros driver, such as 
`roslaunch realsense2_camera rs_camera.launch`

What I modified in  rs_camera.launch is :

```
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>  
  <arg name="enable_sync"         default="true"/>
  <arg name="unite_imu_method"          default="copy"/>
```
The output:

```
PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
[... 116 lines cut ...]
[ INFO] [1720754254.732126649]: Setting Dynamic reconfig parameters.
**[ WARN] [1720754254.771833530]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.**
[ INFO] [1720754254.776552875]: Done Setting Dynamic reconfig parameters.
[ INFO] [1720754254.776919033]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[ INFO] [1720754254.777310692]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1720754254.777333495]: gyro stream is enabled - fps: 200
[ INFO] [1720754254.777348880]: accel stream is enabled - fps: 63
[ INFO] [1720754254.777363688]: setupPublishers...
[ INFO] [1720754254.779183884]: Expected frequency for depth = 15.00000
[ INFO] [1720754254.903845009]: Expected frequency for color = 15.00000
[ INFO] [1720754254.932067270]: Start publisher IMU
[ INFO] [1720754254.933136891]: setupStreams...
[ WARN] [1720754254.937234440]: 
**[ WARN] [1720754254.937278300]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.**
[ INFO] [1720754254.941175177]: SELECTED BASE:Depth, 0
[ INFO] [1720754254.951346380]: RealSense Node Is Up!
 **12/07 11:17:39,942 WARNING [139667387160320] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds**
 12/07 11:17:39,946 WARNING [139667034863360] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 12/07 11:17:44,946 WARNING [139667387160320] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 12/07 11:17:44,950 WARNING [139667034863360] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 12/07 11:17:49,950 WARNING [139667387160320] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 12/07 11:17:49,954 WARNING [139667034863360] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 12/07 11:17:54,954 WARNING [139667387160320] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 12/07 11:17:54,958 WARNING [139667034863360] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
```

MartyG-RealSense, 2024-07-12:
Hi @HowieHepburn  The RealSense ROS1 wrapper 2.3.2 is designed to work with SDK 2.50.1 and firmware 5.13.0.50.  It will also work with 2.51.1.  Some people are able to use the 2.3.2 wrapper with the current latest 2.55.1 SDK and 5.16.0.1 firmware versions but the results may be unpredictable in regards to warnings / errors.

In regard to your questions:

1.  On RealSense camera models whose minimum IMU speeds are 63 Accel and 200 Gyro, the higher supported speeds are 250 Accel and 400 Gyro.

2.  The RealSense Viewer has a known issue for some RealSense users where enabling three streams that include the Motion Module (such as Depth, RGB and Motion Module) can cause one of the streams to cease delivering frames.  There is no fix in the Viewer for this issue, so the only action that can be taken is to only enable two streams (IMU + Depth or IMU + RGB) or only enable the Motion Module on its own.  The trigger for the problem is the enabling of the Motion Module, so changing the order that the streams are enabled in makes no difference.

The RealSense ROS wrapper is not affected by this three stream issue.

If you set **initial_reset** to true then the camera will reset at launch, which can sometimes help to clear problems.

HowieHepburn, 2024-07-12:
Thank you for your prompt reply. @MartyG-RealSense 

> Hi @HowieHepburn The RealSense ROS1 wrapper 2.3.2 is designed to work with SDK 2.50.1 and firmware 5.13.0.50. It will also work with 2.51.1. Some people are able to use the 2.3.2 wrapper with the [...]

Re: Thanks for your reply, this is heplful. And I would like to know how to roll back the SDK and firmware to SDK 2.50.1 and 5.13.0.50 versions compatible with ROS1?

> On RealSense camera models whose minimum IMU speeds are 63 Accel and 200 Gyro, the higher supported speeds are 250 Accel and 400 Gyro.

Re: Since it can support such a high-frequency IMU output, how should I set rosparams to achieve such a high-frequency output?

> The RealSense Viewer has a known issue for some RealSense users where enabling three streams that include the Motion Module (such as Depth, RGB and Motion Module) can cause one of the streams to [...]

Re: That's heplful and the main problems is focused on the two above, and I need your further help. Thanks again

MartyG-RealSense, 2024-07-12:
The easiest way to change the firmware version is in the RealSense Viewer tool.  You can update it by first going to the firmware downloads page at the link below and downloading a **.bin** file for the 5.13.0.50 firmware.

https://dev.intelrealsense.com/docs/firmware-releases-d400

Then go to the 'More' option near the top of the Viewer's options side-panel and select from its drop-down menu **Update Firmware**.  A file selector box will pop up.  Navigating to the location that you downloaded the .bin file to and selecting the file will automatically begin the firmware update.

**********

My advice for removing the previous librealsense version will depend on what method you used to install it.  Can you tell me how you installed librealsense, please?

Let's try adjusting the gyro and accel FPS using the information below first to see whether you actually need to change your librealsense and firmware versions.

***********

You can set the accel and gyro speeds in ROS in the launch instruction with `accel_fps:=250 gyro_fps:=400` or you can edit the accel_fps and gyro_fps parameters in the rs_camera launch file using the code below.

```
<arg name="gyro_fps" default="400"/>
<arg name="accel_fps" default="250"/>
```

HowieHepburn, 2024-07-13:
> The easiest way to change the firmware version is in the RealSense Viewer tool. You can update it by first going to the firmware downloads page at the link below and downloading a .bin file for the [...]
Re：Thanks for your detailed method， I have rolled back the firmware to 5.13.0.50.

> My advice for removing the previous librealsense version will depend on what method you used to install it. Can you tell me how you installed librealsense, please?
>
> Let's try adjusting the gyro and [...]
Re: What I did is just like this:
```
sudo apt-get install librealsense2-dkms
sudo apt-get install librealsense2-utils
```
And I cloned this repository

> You can set the accel and gyro speeds in ROS in the launch instruction with accel_fps:=250 gyro_fps:=400 or you can edit the accel_fps and gyro_fps parameters in the rs_camera launch file using the code below.
> 
> <arg name="gyro_fps" default="400"/>
> <arg name="accel_fps" default="250"/>
I set the parameters as you said 
```
[ INFO] [1720838165.569389861]: RealSense ROS v2.3.2
[ INFO] [1720838165.569467829]: Built with LibRealSense v2.55.1
[ INFO] [1720838165.569514515]: Running with LibRealSense v2.55.1

[ INFO] [1720838165.678652841]: Device FW version: 5.13.0.50
```
), When I launch the node, I think it works! Thank you so much. 
```
[ INFO] [1720838165.718569503]: gyro stream is enabled - fps: 400
[ INFO] [1720838165.718592234]: accel stream is enabled - fps: 250
```
Furthermore,  I would like to know what values ​​these two parameters （"gyro_fps" "accel_fps"） can take? And when I set ”unite_imu_method“ to linear interpolation or copy, what is the relationship between the IMU frequency in the final merged IMU topic and these three parameters?

Finally it still have this warns:
```
[ WARN] [1720838165.643200900]: Device 042222071449 is connected using a 2.1 port. Reduced performance is expected.

[ WARN] [1720838165.714028686]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.

 13/07 10:36:10,771 WARNING [140158313678592] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
```
Is this because i use 2.1 port? Or should I change the RealSense ROS wrapper version? 
Thanks again for your hard work and detailed guidance

MartyG-RealSense, 2024-07-13:
I'm pleased to hear that you had positive progress after the latest changes.

The IMU in your D435i supports minimum and maximum values of 63 / 250 Accel and 200 / 400 Gyro.

There is a significant speed difference in the data transfer rate of USB 2.1 and 3.2.  USB2 is 480 Mbps (megabits per second), whilst USB3 is 5000 Mbps.

In regard to **unite_imu_method**, the ROS wrapper documentation provides the following explanations of the differences between the copy and linear interpolation modes.

copy: Every gyro message will be attached by the last accel message.  This creates a united message for each incoming gyro or accel message.  In copy mode, if gyro was set to 200 and accel was set to 250 then the output rate will be approximately 200+250 = 450 Hz.

linear_interpolation: Every gyro message will be attached by an accel message which is interpolated to gyro's timestamp.

In regard to your librealsense installation method, because you installed from packages, if you wanted to uninstall librealsense then you could remove all packages with 'realsense' in their name from your computer with the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

HowieHepburn, 2024-07-13:
> I'm pleased to hear that you had positive progress after the latest changes.

Re: Thanks so much for your help.

> The IMU in your D435i supports minimum and maximum values of 63 / 250 Accel and 200 / 400 Gyro.

Re: So the two parameters can only take the minimum or maximum value, thanks for your detailed reply.
```
[ WARN] [1720840291.459360669]: No mathcing profile found for accel with fps=200
[ WARN] [1720840291.459388151]: Using default profile instead
```
> There is a significant speed difference in the data transfer rate of USB 2.1 and 3.2. USB2 is 480 Mbps (megabits per second), whilst USB3 is 5000 Mbps.

Re: Ok, I am not sure will this cause any problems for me. I will try to purchase a 3.2. USB2 for it.

> In regard to **unite_imu_method**, the ROS wrapper documentation provides the following explanations of the differences between the copy and linear interpolation modes.
> copy: Every gyro message will be attached by the last accel message. This creates a united message for each incoming gyro or accel message.
> linear_interpolation: Every gyro message will be attached by an accel message which is interpolated to gyro's timestamp.
> Additional information about copy mode is that if gyro was set to 200 and accel was set to 250 then the output rate will be approximately 200+250 = 450 Hz.

Re: Thanks so much, this is helpful, I understand the situation when I set unite_imu_method to copy. So if I set unite_imu_method to linear interpolation and fyro_fps to 200, does it mean that no matter what accel_fps is set to, the final IMU joint output will also be 200?

> In regard to your librealsense installation method, because you installed from packages, if you wanted to uninstall librealsense then you could remove all packages with 'realsense' in their name [...]

Re: Okay, I will try to uninstall librealsense. But Which version should I install? No SDK version matchs 2.50.1, I am a little confused.
[image]
Thanks again.

MartyG-RealSense, 2024-07-13:
It is possible to install a specific version of the packages by using the **apt-cache showpkg librealsense2** command to list the filenames of specific versions of the librealsense2-dkms and librealsense2-utils packages.  Then do sudo apt-get install with those files.

HowieHepburn, 2024-07-13:
Re：Okay, thanks a lot, and I want to confirm which ros1 wrapper version is the best? The version 2.50.1 you mentioned before is not exsit.

```
howie@howiePC-Legion:~$ apt-cache showpkg librealsense2 | grep 2.5
2.55.1-0~realsense.12473 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages) (/var/lib/dpkg/status)
2.55.1-0~realsense.12428 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages)
2.55.1-0~realsense.12425 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages)
2.55.1-0~realsense.12422 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages)
2.54.2-0~realsense.10772 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages)
2.54.1-0~realsense.9590 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages)
2.54.1-0~realsense.9587 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages)
2.53.1-0~realsense0.8250 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages)
2.51.1-0~realsense0.7526 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages)
2.51.1-0~realsense0.7520 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages)
2.50.0-0~realsense0.6128 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-amd64_Packages)
  librealsense2-utils,librealsense2 2.55.1-0~realsense.12473
  librealsense2-gl,librealsense2 2.55.1-0~realsense.12473
  librealsense2-dbg,librealsense2 2.55.1-0~realsense.12473
  librealsense2-dev,librealsense2 2.55.1-0~realsense.12473
  librealsense2-gl,librealsense2 2.55.1-0~realsense.12428
  librealsense2-dbg,librealsense2 2.55.1-0~realsense.12428
  librealsense2-dev,librealsense2 2.55.1-0~realsense.12428
  librealsense2-utils,librealsense2 2.55.1-0~realsense.12428
  librealsense2-utils,librealsense2 2.55.1-0~realsense.12425
  librealsense2-dev,librealsense2 2.55.1-0~realsense.12425
  librealsense2-gl,librealsense2 2.55.1-0~realsense.12425
  librealsense2-dbg,librealsense2 2.55.1-0~realsense.12425
  librealsense2-dbg,librealsense2 2.55.1-0~realsense.12422
  librealsense2-dev,librealsense2 2.55.1-0~realsense.12422
  librealsense2-gl,librealsense2 2.55.1-0~realsense.12422
  librealsense2-utils,librealsense2 2.55.1-0~realsense.12422
  librealsense2-dev,librealsense2 2.54.2-0~realsense.10772
  librealsense2-dbg,librealsense2 2.54.2-0~realsense.10772
  librealsense2-utils,librealsense2 2.54.2-0~realsense.10772
  librealsense2-gl,librealsense2 2.54.2-0~realsense.10772
  librealsense2-dbg,librealsense2 2.54.1-0~realsense.9590
  librealsense2-gl,librealsense2 2.54.1-0~realsense.9590
  librealsense2-dev,librealsense2 2.54.1-0~realsense.9590
  librealsense2-utils,librealsense2 2.54.1-0~realsense.9590
  librealsense2-gl,librealsense2 2.54.1-0~realsense.9587
  librealsense2-dbg,librealsense2 2.54.1-0~realsense.9587
  librealsense2-dev,librealsense2 2.54.1-0~realsense.9587
  librealsense2-utils,librealsense2 2.54.1-0~realsense.9587
  librealsense2-dbg,librealsense2 2.53.1-0~realsense0.8250
  librealsense2-utils,librealsense2 2.53.1-0~realsense0.8250
  librealsense2-utils,librealsense2 2.51.1-0~realsense0.7526
  librealsense2-utils,librealsense2 2.51.1-0~realsense0.7520
  librealsense2-utils,librealsense2 2.50.0-0~realsense0.6128
  librealsense2-net,librealsense2 2.53.1-0~realsense0.8250
  librealsense2-net,librealsense2 2.51.1-0~realsense0.7526
  librealsense2-net,librealsense2 2.51.1-0~realsense0.7520
  librealsense2-net,librealsense2 2.50.0-0~realsense0.6128
  librealsense2-gl,librealsense2 2.53.1-0~realsense0.8250
  librealsense2-gl,librealsense2 2.51.1-0~realsense0.7526
  librealsense2-gl,librealsense2 2.51.1-0~realsense0.7520
  librealsense2-gl,librealsense2 2.50.0-0~realsense0.6128
  librealsense2-dev,librealsense2 2.53.1-0~realsense0.8250
  librealsense2-dev,librealsense2 2.51.1-0~realsense0.7526
  librealsense2-dev,librealsense2 2.51.1-0~realsense0.7520
  librealsense2-dev,librealsense2 2.50.0-0~realsense0.6128
  librealsense2-dbg,librealsense2 2.51.1-0~realsense0.7526
  librealsense2-dbg,librealsense2 2.51.1-0~realsense0.7520
  librealsense2-dbg,librealsense2 2.50.0-0~realsense0.6128
2.55.1-0~realsense.12473 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.17) libgcc-s1 (2 3.0) libstdc++6 (2 9) librealsense2-udev-rules (0 (null)) 
2.55.1-0~realsense.12428 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.17) libgcc-s1 (2 3.0) libstdc++6 (2 9) librealsense2-udev-rules (0 (null)) 
2.55.1-0~realsense.12425 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.17) libgcc-s1 (2 3.0) libstdc++6 (2 9) librealsense2-udev-rules (0 (null)) 
2.55.1-0~realsense.12422 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.17) libgcc-s1 (2 3.0) libstdc++6 (2 9) librealsense2-udev-rules (0 (null)) 
2.54.2-0~realsense.10772 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.29) libgcc-s1 (2 3.0) libstdc++6 (2 9) librealsense2-udev-rules (0 (null)) 
2.54.1-0~realsense.9590 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.29) libgcc-s1 (2 3.0) libstdc++6 (2 9) librealsense2-udev-rules (0 (null)) 
[... cut ...]
[ WARN] [1720875493.863363169]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 848, Height: 480, FPS: 30, Format: RGB8
 WARN] [1720875494.867586808]: Hardware Notification:Motion Module failure,1.72088e+12,Error,Hardware Error
 13/07 20:58:18,948 WARNING [139848315758336] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 13/07 20:58:18,951 WARNING [139848304891648] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
```

MartyG-RealSense, 2024-07-13:
If you want the Infra1 and Infra2 streams then you should leave `enable_infra` as false.  Also, your USB connection type was listed in the log as 2.1, and the Infra2 topic is only available on a USB3 connection.

The best ROS1 wrapper version is 2.3.2, as development of the ROS1 wrapper has ceased (the ROS2 one continues) and there were no more ROS1 wrapper versions after 2.3.2.

What happens if you try the package installation command below to install librealsense 2.50.0?

`sudo apt install librealsense2=2.50.0-0~realsense0.6128`

HowieHepburn, 2024-07-13:
I use this to install librealsense2:
`sudo apt install librealsense2-dev=2.50.0-0~realsense0.6128`
The key information in the output and the modified parameters when launch rs_camera:
```
  <arg name="enable_gyro"         default="true"/>
 <arg name="enable_accel"        default="true"/>
 <arg name="unite_imu_method"          default="linear_interpolation"/> 
 <arg name="initial_reset"             default="true"/>
```

```
[ INFO] [1720882487.526661556]: Initializing nodelet with 8 worker threads.
[ INFO] [1720882487.856429986]: RealSense ROS v2.3.2
[ INFO] [1720882487.856463917]: Built with LibRealSense v2.50.0
[ INFO] [1720882487.856480807]: Running with LibRealSense v2.50.0

[ WARN] [1720882487.894779060]: Device 042222071449 is connected using a 2.1 port. Reduced performance is expected

[ WARN] [1720882494.464276623]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
```

MartyG-RealSense, 2024-07-13:
A lot of people get the power_line_frequency warning.  You can safely ignore that.

The ROS1 wrapper can still work with a USB 2.1 connection.

MartyG-RealSense, 2024-07-21:
Hi @HowieHepburn  Do you require further assistance with this case, please?  Thanks!

HowieHepburn, 2024-07-22:
> Hi @HowieHepburn Do you require further assistance with this case, please? Thanks!
Sorry for the late response. I purchase a USB3.2 wire in these days. And work on sync the time between my sensors.
And I have figured out the relationship between the imu parameters and output FPS.
Now the output of the camera is :
```
[ INFO] [1721621098.015974100]: RealSense ROS v2.3.2
[ INFO] [1721621098.015997632]: Built with LibRealSense v2.50.0
[ INFO] [1721621098.016006859]: Running with LibRealSense v2.50.0
[ INFO] [1721621098.035622321]:  
[ INFO] [1721621098.278269216]: Device with serial number 042222071449 was found.

[ INFO] [1721621098.278317004]: Device with physical ID 2-1-2 was found.
[ INFO] [1721621098.278329076]: Device with name Intel RealSense D435I was found.
[ INFO] [1721621098.278655498]: Device with port number 2-1 was found.
[ INFO] [1721621098.278673181]: Device USB type: 3.2
[ INFO] [1721621098.280643526]: getParameters...
[ INFO] [1721621098.298713053]: setupDevice...
[ INFO] [1721621098.298727565]: JSON file is not provided
[ INFO] [1721621098.298734670]: ROS Node Namespace: camera
[ INFO] [1721621098.298744245]: Device Name: Intel RealSense D435I
[ INFO] [1721621098.298752609]: Device Serial No: 042222071449
[ INFO] [1721621098.298761578]: Device physical port: 2-1-2
[ INFO] [1721621098.298768802]: Device FW version: 05.13.00.50
[ INFO] [1721621098.298777411]: Device Product ID: 0x0B3A
[ INFO] [1721621098.298786255]: Enable PointCloud: Off
[ INFO] [1721621098.298794419]: Align Depth: Off
[ INFO] [1721621098.298801294]: Sync Mode: On
[ INFO] [1721621098.298830253]: Device Sensors: 
[ INFO] [1721621098.357137383]: Stereo Module was found.
[ INFO] [1721621098.366373999]: RGB Camera was found.
[ INFO] [1721621098.366484910]: Motion Module was found.
[ INFO] [1721621098.366505085]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1721621098.366523124]: num_filters: 0
[ INFO] [1721621098.366535289]: Setting Dynamic reconfig parameters.
[ INFO] [1721621100.716509961]: Done Setting Dynamic reconfig parameters.
[ INFO] [1721621100.716688420]: gyro stream is enabled - fps: 200
[ INFO] [1721621100.716716495]: accel stream is enabled - fps: 250
[ INFO] [1721621100.716741493]: setupPublishers...
[ INFO] [1721621100.717866429]: Start publisher IMU
[ INFO] [1721621100.718335085]: setupStreams...
[ INFO] [1721621101.041665108]: SELECTED BASE:Depth, 0
[ WARN] [1721621101.042132851]: 
[ WARN] [1721621101.042160704]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[ INFO] [1721621101.044252969]: RealSense Node Is Up!
 22/07 12:05:01,496 WARNING [140032903997184] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```
And I found that the timestamp is not we expected, I am not sure is this related to the warning above? ( frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically. && control_transfer returned error )
So how can I synchronised the camera timestamp?

`
System time:
r1436@r1436-miniPC:~$ sudo phc_ctl /dev/ptp0 get
phc_ctl[6615.957]: clock time is 1721621633.318417467 or Mon Jul 22 12:13:53 2024
`
`Lidar time:
r1436@r1436-miniPC:~$ rostopic echo /hesai/pandar | grep sec
    secs: 1721621635
    nsecs: 261912000
    
r1436@r1436-miniPC:~$ rostopic echo /camera/imu | grep sec
    secs: 1721621576
    nsecs: 680224895
`

MartyG-RealSense, 2024-07-22:
Having the time domain as HARDWARE_CLOCK may not necessarily affect the data negatively.  If you are concerned though then you could try adding **enable_sync:=true** to your roslaunch instruction.  When true this parameter gathers the closest frames of different sensors, such as infrared, color and depth, to be sent with the same timetag.