realsenseai/librealsense #13152: Issue with Depth Mismatch in D405 Camera

qidihan, 2024-07-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D405 } | 
| Firmware Version                   | The newest | 
| Operating System & Version |   {Linux (Ubuntu 22)  | 
| Kernel Version (Linux Only)    |  The newest                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy}                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
Hello,

I am experiencing frequent depth mismatches with my D405 camera. There are often areas where the depth data is not accurate. Sometimes the camera works normally, but other times, it fails to provide correct depth information. I have tried adding a ring light source, which seems to help occasionally, but the improvement is not consistent.

Attached is a screenshot showing the issue. Any insights or suggestions on how to resolve this would be greatly appreciated.

Thank you!

Below is the examples
[image]
[image]
[image]

MartyG-RealSense, 2024-07-13:
Hi @qidihan  It looks as though the room that the camera is being used in has ceiling strip lights.  These could be disruptive to the camera image depending on the power frequency that the lights are operating at.

[image]

As the D405 does not have RGB controls because the RGB image comes from the depth sensor instead of a dedicated RGB sensor, you could instead try setting the FPS speed of depth and color to 60 in the RealSense Viewer tool to see whether it improves the depth image.  If there is interference from overhead lights then the nearer that the FPS is to the 'Hz' operating frequency of the lights, the lower the interference should be.

qidihan, 2024-07-13:
I try FPS with 30 60 90, I got all same results. 
[image]
[image]

In addition to the overhead lighting, I also placed a ring light around the camera. The result did't change too much.
[image]

MartyG-RealSense, 2024-07-13:
In the RealSense Viewer, what does the depth image look like if you change the Preset near the top of the Viewer's side-panel from the default 'Custom' to 'High Accuracy'

closed 2024-07-30 by MartyG-RealSense