realsenseai/librealsense #13161: Why is different between hardware size and display size of depth frames in RealsenseViewer?

Guan9Wu, 2024-07-15:
[image]

I use RealsenseViewer to visualize my `.bag` record file, but the hardware size and display size of depth frames is different, that makes me  can't get the correct depth frame by the capture function. I definately believe my recorded depth frame size is 1280*720. How to fix it?

MartyG-RealSense, 2024-07-16:
Hi @Guan9Wu  This halving of the resolution occurs because a post-processing filter called **Decimation Filter** is enabled by default in the 'Stereo Module > Post-Processing Filters' section of the Viewer's options side-panel.

If you expand open the list of filters in the side-panel and click on the blue icon beside Decimation Filter to turn the icon red (disabled) then the full 1280x720 resolution will be used for the depth frame.  

It is also worth mentioning that post-processed data is not saved to a bag file, only the raw stream data.  The resolution of the depth data in your bag is 1280x720, but when you load the bag into the Viewer, the Decimation filter will by default halve ('downsample') the resolution in real-time unless the filter is disabled.

Guan9Wu, 2024-07-22:
Thans for the information. It clarifies things for me now.

closed 2024-07-22 by Guan9Wu

MartyG-RealSense, 2024-07-22:
You are very welcome.  I'm pleased that I could help!