realsenseai/librealsense #13162: Device not detected

bingogome, 2024-07-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i, D435if, D405 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu22.04} | 
| Kernel Version (Linux Only)    |  (6.5.0-41-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.55.1 }                          | 

### Issue Description
My D435i, D435if, D405 cameras can't be detected on Ubuntu. 

realsense-viewer: 

 15/07 17:57:02,402 INFO [124285043915968] (backend-v4l2.cpp:646) Cannot access /sys/class/video4linux
 15/07 17:57:02,624 INFO [124285043915968] (backend-v4l2.cpp:646) Cannot access /sys/class/video4linux
 15/07 17:57:02,624 INFO [124285043915968] (context.cpp:128) Found 0 RealSense devices (0xff requested & 0xff from device-mask in settings)

rs-enumerate-devices:

No device detected. Is it plugged in?

Tried to reinstall the SDK from [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md). Also unplugged and replugged. Also made sure the USB ports are 3.0 and above.

lsusb does detect it:
Bus 002 Device 003: ID 8086:0b5b Intel Corp. Intel(R) RealSense(TM) Depth Camera 405

MartyG-RealSense, 2024-07-16:
Hi @bingogome  What is the make / model of the PC that you are using the cameras on, please?

bingogome, 2024-07-16:
Thanks for your message @MartyG-RealSense , the model of the PC is [this](https://psref.lenovo.com/syspool/Sys/PDF/Legion/Legion_Pro_5_16IRX8/Legion_Pro_5_16IRX8_Spec.pdf). It is a Legion Pro 5 16IRX8.

MartyG-RealSense, 2024-07-16:
What method are you using to run Ubuntu 22.04 on your Windows laptop, please?

For example, if you are using Microsoft's **Windows Subsystem for Linux 2 (WSL2)** to install your Ubuntu environment in a virtual machine (VM) then RealSense cameras usually have a lot more problems with WSL2 than with an Ubuntu 'dual boot' setup where Windows and Ubuntu are installed on the same computer and you select which operating system to use when starting up the computer.

bingogome, 2024-07-16:
@MartyG-RealSense  The Ubuntu is a dual boot on the machine (booted from GRUB option). I am not using WSL. 

Just to confirm, should my kernel version 6.5.0-41-generic be fine?

MartyG-RealSense, 2024-07-16:
If you are using librealsense 2.55.1 then kernel 6.5 is supported.  In situations where there might be a conflict between librealsense and the kernel though, it is recommended to build librealsense from source code with CMake instead of using packages and include the flag **-DFORCE_RSUSB_BACKEND=TRUE** in the CMake build instruction so that librealsense bypasses the kernel.

An easy way to test whether a kernel bypass would help would be:

1.  Remove all packages from your computer containing the word 'realsense' by inputting the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

2.  Run the **./libuvc_installation.sh** script to perform an automated source code build of librealsense that bypasses the kernel.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

3.  Test whether the cameras can be detected in this libuvc build of librealsense.

********

First though before doing that, you could try running RealSense programs in sudo admin permissions mode with commands such as **sudo realsense-viewer**

bingogome, 2024-07-16:
The sudo did not help, but the script helped! The installation is working. Thank you for your help.

Are there "clean up" steps for this? Would keeping using this installation from the script fine? Or, I should have a new installation using the DFORCE_RSUSB_BACKEND flag.

Thanks again

MartyG-RealSense, 2024-07-17:
You are very welcome.  I'm pleased to hear that you achieved a working installation.

The **libuvc backend** installation method is the predecessor of RSUSB Backend and both installation methods work the same.  So you are not likely to need to perform an RSUSB installation. 

If you do need to uninstall then go to the 'build' directory of the librealsense source code directory that was created by the libuvc script at **librealsense > build** and use the CMake command below.

`sudo make uninstall && make clean`

closed 2024-07-17 by bingogome

MartyG-RealSense, 2024-07-18:
You are very welcome.  :)