realsenseai/librealsense #13164: 树莓派5连接intel realsense D455,使用realsense-viewer可以打开摄像头，但是运行代码时显示module 'pyrealsense2' has no attribute 'pipeline'

ddl13140, 2024-07-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ddl13140, 2024-07-17:
麻烦各位大神帮忙解决一下

MartyG-RealSense, 2024-07-17:
Hello @ddl13140  If you are using Linux then a solution for the `module 'pyrealsense2' has no attribute 'pipeline'` error is to copy two .so files - **librealsense2.so** and **pyrealsense2.so** - into the same folder that your pyrealsense2 project script is in.  

If you have built the pyrealsense2 wrapper from source code then you should be able to find the pyrealsense2.so file in the librealsense source code folder, at the location **librealsense > build >wrappers > Python**

When using Python 3, the pyrealsense2.so file will have a long automatically generated filename such as pyrealsense2.cpython-35m-arm-linux-gnueabihf.so

*********

如果您使用的是 Linux，那么解决模块“pyrealsense2”没有属性“pipeline”错误的方法是将两个 .so 文件（librealsense2.so 和 pyrealsense2.so）复制到 pyrealsense2 项目脚本所在的同一文件夹中。

如果您已从源代码构建了 pyrealsense2 包装器，那么您应该能够在 librealsense 源代码文件夹中找到 pyrealsense2.so 文件，位置为 librealsense > build >wrappers > Python

使用 Python 3 时，pyrealsense2.so 文件将具有一个较长的自动生成的文件名，例如 pyrealsense2.cpython-35m-arm-linux-gnueabihf.so

ddl13140, 2024-07-17:
> 你好@ddl13140 如果您使用的是Linux，那么解决该`module 'pyrealsense2' has no attribute 'pipeline'`错误的方法只需两个.so文件 - **librealsense2.so**和**pyrealsense2.so** - 复制到pyrealsense2项目脚本具有相同的文件夹中。
> 
> 如果您已经从源代码构建了pyrealsense2包装器，那么您应该能够在librealsense源代码文件夹中找到pyrealsense2.so文件，位置为**librealsense > build >wrappers > Python**
> 
> 当使用Python 3时，pyrealsense2.so文件将有一个大量的自动生成的文件名，例如pyrealsense2.cpython-35m-arm-linux-gnueabihf.so
> 
> 如果您使用的是Linux，那么解决模块“pyrealsense2”属性没有“pipeline”错误的方法只需两个.so文件（librealsense2.so和pyrealsense2.so）复制到pyrealsense2项目脚本有相同的文件夹中。
> 
> 如果您已从源代码构建了pyrealsense2包装器，那么您应该能够在librealsense源代码文件夹中找到pyrealsense2.so文件，位置为librealsense > build >wrappers > Python
> 
> 使用Python 3时，pyrealsense2.so文件将具有大量自动生成的文件名，例如pyrealsense2.cpython-35m-arm-linux-gnueabihf.so

您好，把相应得文件加进去了，还是不行

MartyG-RealSense, 2024-07-17:
What happens if you input the three lines below before running your script?

Change **python3.6** to the version number of Python 3 that you are using, such as python3.11

```
export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH: /usr/local/lib
export PYTHONPATH=$PYTHONPATH: /usr/local/lib/python3.6/pyrealsense2
```

**********

如果在运行脚本之前输入以下三行，会发生什么情况？

将 python3.6 更改为您正在使用的 Python 3 的版本号，例如 python3.11

```
export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH: /usr/local/lib
export PYTHONPATH=$PYTHONPATH: /usr/local/lib/python3.6/pyrealsense2
```

MartyG-RealSense, 2024-07-30:
Do you require further assistance with this case, please?  Thanks!

您需要进一步协助处理此案吗？谢谢！

MartyG-RealSense, 2024-08-09:
Case closed due to no further comments received.

由于没有收到进一步的评论，案件已结。

closed 2024-08-09 by MartyG-RealSense