realsenseai/librealsense #13167: Firmware upgrade of 2 D457 not working properly

tonynajjar, 2024-07-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 22)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson Orin  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Upgrading the firmware with two D457 attached to a Jetson Orin is giving unpredictable results and I'm not sure if it's just the reporting of  `rs-enumerate-devices -s` that is wrong or if it's actually wrong. The issues are:

1- upgrading one realsense results in the upgrade of the other
2- the second realsense is never reported to be upgraded.

Can you reproduce?

```
user@jetson:~$ rs-enumerate-devices -s
Device Name                   Serial Number       Firmware Version    
Intel RealSense D457          243122301179        5.13.1.53           
Intel RealSense D457          243122300693        5.13.1.53           
user@jetson:~$ cd /tmp/
user@jetson:/tmp$ rs-fw-update -f Signed_Image_UVC_5_16_0_1.bin -s 243122301179

Search for device with serial number: 243122301179

Updating device: 
Name: Intel RealSense D457, serial number: 243122301179, update serial number: 252143110086, firmware version: 5.13.1.53, USB type: unknown

Update can take up to 2 minutes
    
Firmware update done
user@jetson:/tmp$ rs-enumerate-devices -s
Device Name                   Serial Number       Firmware Version    
Intel RealSense D457          243122301179        5.13.1.53           
Intel RealSense D457          243122300693        5.16.0.1  

user@jetson:/tmp$ rs-fw-update -f Signed_Image_UVC_5_16_0_1.bin -s 243122300693

Search for device with serial number: 243122300693

Updating device: 
Name: Intel RealSense D457, serial number: 243122300693, update serial number: 252143110082, firmware version: 5.16.0.1, USB type: unknown

Update can take up to 2 minutes
    
Firmware update done
user@jetson:/tmp$ rs-enumerate-devices -s
Device Name                   Serial Number       Firmware Version    
Intel RealSense D457          243122301179        5.13.1.53           
Intel RealSense D457          243122300693        5.16.0.1
```

MartyG-RealSense, 2024-07-17:
Hi @tonynajjar  The librealsense SDK's release notes state that the rs-fw-update tool supports the update of multiple MIPI devices as of librealsense 2.55.1.  However, it adds that the udev rules and driver v1.0.1.21 (at the link below) should be reinstalled.

https://github.com/IntelRealSense/realsense_mipi_platform_driver/releases/tag/v1.0.1.21

[image]

tonynajjar, 2024-07-17:
Thanks for the answer. I don't use the `realsense_mipi_platform_driver` directly because I have this: https://connecttech.com/product/intel-realsense-rudi-nx-embedded-system-with-nvidia-jetson-xavier-nx/
(I don't know if internally they use that driver)

> Please reinstall udev rules and [driver](https://github.com/IntelRealSense/realsense_mipi_platform_driver/releases/tag/v1.0.1.21) (v1.0.1.21)

but anyway just to understand this statement; when is this supposed to be done? after upgrading to 2.55.1?

> If you have two cameras then it may be easier to update the firmware with only one D457 at a time attached to the computer.

I guess this is an option but it's not very convenient for me as a I have 5 times this setup in remote areas that I can't reach to disconnect the cameras one by one -> is there a way to "disconnect them via software" for the sake of upgrading?

MartyG-RealSense, 2024-07-17:
As support for firmware update of multiple MIPI devices with rs-fw-update was added in 2.55.1 then yes, update librealsense to 2.55.1 first to use the 2.55.1 version of rs-fw-update.

Connect Tech provide their own drivers called BSP (Board Support Package).  The list of drivers on their support page is very long, but you can quickly find the section listing their D457 product drivers by performing a browser 'find' search on the page for the word 'd457'.

https://connecttech.com/supported-cameras/

Intel do not provide technical support for products supplied by non-Intel companies that use RealSense technology because support for those products is the responsibility of their manufacturer.  I would recommend contacting Connect Tech's support to enquire about this subject.

https://connecttech.com/support/

tonynajjar, 2024-07-17:
Got it, do you have however any tip regarding:

> is there a way to "disconnect them via software" for the sake of upgrading?

tonynajjar, 2024-07-19:
Okay thank you I will close this ticket until I try it and report back if that doesn't work