realsenseai/librealsense #13169: Depth stream "No frame received" on intel realsense viewer in PC

0417wang, 2024-07-18:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0264203|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003a60bc0c0|
|**Firmware Update Id**|0003a60bc0c0|

Please provide a description of the problem

0417wang, 2024-07-18:
[image]

MartyG-RealSense, 2024-07-18:
Hi @0417wang  I note that you have a **Fatal error occurred (16)** message in the log at the bottom of the RealSene Viewer window.  This unfortunately means that there is a fault in the camera hardware that cannot be repaired.  

If you purchased the camera less than a year ago then please contact the retailer that you purchased the L515 from to enquire about an **RMA return** of the camera.

closed 2025-01-06 by MartyG-RealSense