realsenseai/librealsense #13171: failed to set power state, using `for (rs2::device dev : list)`

bingogome, 2024-07-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |  Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  6.5.0-44-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            | 2.55.1.0                          | 
| Language                            |  C                          | 

### Issue Description

[PC model](https://psref.lenovo.com/syspool/Sys/PDF/Legion/Legion_Pro_5_16IRX8/Legion_Pro_5_16IRX8_Spec.pdf)

failed to set power state

Error happens when iterating device list using the following and the iterator reaches a device that is in use:
`
rs2::device_list list = ctx_.query_devices();
for (rs2::device dev : list)
{
...
}
`
Had to use a variable to track device in use and iterate by index.

MartyG-RealSense, 2024-07-19:
Hi @bingogome  The links below have a couple of examples of using rs2::device_list in C language.

**rs-depth**
https://github.com/IntelRealSense/librealsense/blob/master/examples/C/depth/rs-depth.c#L81-L83

**rs-color**
https://github.com/IntelRealSense/librealsense/blob/master/examples/C/color/rs-color.c#L36-L38

Please note that in both examples there is an asterisk - * - on the end of rs2_device_list.

`rs2_device_list*`

MartyG-RealSense, 2024-07-30:
Hi @bingogome  Do you require further assistance with this case, please?  Thanks!

bingogome, 2024-07-31:
We solved the problem with a USB-C hub. Any idea that might have been the solution? But no, no further assistance is needed for now! Please feel free to close the issue. Thanks!

MartyG-RealSense, 2024-07-31:
A hub uses its own **USB controller** for its ports that is separate from the USB controller on a computer's built-in USB ports.  So one brand of USB controller chosen by the hub's manufacturer might work better with RealSense cameras than another brand of controller used by the computer's manufacturer.

As you suggested, I will close this issue.  Thanks very much for the update!

closed 2024-07-31 by MartyG-RealSense