realsenseai/librealsense #13172: Can't update firmware on D457 on NVIDIA Jetson Orin

micah-cobot, 2024-07-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457 | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Jetpack 5.1.1 / R35 (release), REVISION: 3.1   | 
| Kernel Version (Linux Only)    |  Jetpack 5.1.1 / R35 (release), REVISION: 3.1                                         | 
| Platform                                 | NVIDIA Jetson Orin / [ANVIL Orin](https://connecttech.com/product/anvil-embedded-system-with-nvidia-jetson-agx-orin/)  |
| SDK Version                            |  2.55.1                           | 
| Language                            |  N/A                          | 
| Segment			|  Robot                          | 

### Issue Description
Cannot update firmware on D457 using `rs-fw-update` or the RealsenseViewer GUI. Command run:

```bash
rs-fw-update -s 234322303524 -f /home/nvidia/Downloads/Signed_Image_UVC_5_16_0_1__3/Signed_Image_UVC_5_16_0_1/Signed_Image_UVC_5_16_0_1.bin
```

The output is:
```
Search for device with serial number: 234322303524

Updating device FW: 
[GMSL] Intel RealSense D457 s/n 234322303524, update serial number: 238143110464, firmware version: 5.15.0.2

Update can take up to 2 minutes

Firmware Update failed - wrong path or permissions missing
```

I have verified that `/home/nvidia/Downloads/Signed_Image_UVC_5_16_0_1__3/Signed_Image_UVC_5_16_0_1/Signed_Image_UVC_5_16_0_1.bin` is a valid and correct file path to the firmware which I downloaded from the official releases: https://dev.intelrealsense.com/docs/firmware-releases-d400. I have tried using both the command line `rs-fw-update` tool and the "Update firmware" option in the RealSense Viewer GUI, both of which produce the above output

### Hardware Info
I have the camera connected via GMSL to an Anvil Embeded System: https://connecttech.com/product/anvil-embedded-system-with-nvidia-jetson-agx-orin/ running Jetpack 5.1.1; this computer has a board for GMSL, and the D457 is supported as per this table from ConnectTech: https://connecttech.com/supported-cameras/:

[image]

The output of `cat /etc/nv_tegra_release` is :
```
# R35 (release), REVISION: 3.1, GCID: 32827747, BOARD: t186ref, EABI: aarch64, DATE: Sun Mar 19 15:19:21 UTC 2023
```

I installed librealsense via Debian package as per this tutorial: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md. The output of `dpkg -l | grep librealsense` is:

```
ii  librealsense2:arm64                        2.55.1-0~realsense.3336                          arm64        Intel(R) RealSense(tm) Cross Platform API - runtime
ii  librealsense2-dev:arm64                    2.55.1-0~realsense.3336                          arm64        Intel(R) RealSense(tm) Camera Capture API - development files
ii  librealsense2-gl:arm64                     2.55.1-0~realsense.3336                          arm64        Intel(R) RealSense(tm) - GLSL-enabled extensions
ii  librealsense2-udev-rules:arm64             2.55.1-0~realsense.3336                          arm64        Intel(R) RealSense(tm) Camera Capture API - udev rules
ii  librealsense2-utils:arm64                  2.55.1-0~realsense.3336                          arm64        Intel(R) RealSense(tm) Camera Capture API - utils and demos
```

MartyG-RealSense, 2024-07-19:
Hi @micah-cobot  Please try moving your .bin file to the same folder as rs-fw-update so that you do not need to provide a path, just the .bin filename.  rs-fw-update should be in the **usr/local/bin** folder.

`rs-fw-update -s 234322303524 -f Signed_Image_UVC_5_16_0_1.bin`

micah-cobot, 2024-07-19:
Hello @MartyG-RealSense ,

I have tried doing this (my `rs-fw-update` is located in /usr/bin) and receive the same output:

```
rs-fw-update -s 234322303524 -f Signed_Image_UVC_5_16_0_1.bin
```

```
Search for device with serial number: 234322303524

Updating device FW: 
[GMSL] Intel RealSense D457 s/n 234322303524, update serial number: 238143110464, firmware version: 5.15.0.2

Update can take up to 2 minutes

Firmware Update failed - wrong path or permissions missing
```

I even tried changing the permissions of the `.bin` file with `chmod 777 Signed_Image_UVC_5_16_0_1.bin`, same result, and also tried first going into root mode with `sudo su` before executing the rs-fw-update command, same result.

micah-cobot, 2024-07-19:
Hello @MartyG-RealSense ,

I've tracked down the issue to this line returning an empty string:

```
dev.get_info(RS2_CAMERA_INFO_DFU_DEVICE_PATH)
```

which is located here: https://github.com/IntelRealSense/librealsense/blob/master/tools/fw-update/rs-fw-update.cpp#L167

any ideas on what's going wrong / how to fix?

MartyG-RealSense, 2024-07-20:
This error message has not been previously reported.

Would it be possible for you to temporarily switch your D457 from GMSL mode to USB mode using the physical mode switch inside the camera's casing, perform a firmware update on a USB cable and then switch the camera back to GMSL mode, please?

Instructions for using the mode switch can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/14840675121043/comments/14843232242835

micah-cobot, 2024-07-23:
@MartyG-RealSense I've switched my camera from GMSL to USB and updated the firmware to 5.16.0.1 and im running SDK version 2.55.1. The firmware update completes successfully, but now IMU data does not stream from the device when connected over GMSL. See below for screenshots from realsense viewer. There are errors about "Data did not come within 5 seconds" and "Out of frame resources"... 

[image]
[image]

Nir-Az, 2024-07-24:
Can you please share the MIPI driver version you work with?

MartyG-RealSense, 2024-07-30:
Hi @micah-cobot  Could you please share your MIPI driver version as requested by @Nir-Az in the comment above, please?  Thanks!

closed 2024-08-17 by MartyG-RealSense

MartyG-RealSense, 2025-09-21:
Thanks so much @kaidokert for sharing the solution that worked for you!