realsenseai/librealsense #13173: Probe-commit control transfer failed with error: RS2_USB_STATUS_IO

LQ0212, 2024-07-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu20.04 | 
| Kernel Version (Linux Only)    |  (L4T 5.10.192)                                         | 
| Platform                                 | NVIDIA Jetson orin nano  |
| SDK Version                            |  { legacy / 2.5.0 }                          | 
| realsense-ros Version                            |  { 2.3.2 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
After I use the RSUSB backend without the kernel patching and build the SDK using the RSUSB method it still have error when i lanuch the rs_d435_camera_with_model.launch 
ERROR [281472375587184] (uvc-device.cpp:862) Probe-commit control transfer failed with error: RS2_USB_STATUS_IO
[ERROR] [1721367780.707439652]: An exception has been thrown: 
Failed to resolve the request: 
	Format: RGB8, width: 640, height: 480

Into:
	Formats: 
	 YUYV

[ERROR] [1721367780.709325917]: Exception: 
Failed to resolve the request: 
	Format: RGB8, width: 640, height: 480

Into:
	Formats: 
	 YUYV

MartyG-RealSense, 2024-07-19:
Hi @LQ0212  Have you defined a custom FPS speed for the 640x480 resolution, please?  The **Failed to resolve the request** error usually occurs because a stream configuration has been requested that the camera is not able to provide.

For example, if a custom FPS of 6 was set on a D455 or D456 then this error would occur because these particular camera models have a minimum FPS of 5.

LQ0212, 2024-07-19:
Thank you for your attention, i did't modify the launch file. I think the fps is be set correctly. There is the my launch info.
roslaunch realsense2_camera rs_d435_camera_with_model.launch 
... logging to /home/cc/.ros/log/28ecf2e4-45a5-11ef-bdbb-488f4cff0b90/roslaunch-ubuntu-64174.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://ubuntu:35177/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 103 lines cut ...]
[ INFO] [1721376121.145150635]: setupStreams...
 19/07 16:02:01,591 WARNING [281472448123248] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 19/07 16:02:01,905 WARNING [281472216850800] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 19/07 16:02:01,909 WARNING [281472023916912] (messenger-libusb.cpp:42) control_transfer returned error, index: 4, error: Resource temporarily unavailable, number: b
 19/07 16:02:01,911 WARNING [281472023916912] (messenger-libusb.cpp:42) control_transfer returned error, index: 4, error: Resource temporarily unavailable, number: b
 19/07 16:02:01,914 WARNING [281472023916912] (messenger-libusb.cpp:42) control_transfer returned error, index: 4, error: Resource temporarily unavailable, number: b
 19/07 16:02:01,917 WARNING [281472023916912] (messenger-libusb.cpp:42) control_transfer returned error, index: 4, error: Resource temporarily unavailable, number: b
 19/07 16:02:01,919 WARNING [281472023916912] (messenger-libusb.cpp:42) control_transfer returned error, index: 4, error: Resource temporarily unavailable, number: b
 19/07 16:02:01,922 WARNING [281472023916912] (messenger-libusb.cpp:42) control_transfer returned error, index: 4, error: Resource temporarily unavailable, number: b
 19/07 16:02:01,922 ERROR [281472448123248] (uvc-device.cpp:862) Probe-commit control transfer failed with error: RS2_USB_STATUS_IO
[ERROR] [1721376122.115376196]: An exception has been thrown: 
Failed to resolve the request: 
	Format: RGB8, width: 640, height: 480

Into:
	Formats: 
	 YUYV

[ERROR] [1721376122.115534120]: Exception: 
Failed to resolve the request: 
	Format: RGB8, width: 640, height: 480

Into:
	Formats: 
	 YUYV

MartyG-RealSense, 2024-07-19:
Thank you very much for the log.

The recommended firmware to use with librealsense 2.50.0 is **5.13.0.50** rather than 5.16.0.1, which is designed for librealsense 2.55.1.

The launch file **rs_d435_camera_with_model.launch** is typically only used when performing robot simulation with Gazebo.   Do you experience the error if you use the **rs_camera** launch file?

`roslaunch realsense2_camera rs_camera.launch`

LQ0212, 2024-07-19:
yes, if i modify the launch file like below it also has the same error. the origin rs_camera.launch can run without error but i can't get the topic /camera/depth/color/points which is necessary to my project
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="-1"/>
  <arg name="depth_height"        default="-1"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="-1"/>
  <arg name="color_height"        default="-1"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="true"/>
  <arg name="ordered_pc"                default="true"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="wait_for_device_timeout"   default="-1.0"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>

  <arg name="stereo_module/exposure/1"  default="7500"/>
[... 56 lines cut ...]

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="reconnect_timeout"        value="$(arg reconnect_timeout)"/>
      <arg name="wait_for_device_timeout"  value="$(arg wait_for_device_timeout)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="stereo_module/exposure/1" value="$(arg stereo_module/exposure/1)"/>
      <arg name="stereo_module/gain/1"     value="$(arg stereo_module/gain/1)"/>
      <arg name="stereo_module/exposure/2" value="$(arg stereo_module/exposure/2)"/>
      <arg name="stereo_module/gain/2"     value="$(arg stereo_module/gain/2)"/>

      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
      
    </include>
  </group>
</launch>

MartyG-RealSense, 2024-07-19:
The pointcloud is disabled by default in the launch file.  It can be enabled using the launch command below.

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#point-cloud

LQ0212, 2024-07-19:
I also try this, it can run.  However, when i try to echo the topic /camera/depth/color/points  there are the log about
[ INFO] [1721380215.120778310]: publishing unordered pointcloud.
[ WARN] [1721380215.346897315]: No stream match for pointcloud chosen texture Process - Color
and there is no output in the rostopic echo

MartyG-RealSense, 2024-07-19:
`No stream match for pointcloud chosen texture Process - Color` means that color frames are missing because they are being dropped.

You can publish an ordered pointcloud instead of an unordered one by adding **ordered_pc:=true** to your roslaunch instruction.

If this does not make a difference then you could try texturing the pointcloud with the infrared image instead of the RGB color image.

roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:="RS2_STREAM_INFRARED"

*******

Sometimes the pointcloud filter of the ROS1 wrapper will not work with Jetson boards.  A workaround is to perform an **RGBD** launch instead of rs_camera.

1.  Install support for an RGBD launch with the command below.

`sudo apt-get install ros-noetic-rgbd-launch`

2.  Use the rs_rgbd.launch file.

`roslaunch realsense2_camera rs_camera.launch`

You do not need to enable the pointcloud filter on this launch file as it is already enabled by default.

The pointcloud will be published to **depth_image_proc** instead of /camera/depth/color/points

/depth/image/proc

LQ0212, 2024-07-19:
thanks for your help. I will try it on Monday and update the results @MartyG-RealSense

MartyG-RealSense, 2024-07-19:
You are very welcome.  I look forward to your next update.  Good luck!

LQ0212, 2024-07-22:
Sorry to bother you @MartyG-RealSense . I try the method which use orderd_pc doesn't make any difference. And when I use infrared image the log output as below.
[ INFO] [1721609567.428119131]: publishing unordered pointcloud.
[ WARN] [1721609567.745491636]: No stream match for pointcloud chosen texture Process - Infrared

I try to use rs_rgbd.launch file 
`roslaunch realsense2_camera rs_rgbd.launch`
I can't find any topic with depth_image_proc.

MartyG-RealSense, 2024-07-22:
You're very welcome.  I'm pleased to hear that rs_rgbd worked for you!

MartyG-RealSense, 2024-07-30:
Hi @LQ0212  Do you require further assistance with this case, please?  Thanks!

LQ0212, 2024-07-30:
Hi @MartyG-RealSense thanks for your assistance, I have no further issue. I will close this issue.

closed 2024-07-30 by LQ0212