realsenseai/librealsense #13174: Resolution error

du-xinyi, 2024-07-19:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|

[image]
当我在笔记本上启动realsense-viewer时，在笔记本上的屏幕无法正常显示界面，显示出来的界面是这样的，我cpu是14900hx，显卡是rtx4060

MartyG-RealSense, 2024-07-19:
Hello @du-xinyi  There are suggestions about solving this graphics scaling problem in the RealSense Viewer at https://github.com/IntelRealSense/librealsense/issues/5511

******

你好 @du-xinyi 有关于解决 RealSense Viewer 中图形缩放问题的建议，请访问 https://github.com/IntelRealSense/librealsense/issues/5511

MartyG-RealSense, 2024-07-30:
Hi @du-xinyi  Do you require further assistance with this case, please?  Thanks!

您需要进一步协助处理此案吗？谢谢！

closed 2024-08-09 by MartyG-RealSense