realsenseai/librealsense #13175: No frames received on Ubuntu 22 for D435

heyunwang1, 2024-07-19:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|241222077177|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|241223121250|
|**Firmware Update Id**|241223121250|
|**Dfu Device Path**||

Please provide a description of the problem

Hi, I followed the[ instruction](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md) to install sdk, but The viewer shows no frames received, no other errors.
And also I followed this [link](https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide) installed  dependencies

MartyG-RealSense, 2024-07-19:
Hi @heyunwang1  Please try unplugging the micro-sized end of the USB cable from the side of the camera, reversing the connector's orientation and plugging it back into the camera (USB-C cables are two-way insertion at the micro-sized end). 
 Does doing so make any difference?

heyunwang1, 2024-07-19:
Hi, there are no differece
and also I try to use cheese to open
[image]

MartyG-RealSense, 2024-07-19:
So the camera is detected and displayed in the side-panel of the RealSense Viewer but you receive **No frames received** when enabling streams?  Which stream is it (depth or RGB, or both) that displays this message, please?

Does the Viewer behave the same if you run it in sudo admin permissions mode with **sudo realsense-viewer**

heyunwang1, 2024-07-19:
It also shows No frames received for both with sudo realsense-viewer

MartyG-RealSense, 2024-07-19:
Are you using the official 1 meter long USB cable supplied with the camera or a longer cable of your own choice?

If you are using the official cable, is the camera accessible with Cheese if you follow the procedure below:

1.  Uninstall the SDK packages with this command: 

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge
`
2.  Re-install the SDK from source code using the **libuvc backend** installation method at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

heyunwang1, 2024-07-19:
[image]
Now the software can basically receive a picture in 20 seconds.
“20:01:19 [Error] /home/drt/librealsense_build/librealsense-master/src/uvc/uvc-streamer.cpp:105 - uvc streamer watchdog triggered on endpoint: 132”
“20:01:17 [Warn] /home/drt/librealsense_build/librealsense-master/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11”
“20:04:30 [Error] /home/drt/librealsense_build/librealsense-master/src/ds/ds-options.cpp:93 - Asic Temperature value is not valid!”

For cheese, it does not report an error, but its screen is black
[image]

MartyG-RealSense, 2024-07-19:
The **control_transfer returned** warning indicates a communication problem between the camera and computer, typically with the USB system (a problem with the USB port or the USB cable).

The **Asic Temperature value is not valid!** message is not important.

Do you have a different USB port that you can try the camera in?

heyunwang1, 2024-07-19:
Thank you, I will buy a new cable tommorow

closed 2024-08-09 by MartyG-RealSense