realsenseai/librealsense #13176: Failed to write 'change' to '/sys/devices/platform/fe084040.ir/input/input10/uevent': Cannot allocate memory

w407022008, 2024-07-19:
Hello,
Here I'm building librealsense-2.50.0 on Ubuntu 20.04 Khadas Vim4 arm64 SBC. An error occur when i setup the udev rules by ./scripts/setup_udev_rules.sh:   
`Failed to write 'change' to '/sys/devices/platform/fe084040.ir/input/input10/uevent': Cannot allocate memory
Failed to write 'change' to '/sys/devices/platform/fe084040.ir/input/input11/uevent': Cannot allocate memory
Failed to write 'change' to '/sys/devices/platform/fe084040.ir/input/input12/uevent': Cannot allocate memory`
Let me know how should i fix it pls.

MartyG-RealSense, 2024-07-19:
Hi @w407022008  There have been a couple of past cases of this error at https://github.com/IntelRealSense/librealsense/issues/11393 and https://github.com/IntelRealSense/realsense-ros/issues/2545 that also occurred when installing the udev rules with `./scripts/setup_udev_rules.sh`

One of the RealSense users in those cases found that building the librealsense SDK from source code using the installation script **libuvc_installation.sh** at the link below has worked for them.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

MartyG-RealSense, 2024-07-19:
If you are not able to install the udev rules, are you able to launch programs in sudo admin permissions mode by putting 'sudo' in front of the launch command.  For example for the realsense-viewer tool: 

**sudo realsense-viewer**

MartyG-RealSense, 2024-07-19:
Infrared topics in the ROS wrapper are disabled by default and have to be enabled.  You can do this by adding the parameters below to your ROS launch instruction.

**enable_infra1:=true enable_infra2:=true**

You could also try adding **initial_reset:=true** to reset the camera at launch to see if this reduces the problems.

MartyG-RealSense, 2024-08-09:
Case closed due to no further comments received.

closed 2024-08-09 by MartyG-RealSense

w407022008, 2024-08-27:
It's problem of the host device hardware issue.