realsenseai/librealsense #13182: D415 found, L515 not using python wrapper

fcopp, 2024-07-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 + L515  | 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                           | 2.55.1.6486                | 
| Language                               | python                        | 

### Issue Description
I have a D415 and a L515. Both are detected and work in Intel RealSense Viewer. 
When I run the following snippet only the D415 is detected, the L515 is not. Same USB-C cable plugged into the same port.

```python
import pyrealsense2 as rs

ctx = rs.context()

for d in ctx.devices:
    print ('Found device: ', d.get_info(rs.camera_info.name), ' ', d.get_info(rs.camera_info.serial_number))

else:
    print("No Intel Device connected")
```

Originally tried to run examples from the python examples folder and got this far debugging.

MartyG-RealSense, 2024-07-23:
Hi @fcopp  Support for L515 was removed in SDK 2.55.1 as it is a retired camera model.  SDK 2.54.2 is the final version that supports L515.

fcopp, 2024-07-24:
That would explain it. Thanks for the help

closed 2024-07-29 by fcopp