realsenseai/librealsense #13184: PLEASE I CAN'T INSTALL SDK IN MY WINDOWS

DORIAN1962, 2024-07-23:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2024-07-23:
Hi @DORIAN1962  Which RealSense camera model are you using, please?  For example, if you have an L515 then it would not appear in the RealSense Viewer because support for L515 was removed in SDK 2.55.1 and the last version that supports L515 was 2.54.2.

Does the error appear when the RealSense Viewer first opens, or does it appear when an option is selected?

This error sometimes occurs when trying to access the RGB camera from the Add Source option, like in https://github.com/IntelRealSense/librealsense/issues/11338

DORIAN1962, 2024-07-23:
How do I know which camera I'm using? Since I want to read with the pc
camera directly and please tell me how to solve this problem. Thank you for
your precious help

MartyG-RealSense, 2024-07-23:
If the camera is plugged in then its details should appear in the options side-panel of the Viewer, with the camera name at the top of the panel.

[image]

It sounds though as if you want to use your PC's built in RGB webcam in the Viewer instead of a RealSense camera and access it using the 'Add Source' button at the top of the Viewer's side-panel.  Is that correct please?  If that is what you are aiming to do, then unfortunately not all computer webcams work with the Add Source option and there is no fix if the PC webcam is not compatible.

DORIAN1962, 2024-07-23:
Okay thank you but which model would be compatible according to you so I
have to download which version to use the WEBCAM of my pc huawei raizen 5

MartyG-RealSense, 2024-07-23:
There is no way to know in advance which computer's built-in webcam, or which external USB webcam, will work with the Add Source button.  

For example, I have a laptop where the built-in webcam does not work with Add Source and another laptop where the camera does work with it.

If you have a Windows computer then you could instead try accessing your PC's camera with the built-in Windows camera application.  You can find this by typing 'camera' into the search box at the bottom of your Windows screen.

DORIAN1962, 2024-07-23:
Thanks, I'll check it out.

closed 2024-08-09 by MartyG-RealSense