realsenseai/librealsense #13188: error for "cmake../"

monson-li, 2024-07-24:
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.55.1
-- Setting Unix configurations
-- No output directory set; using /home/sti/librealsense-master/build/Release/
-- Building libcurl enabled
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libcrypto.so (found version "1.1.1") 
You have called ADD_LIBRARY for library realsense2 without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Performing Test SUPPORTS_CXX14
-- Performing Test SUPPORTS_CXX14 - Success
-- using RS2_USE_V4L2_BACKEND
-- Found usb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libudev'
--   Found libudev, version 237
-- Found Udev: /usr/include  
-- Fetching nlohmann/json...
Cloning into 'json'...
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
Cloning into 'json'...
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
Cloning into 'json'...
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
CMake Error at /home/sti/librealsense-master/build/external-projects/json-download/tmp/nlohmann_json-gitclone.cmake:66 (message):
  Failed to clone repository: 'https://github.com/nlohmann/json.git'

make[2]: *** [src/nlohmann_json-stamp/nlohmann_json-download] Error 1
make[1]: *** [CMakeFiles/nlohmann_json.dir/all] Error 2
make: *** [all] Error 2
CMake Error at CMake/external_json.cmake:34 (message):
  Failed to download nlohmann/json
Call Stack (most recent call first):
  CMake/external_json.cmake:50 (get_nlohmann_json)
  third-party/CMakeLists.txt:3 (include)
  CMakeLists.txt:58 (include)

-- Configuring incomplete, errors occurred!
See also "/home/sti/librealsense-master/build/CMakeFiles/CMakeOutput.log".

How can I solve this

MartyG-RealSense, 2024-07-24:
Hi @monson-li  It looks as though you are building building librealsense 2.55.1 from source code with CMake.  Which installation method did you use, please?

For example, you can download the source code for 2.55.1 as a zip file and build it using the instructions at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

Do you experience the same problem with building librealsense 2.55.1 from source code if you do so using the **libuvc backend** build script at the link below, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

MartyG-RealSense, 2024-08-09:
Case closed due to no further comments received.

monson-li, 2024-08-22:
Hi, MartyG. I have downloaded the zip file "librealsense-master" and unzip that. When I cmale../ in the "build" file, it shows me RPC failed.

MartyG-RealSense, 2024-08-22:
Possible causes and fixes for the 'RPC failed' are listed in your earlier issue at https://github.com/IntelRealSense/librealsense/issues/13250#issuecomment-2277486376

Lithinem, 2024-10-24:
> Hi @monson-li It looks as though you are building building librealsense 2.55.1 from source code with CMake. Which installation method did you use, please?
> 
> For example, you can download the source code for 2.55.1 as a zip file and build it using the instructions at [#9931 (comment)](https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692)
> 
> Do you experience the same problem with building librealsense 2.55.1 from source code if you do so using the **libuvc backend** build script at the link below, please?
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Hello kind person,

I am a newbe to this stuff, but I have the same issue. Can you help me please?

I tried to install ROS2 on my Win10 Laptop so I can use my IntelRealSense D4XX. I dollowed https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#installation-on-windows for installation but Step 3.5 "Build librealsense2 package only" fails with 

CMake Error at CMake/external_json.cmake:34 (message):
  Failed to download nlohmann/json
Call Stack (most recent call first):
  CMake/external_json.cmake:50 (get_nlohmann_json)
  third-party/CMakeLists.txt:3 (include)
  CMakeLists.txt:57 (include)

My firewall is not the problem. And it seems there are at least some files in one of my folders with nlohmann in the name. So maybe there is a sequencing error? Or my Installation does not know about nlohmann bein downloaded? My path is C:\realsense-ros-4.56.1 with C:\realsense-ros-4.56.1\librealsense-2.56.2 being the source files of the sdk. Installing the SDK from WIN10 Binaries did not fix the problem.

Im looking forward to your answer. 

In hoping expectations.
Lith

MartyG-RealSense, 2024-10-24:
@Lithinem  I looked at this issue carefully but was unable to find a solution, unfortunately.  I sincerely apologise.  I recommend using the ROS2 wrapper on Ubuntu whenever possible, as attempts to install the wrapper on Windows have always had problems.

MartyG-RealSense, 2024-10-27:
Hi @monson-li and @Lithinem   My Intel RealSense colleagues looked at your case and said that it looks like a communication issue on your end, such as a problem with ethernet or a VPN.  They quoted the following section of the log of @monson-li 

```
Cloning into 'json'...
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
```

@monson-li - you previously also had an 'RPC failed' error a couple of months ago when installing librealsense over vcpkg at https://github.com/IntelRealSense/librealsense/issues/13250

MartyG-RealSense, 2024-11-04:
Does anyone require further assistance with this case, please?  Thanks!

closed 2024-11-15 by MartyG-RealSense