realsenseai/librealsense #13194: L515 not working in ubuntu 20.04 realsense-viewer

Daya-AWR1642, 2024-07-25:
Hi, 
I had followed the all the necessary steps to install realsense viewer for ubuntu 20.04
[installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) 
but when I run the command _realsense-viewer_ 
it's not getting 

[image]

MartyG-RealSense, 2024-07-25:
Hi @Daya-AWR1642  If you used the **distribution_linux.md** guide then it is likely that the current latest **2.55.1** version of the RealSense SDK has been installed from dkms packages.

Support for L515 has been removed in 2.55.1 onwards as it is a retired camera model, so it is necessary to install the previous **2.54.2** SDK version in order to use L515 with it.

You can either build 2.54.2 from source code or use the **showpkg** command to find the filenames of the packages for the 2.54.2 version, as described at https://github.com/IntelRealSense/librealsense/issues/12892#issuecomment-2097781830

MartyG-RealSense, 2024-08-03:
Hi @Daya-AWR1642  Do you require further assistance with this case, please?  Thanks!

Daya-AWR1642, 2024-08-04:
No thanks

closed 2024-08-04 by Daya-AWR1642