realsenseai/librealsense #13196: MIPI issue

oaragones, 2024-07-25:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|238222073000|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|237223025099|
|**Firmware Update Id**|237223025099|
|**Dfu Device Path**||

Please provide a description of the problem

Hello, when we try to connect to Real Sense device into stereo mode, an error appear:

Right MIPI Error.

10:20:15 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\notifications.cpp:511 - Right MIPI error
10:20:16 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\core\notification.cpp:22 - IR stream start failure
10:20:16 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\notifications.cpp:511 - IR stream start failure
10:20:17 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\core\notification.cpp:22 - Right MIPI error
10:20:18 [Info]  - Stopping streaming

With the RGB camera no problems detected.

We can't operate with the stereo vision. What can we do?

We update the firmware to the latest version.

We try to make a Hardware reset with no effects...

[image]

MartyG-RealSense, 2024-07-25:
Hi @oaragones  Are you using the official 1 meter USB cable supplied with the cable or a cable of your own choice, please?

Does it make a difference if you unplug the micro-sized end of the USB cable from the side of the camera, turn the connector around the other way and re-insert it (USB-C cables are two-way insertion at the micro-sized end).

oaragones, 2024-07-25:
Dear Marty, 

Thanks for your fast answer.

We don't have the supplied cable from the camera but we try several cables (from 1 m to other lengths) without success.

Is the MIPI something to be for the IO interface from the camera? we don't need it, is it possible to deactivate or something like that?

Thanks in advance,

MartyG-RealSense, 2024-07-25:
Cable quality is important with RealSense cameras.  It should be one that is designed for data transfer instead of an inexpensive one suited only to device charging.  Cable quality is especially important for lengths of 2 m and greater, otherwise the camera connection can degrade or disconnect.

MIPI errors can be related to a cable that joins together two separate circuit boards inside the camera casing.

Do you know whether your camera has been stored inactive in a location such as a drawer or cupboard for weeks or months before you experienced this error?  In rare cases, such storage can result in a fault in the camera hardware that could cause this issue with the depth stream.

oaragones, 2024-07-25:
Dear Marty,

We will contact to our provider to get this information.

We are the development team and we don't know about that.

We cables that we use are 1 meter and less.

Thanks in advance!

MartyG-RealSense, 2024-07-25:
You are very welcome.  I look forward to your next update.  Thanks very much!

closed 2024-08-17 by MartyG-RealSense