realsenseai/librealsense #13198: Error in using Python Wrapper

titan1230, 2024-07-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    | 6.6.41-current-bcm2712 | 
| Platform                                 | Raspberry Pi|
| SDK Version                            |   2.55.1                           | 
| Language                            |  Python                           | 
| Segment			|  Robot                           | 

### Issue Description
The camera is working properly on the Intel Realsense Viewer, but we get this error when we try to run it on python.
I installed the SDK by following [this guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md)
Installed pyrealsense2 via pip.

# Error Message
```
Traceback (most recent call last):
  File "/home/pi/tests/RealSense/error.py", line 14, in <module>
    frames = pipe.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000
```

# Code
```py
import cv2
import pyrealsense2 as rs
import numpy as np

pipe = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
pipe.start(config)

try:
    while True:
        frames = pipe.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        
        images = np.hstack((color_image, depth_image))
        
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)
finally:
    pipe.stop()
```

MartyG-RealSense, 2024-07-27:
Hi @titan1230  Can you confirm which model of RealSense 400 Series camera you have, please (D415, D435, D455, etc).

titan1230, 2024-07-27:
@MartyG-RealSense
Thank you for the reply.
I am using D435i

MartyG-RealSense, 2024-07-27:
Thanks very much for the model confirmation.

Does the error still occur if you place the value '5000' inside the brackets of the wait_for_frames() instruction?

`frames = pipe.wait_for_frames(5000)`

titan1230, 2024-07-28:
That doesn't work either.

MartyG-RealSense, 2024-07-28:
Sometimes with **Frame didn't arrive within 5000** errors on Raspberry Pi, the problem is with the Pi itself and the code works fine on a PC.  RealSense cameras typically experience more problems when used on Pi boards than when used with PCs.  One of these known problems that can occur is the camera working with the Viewer but not with pyrealsense2.

I note that you installed the pyrealsense2 wrapper with pip install.  Whilst pyrealsense2 does have Arm packages for the 'pip install pyrealsense2' installation method, they (a) only work for Python 3.7 to 3.9, and (b) do not work correctly when used on Raspberry Pi or Nvidia Jetson boards (Jetson is also an Arm device).  It is therefore recommendable to install the pyrealsense2 wrapper from source code instead of with pip install on Pi and Jetson.

A method of installing librealsense that works well on Pi is to install it from source code using the **libuvc backend** build script.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

You can build librealsense and the pyrealsense2 wrapper at the same time if you edit line 46 of the script to add the flag **-DBUILD_PYTHON_BINDINGS:bool=true** to the CMake build instruction.

If you want the RealSense Viewer to be included in the build then also add the flags **-DBUILD_EXAMPLES=TRUE -DBUILD_GRAPHICAL_EXAMPLES=TRUE**

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

************

Before changing your installation though, you could first test whether the RGB color stream is the cause of the problem by using the infrared stream instead.

```
import cv2
import pyrealsense2 as rs
import numpy as np

pipe = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.infrared, 640, 480, rs.format.y8, 30)
pipe.start(config)

try:
    while True:
        frames = pipe.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        infrared_frame = frames.get_infrared_frame()
        
        depth_image = np.asanyarray(depth_frame.get_data())
        infrared_image = np.asanyarray(infrared_frame.get_data())
        
        images = np.hstack((infrared_image, depth_image))
        
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)
finally:
    pipe.stop()
```

titan1230, 2024-07-28:
I ended up using ROS with C++ and that worked perfectly fine.
Thanks for the help.

closed 2024-07-28 by titan1230

MartyG-RealSense, 2024-07-28:
You are very welcome.  I'm pleased to hear that you achieved a solution.  Thanks very much for the update!