realsenseai/librealsense #13199: tool[ rs_server ]  "Segmentation fault (core dumped)" issues

BoydenYubin, 2024-07-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.12.03.00) | 
| Operating System & Version |   {Linux (Ubuntu 22.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 5.05.0-1058-raspi)                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  { legacy / 2.<53>.<1> }                          | 
| Language                            |  {C }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I have built the lib-realsense based on the Raspberry Pi 4B and the sdk version is 2.53.1. This SDK is the latest version with tool rs-sever. But when I use rs-server runing on Raspberry Pi, and VLC media player as client, every time the connection has build, rs-server will corrupt with `Segmentation fault (core dumped)`. So what can I do to solve the problem. Thanks. 
[rtsp.zip](https://github.com/user-attachments/files/16400180/rtsp.zip)
I have uploaded the tcpdump package to make it more clear.

MartyG-RealSense, 2024-07-27:
Hi @BoydenYubin  If you are using SDK 2.53.1 then the firmware version that should be installed in the camera is **5.14.0.0**.

The firmware version is not likely to be the main cause of your problem though.  Usually, trying to access the RTSP stream of rs-server with non-RealSense tools instead of with the RealSense Viewer tool or RealSense SDK script code does not work.  There is a discussion about it at https://github.com/IntelRealSense/librealsense/issues/10345

BoydenYubin, 2024-07-27:
If I use [mediamtx](https://github.com/bluenviron/mediamtx) that transfer the rstp protocol, it seems that the stream still can't be found.

MartyG-RealSense, 2024-07-27:
The only example I have seen of rs-server being accessed by a method other than the RealSense Viewer or SDK program code is when a RealSense user accessed from ROS by changing the RTSP protocol for ROS protocols, as described at https://github.com/IntelRealSense/realsense-ros/issues/1808

All past attempts to access rs-server's RTSP stream with a non-RealSense tool such as VLC or GStreamer have not been successful.

closed 2024-08-09 by MartyG-RealSense