realsenseai/librealsense #13201: Jetson orin nano /camera/imu

CSARobotics, 2024-07-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435i  | 
| Firmware Version                   | (5.16.0.1) | 
| Operating System & Version |   Linux (Ubuntu 20.04)   | 
| Kernel Version (Linux Only)    | 5.10                                        | 
| Platform                                 | NVIDIA Jetson Orin Nano |
| SDK Version                            |  legacy / 2.55.1.0                         | 
| Language                            |  python                          | 
| Segment			|  Autonomous Robot }                         | 

### Issue Description
When I tried to use /camera/imu topic, any data does not come and also i couldn't visualize in rviz. Realsense-viewer send some error for motion module.  

[image]
[image]
[image]

I tried to solve this problem with this issue solution.
https://github.com/IntelRealSense/realsense-ros/issues/2563 
I got this

https://github.com/user-attachments/assets/b8256df1-6baf-4086-b3f1-9058068bb719

MartyG-RealSense, 2024-07-29:
Hi @CSARobotics  When the **control_transfer returned** message generates continuously, it can indicate that there is a communication problem between the camera and the computer / computing board, typically an issue with the US port or the USB cable.

In the RealSense ROS wrapper, the IMU topics are disabled by default.  They can be enabled and published in the **/imu** topic by adding the commands below to your ROS launch instruction.

**ROS1**
`enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation`

**ROS2**
`enable_gyro:=true enable_accel:=true unite_imu_method:=2`

MartyG-RealSense, 2024-08-03:
Hi @CSARobotics  Do you have an update about this case that you can provide, please?  Thanks!

CSARobotics, 2024-08-08:
Thank you for your interest. When I try it on my laptop with the same cable, there is no problem. So I don't think it's the cable. There does not appear to be a problem with ROS Wrapper. I think the problem is with the SDK. I installed the SDK using this repo https://github.com/jetsonhacks/installRealSenseSDK. There is no problem with RGB and Stereo camera, but I cannot view IMU data even with Realsense Viewer.

MartyG-RealSense, 2024-08-08:
When using the installRealSenseSDK repo, did you install from packages with the **installLibrealsense.sh** script or build from source code with the **buildLibrealsense.sh** script?

The buildLibrealsense script should have a similar effect to using the **libuvc backend** procedure mentioned earlier in this discussion, as both the libuvc script and the buildLibrealsense script include the CMake build flag `-DFORCE_LIBUVC=ON`.  

However, if you used installLibrealsense.sh to build from packages then the IMU would not be supported on JetPack 6.

CSARobotics, 2024-08-08:
I used the buildLibrealsense script. The flags were as follows: /usr/bin/cmake ../ -DBUILD_EXAMPLES=true -DFORCE_LIBUVC=ON -DBUILD_WITH_CUDA="$USE_CUDA" -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python3)

I'm currently using JetPack 6, but it doesn't matter to me. I can also switch to Jetpack 5 if it will work there.

MartyG-RealSense, 2024-08-08:
Yes, I would strongly recommend trying JetPack 5.1 if you do not mind trying a downgrade of your JetPack.

CSARobotics, 2024-08-08:
So, JetPack 5.1.2 or 5.1.3? Also, when I tried them with the buildLibrealsense script before, I got the same result. Is there an SDK installation guide where I can run IMU for JetPack 5?

CSARobotics, 2024-08-08:
I'm currently flashing JetPack 5.1.3. I will try it as soon as it is finished and write the results.

CSARobotics, 2024-08-08:
Realsense d435i became operational with all its functions.

Installation Method:

1 - Install Jetpack 5.1.3.

2 - Clone this repo https://github.com/jetsonhacks/installRealSenseSDK.

3 - Run installLibrealsense.sh

4 - Congratulations, the installation is completed.

Thank you very much for your help @MartyG-RealSense

closed 2024-08-08 by CSARobotics

MartyG-RealSense, 2024-08-08:
You are very welcome, @CSARobotics - it's great to hear that JetPack 5.1.3 was a successful solution for you.  Thanks very much for the update and for sharing your method!