realsenseai/librealsense #13203: Build the SDK from source code on Windows with CMake

guotonglucky, 2024-07-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | ( 05.12.14.50 ) | 
| Operating System & Version | { Win 10 } | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            | { v2.47.0 }                          | 
| Language                            |  { C++ }                          | 
| Segment			|                      | 

[image]

### Issue Description
I need to connect to the camera over network. I build the SDK from source code on Windows with CMake. It works fine over Ethernet. But it doesn't work over USB. The **realsense-viewer.exe built from the SDK** also does not recognize the camera. The **Intel.RealSense.Viewer.exe downloaded from the official website** can recognize the camera.
[image]
[image]

I run the example **rs-hello-realsense**. Then came the error. Do I need to configure something else?
[image]

MartyG-RealSense, 2024-07-29:
Hi @guotonglucky  Please try building the SDK without **FORCE_RSUSB_BACKEND** set to true.  This flag behaves differently on Windows from how it behaves on Linux.  On Linux it enables the kernel to be bypassed.  On Windows it causes the SDK's default backend to be changed from Microsoft Media Foundation to a WinUSB rewrite of the UVC protocol, and the flag is set to false (unticked) in the Windows source code build instructions.

************

If the camera still does not work with USB afterwards then you could try a workaround.  After the SDK is built, find its folder on your computer.  When the SDK is built using the **Intel.RealSense.SDK-WIN10** installer file then it is located at **C:\Program Files (x86)\Intel RealSense SDK 2.0\tools** so I recommend looking for it there first.

After renaming the Intel RealSense SDK 2.0 folder, run the Intel.RealSense.SDK-WIN10 installer file for SDK 2.47.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.47.0

This will install the non-networked version of the SDK at C:\Program Files (x86)\Intel RealSense SDK 2.0\tools

You may then be able to use the SDK at the renamed source code folder for network access and the non-networked SDK for everything else.

guotonglucky, 2024-07-29:
The first method works. Set **FORCE_RSUSB_BACKEND** to false. Thanks

MartyG-RealSense, 2024-07-29:
You are very welcome.  I'm pleased that I could help!

MartyG-RealSense, 2024-08-09:
Hi @guotonglucky  Do you require further assistance with this case, please?  Thanks!

guotonglucky, 2024-08-12:
No, thanks.

closed 2024-08-12 by guotonglucky