realsenseai/librealsense #13204: "Frame didn't arrive within 5000" in pyrealsense2 in docker environment

tkitagawa1999, 2024-07-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D456 | 
| Operating System & Version | Linux (Ubuntu 20.04) | 
| Platform                                 | NVIDIA Jetson Orin NX 8GB |
| SDK Version                            | pyrealsense2  2.55.1.6486| 
| Language                            | python3.8                           |                       | 

### Issue Description
In a local environment using Jetson, it is possible to install pyrealsense2 using pip and run the following code to get the frame, but if you run it using the following docker compose, you will get “RuntimeError: Frame didn't arrive However, when the following code is executed using docker compose, the camera is recognized but frame acquisition fails with “RuntimeError: Frame didn't arrive within 5000. “ Is there anything else I need to do besides mounting the device?

**test.py**
```python
import pyrealsense2 as rs

ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
    dev.hardware_reset()

config = rs.config()

pipeline = rs.pipeline()
profile = pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        print(frames)

finally:
    pipeline.stop()
```

**docker-compose.yml**
```
version: "3.9"
services:
  test:
    build:
      context: .
      dockerfile_inline: |
        FROM ubuntu:focal
        ARG DEBIAN_FRONTEND=noninteractive
        RUN apt-get -y update
        RUN apt-get -y install python3.8 python3.8-dev python3-distutils python3-pip libusb-1.0-0-dev
        RUN python3.8 -m pip install pyrealsense2
        WORKDIR /
        COPY test.py /test.py
        CMD ["/usr/bin/python3.8", "/test.py"]
    privileged: true
    runtime: nvidia
    restart: always
    volumes:
      - /dev/video0:/dev/video0
      - /dev/video1:/dev/video1
      - /dev/video2:/dev/video2
      - /dev/video3:/dev/video3
      - /dev/video4:/dev/video4
      - /dev/video5:/dev/video5
      - /dev/video6:/dev/video6
      - /dev/video7:/dev/video7
      - /dev/video8:/dev/video8
      - /dev/video9:/dev/video9
      - /dev/video10:/dev/video10
      - /dev/video11:/dev/video11
      - /dev/media1:/dev/media1
      - /dev/media2:/dev/media2
    device_cgroup_rules:
      - 'c 81:* rmw'
      - 'c 189:* rmw'
```

MartyG-RealSense, 2024-07-29:
Hi @tkitagawa1999  Does it make a difference if you comment out the **dev.hardware_reset()** line of your Python test script, as typically you would only need to perform a reset if the script cannot run successfully without it.

tkitagawa1999, 2024-07-29:
Hi @MartyG-RealSense 

I have commented out the dev.hardware_reset() line in my Python test script, and it does not affect the results.

MartyG-RealSense, 2024-07-29:
A Jetson user with a Docker compose yaml file at https://github.com/IntelRealSense/librealsense/issues/12218 had the same problem of their Python test script working locally but experiencing the **Frame didn't arrive within 5000** error in Docker.  There was not a solution reached in that case though as the user stopped commenting.

Another user took the approach of listing their /dev channels as **devices** instead of volumes in their compose file.

https://medium.com/99p-labs/navigating-depth-perception-f9d4874dd88f

```
privileged: true
devices:
- "/dev/HID-SENSOR-2000e1.3.auto:/dev/HID-SENSOR-2000e1.3.auto"
- "/dev/iio:device0:/dev/iio:device0"
- "/dev/iio:device1:/dev/iio:device1"
- "/dev/video0:/dev/video0"
- "/dev/video1:/dev/video1"
- "/dev/video2:/dev/video2"
- "/dev/video3:/dev/video3"
- "/dev/video5:/dev/video5"
- "/dev/video5:/dev/video5"
```

tkitagawa1999, 2024-07-29:
Trying the two methods you suggested did not change the error. Is there any other way to resolve this issue?
・https://github.com/IntelRealSense/librealsense/issues/12218
・https://medium.com/99p-labs/navigating-depth-perception-f9d4874dd88f

MartyG-RealSense, 2024-07-29:
Usually the advice that I give for Docker users is to set privileged to true (which you have already done) and to launch Docker in sudo admin permissions mode with **sudo docker run**.  I do not know though how Docker can be activated in sudo mode when using a compose file (my knowledge of Docker is limited).

TheBlind11, 2024-08-02:
I’m facing the same issue with cam D455 in a docker container on NVIDIA AGX Xavier Developer kit, I don’t know how to solve the problem: it works on the same platform outside of the container and it works on another platform in docker (Dell Ubuntu 22.04), it’s strange

MartyG-RealSense, 2024-08-03:
@tkitagawa1999  and @TheBlind11  Does it make a difference if you reload the **udev** device handling rules?

```
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control –reload-rules && udevadm trigger
```

TheBlind11, 2024-08-03:
No it doesn't make any difference, I tried it. I also tried to put the plugdev user in the container, nothing changes. We are using D455 cam with the latest firmware and the latest version of the sdk installed on the Jetson.

MartyG-RealSense, 2024-08-03:
This is a difficult issue, as there are few references about using Docker with Jetson and pyrealsense2 (there are a greater number of references about Jetson and ROS).

Of the Docker / pyrealsense2 Jetson references that are available, https://github.com/IntelRealSense/librealsense/issues/7849 may be helpful.

TheBlind11, 2024-08-03:
It seems like a guide to install pyrealsense in the container, I don't think I'm looking for it.

MartyG-RealSense, 2024-08-03:
@tkitagawa1999 is using pyrealsense2.  You said that you have the 'same issue'.  Are you not using pyrealsense2, please?

TheBlind11, 2024-08-03:
Yeah, I mean we are using the same library obv, but the problem is not the installation of the library, but the fact that the container doesn't receive any frame. We give the container the privileged flag and all /dev path and we checked that camera is recognised in the container.

MartyG-RealSense, 2024-08-03:
@TheBlind11  The most useful references available on this subject are https://github.com/IntelRealSense/librealsense/issues/12218 mentioned at the start of this issue and https://github.com/IntelRealSense/librealsense/issues/7849

As mentioned above, the other available Jetson / Docker resources mostly relate to ROS specifically, unfortunately, and so would likely not be helpful to resolving a 'no frames received' problem.

Does your docker run instruction include the **-it --rm** parameters?  For example:

`sudo docker run -it --rm --privileged=true`

TheBlind11, 2024-08-03:
Yes we include them

MartyG-RealSense, 2024-08-03:
@TheBlind11  Are you able to use the realsense-viewer tool in your Docker camera and stream frames successfully with it?

If 'no frames received' occurs, are you able to stream successfully if you perform a hardware reset of the camera from the 'More' option in the realsense-viewer side-panel or reset using a **hardware_reset()** instruction inserted into a pyrealsense2 script?

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

TheBlind11, 2024-08-03:
We can successfully stream frames with the realsense viewer outside docker, we haven't tried it inside docker because we don't need the viewer in the container. We have already tried the hardware reset instruction, nothing changes.

MartyG-RealSense, 2024-08-03:
The suggestion to try realsense-viewer within Docker is to confirm whether the problem is with pyrealsense2 specifically.  Sometimes on Jetson boards realsense-viewer will work fine whilst pyrealense2 has 'no frames received'.

TheBlind11, 2024-08-03:
[image]
This is the situation in realsense-viewer whitin Docker container, even in 3D section we can't get any frame.

TheBlind11, 2024-08-03:
We also tried with an arch base image (instead of ubuntu 20.04) and the realsense viewer works fine, in 2D and 3D mode, but still we can't receive any frame in our python script.

MartyG-RealSense, 2024-08-04:
I note that you have included **--device /dev/video0** in your Dockerfile.  By default on RealSense 400 Series cameras, dev/video0 corresponds to the depth stream, /video1 is the infrared stream and /video2 is the RGB stream, as described at https://github.com/IntelRealSense/librealsense/issues/3416#issuecomment-471720605

Is the RGB stream of realsense-viewer able to be displayed in Docker if you add **/dev/video2** to the list of devices in the Dockerfile?

TheBlind11, 2024-08-04:
Yeah we also tried it in another base image of our computer vision microservice, I'm going to provide you the dockerfile and the docker run command.

```
FROM nvcr.io/nvidia/l4t-pytorch:r35.2.1-pth2.0-py3
ENV DEBIAN_FRONTEND=noninteractive

# Disable the non-working nvidia repo in the Docker image
RUN sed -i 's/^deb https:\/\/repo.download.nvidia.com\/jetson\/common r35.2 main/#&/' /etc/apt/sources.list

# Some cleanup of unused packages
RUN apt -y autoremove --purge \
    python2 python2.7 python-is-python2 \
    python3.9 python3.9-minimal python3.9-vpi2

#RUN apt -y autoremove --purge cuda-toolkit-11-4

# Package upgrades
#RUN apt-get update && apt-get upgrade -y --autoremove
RUN python3 -m pip install --upgrade pip

# Pyrealsense2 compiling and installation
RUN apt update && apt -y install libgl1-mesa-glx libusb-1.0-0-dev
WORKDIR /var/tmp
COPY compile_pyrealsense2.sh .
RUN ./compile_pyrealsense2.sh

# OpenCV compiling and installation

# Install tzdata
RUN apt-get update && apt-get -y install tzdata

# Install dependencies
#RUN apt update && apt upgrade && apt -y install \

RUN apt update && apt -y install \
    git \
    git-buildpackage \
    debhelper \
    autoconf \
    libncursesw5-dev

#    python3.8

# Install build dependencies TODO check duplicates
RUN apt update && \
    apt install -y \
    build-essential \
    cmake \
    wget \
    unzip \
    gfortran \
    libatlas-base-dev \
    libavcodec-dev \
    libavformat-dev \
    libavresample-dev \
    libcanberra-gtk3-module \
    libdc1394-22-dev \
    libeigen3-dev \
    libglew-dev \
    libgstreamer-plugins-base1.0-dev \
    libgstreamer-plugins-good1.0-dev \
    libgstreamer1.0-dev \
    libgtk-3-dev \
    libjpeg8-dev \
    libjpeg-turbo8-dev \
    liblapack-dev \
    liblapacke-dev \
    libopenblas-dev \
    libpng-dev \
    libpostproc-dev \
    libswscale-dev \
    libtbb-dev \
    libtbb2 \
    libtesseract-dev \
    libtiff-dev \
    libv4l-dev \
    libxine2-dev \
    libxvidcore-dev \
    libx264-dev \
    pkg-config \
    qv4l2 \
    v4l-utils \
    zlib1g-dev \
    yasm \
    pkg-config \
    libswscale-dev \
    libtbb2 \
    libtbb-dev \
    libavformat-dev \
    libpq-dev \
    libxine2-dev \
    libglew-dev \
    libtiff5-dev \
    zlib1g-dev \
    libjpeg-dev \
    libavcodec-dev \
    libavformat-dev \
    libavutil-dev \
    libswscale-dev \
    libeigen3-dev \
    libtbb-dev \
    libgtk2.0-dev \
    pkg-config \
    libeigen3-dev
#    python-dev \
#    python3.8 \
#    python3.8-dev \
#    git \
#    python3-dev
#    python-numpy \

RUN mkdir "build_opencv"

COPY build_opencv.sh build_opencv/

#RUN apt-get remove -y python3-numpy python3-matplotlib \
#    && apt-get install -y python3-pip \
#    && python3.8 -m pip install pybind11 \
#    && python3.8 -m pip install Cython==0.29.36 \
#    && python3.8 -m pip install setuptools==49.6.0 \
#    && python3.8 -m pip install numpy==1.24.4
#    && python3.8 -m pip install numpy==1.19.4

# pip dependencies for OpenCV
RUN python3 -m pip install pybind11 Cython==0.29.36 setuptools==49.6.0 numpy==1.24.4

RUN cd build_opencv && ./build_opencv.sh

# Create package
RUN cd /tmp/build_opencv/opencv/build \
    && make package

# Post-installation cleanup

#RUN apt -y autoremove --purge \
#    git \
#    git-buildpackage \
#    debhelper \
#    autoconf \
#    gfortran \
#    libatlas-base-dev \
#    libavcodec-dev \
#    libavformat-dev \
#    libavresample-dev \
#    libcanberra-gtk3-module \
#    libdc1394-22-dev \
#    libeigen3-dev \
#    libglew-dev \
#    libgstreamer-plugins-base1.0-dev \
#    libgstreamer-plugins-good1.0-dev \
#    libgstreamer1.0-dev \
#    libgtk-3-dev \
#    libjpeg8-dev \
#    libjpeg-turbo8-dev \
#    liblapack-dev \
#    liblapacke-dev \
#    libopenblas-dev \
#    libpng-dev \
#    libpostproc-dev \
#    libswscale-dev \
#    libtbb-dev \
#    libtbb2 \
#    libtesseract-dev \
#    libtiff-dev \
#    libv4l-dev \
#    libxine2-dev \
#    libxvidcore-dev \
#    libx264-dev \
#    pkg-config \
#    qv4l2 \
#    v4l-utils \
#    zlib1g-dev \
#    yasm \
#    pkg-config \
#    libswscale-dev \
#    libtbb2 \
#    libtbb-dev \
#    libavformat-dev \
#    libpq-dev \
#    libxine2-dev \
#    libglew-dev \
#    libtiff5-dev \
#    zlib1g-dev \
#    libjpeg-dev \
#    libavcodec-dev \
#    libavformat-dev \
#    libavutil-dev \
#    libswscale-dev \
#    libeigen3-dev \
#    libtbb-dev \
#    libgtk2.0-dev \
#    pkg-config \
#    libeigen3-dev \
#    libncursesw5-dev

RUN apt clean && apt autoclean

#RUN rm -rf /tmp/build_opencv

# Copy the code and install the dependencies

COPY ComputerVision/ /code
WORKDIR /code
RUN python3 -m pip install -r requirements.txt

# Workaround for some sort of pytorch bug that was preventing the code from running
# As seen in https://stackoverflow.com/questions/72297590/attributeerror-upsample-object-has-no-attribute-recompute-scale-factor
RUN sed -i '157s/recompute_scale_factor=self.recompute_scale_factor//g' /usr/local/lib/python3.8/dist-packages/torch/nn/modules/upsampling.py

# COPY completo-5.bag /data/video.bag

COPY launch.sh /code
CMD ./launch.sh

RUN apt install opencv-python
```

```
docker run -it --rm --privileged --device=/dev/bus/usb --device /dev/video0 --device /dev/video1 --device /dev/video2 --device /dev/video3 --device /dev/video4 --device /dev/video5 as-computervision
```

Nothing changes.

MartyG-RealSense, 2024-08-05:
I am not familiar enough with Docker unfortunately to comment on the correctness of your Dockerfile, but you did say that the camera is working correctly with Docker on your Dell computer.

In your Dockerfile though I note that on the **RUN curl** line you are installing librealsense from packages using instructions from the **distribution_linux.md** installation instruction page.  Those instructions are for installing packages on PC though.  Jetson boards have different instructions for installing the -utils and -dev packages from an Ubuntu **keyserver**.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

FastCharge-ADAS, 2024-08-05:
Hi I'm TheBlind11 but writing with another account. I followed your advice and tried to run this script ``` ./scripts/patch-realsense-ubuntu-L4T.sh   ```, this is the result:

[image]

I'm afraid about our kernel and l4t version, what do you think?

FastCharge-ADAS, 2024-08-05:
I can't even try the docker image you provided in the repo: I was trying to build it and run it and it doesn't work, I don't really know what to do.. Could we have like a remote call in discord/teams/google meet in which we can discuss better about it? It's fundamental that we resolve the problem, we are going to a FSAE competition with our car and D455 is crucial for our autonomous system..

MartyG-RealSense, 2024-08-05:
If you mean the Librealsense Docker at the link below, it is for use with x86 architecture such as desktop / laptop PCs and not the Arm architecture that Jetson boards have, unfortunately.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/Docker/readme.md

Support is only available publicly through the RealSense GitHub forum sites such as this one.  If you require more advanced RealSense support then you can create a private support ticket on Intel's Zendesk support channel.

****************

Before we get to that stage though, there may be a workaround that can be tried because you have the D455 model.  Your earlier image of the Viewer suggested that you could receive depth but not RGB.

[image]

Instead of accessing the RGB stream, you could try using the **Infrared** stream instead and set its format to **BGR8** instead of the default Y8 infrared format.  This should provide an RGB image from the depth sensor instead of the RGB sensor.

In Python scripting, you can obtain this 'RGB from left infrared sensor' stream with the cfg configuration code below.

`cfg.enable_stream(rs.stream.infrared, 1280, 720, rs.format.bgr8, 30)`

tkitagawa1999, 2024-08-06:
Sorry for the late reply but by using the following environment I was able to get frames from Realsensen in a container using jetson.

Possible causes are
Difference in base image (environment with cuda is required)
Script execution with root privileges is required.

**Dockerfile**
```
ARG BASE_IMAGE
FROM ${BASE_IMAGE}

ARG PROJECT_NAME=depth_docker
ARG USER_NAME=docker
ARG GROUP_NAME=docker
ARG UID=1001
ARG GID=1001
ARG APPLICATION_DIRECTORY=/home/${USER_NAME}/${PROJECT_NAME}/

ENV LC_ALL="C.UTF-8"
ENV TZ=Asia/Tokyo
ENV HOME=/home/${USER_NAME}
ENV PYTHONPATH=:${APPLICATION_DIRECTORY}:${HOME}:${PYTHONPATH}
ENV PATH="${HOME}/.local/bin:$PATH"

RUN apt-get update \
    && DEBIAN_FRONTEND=noninteractive apt-get install --no-install-recommends -y \
    sudo \
    usbutils \
    v4l-utils \
    openssh-client \
    tree \
    tzdata \
    x11-apps \
    python3.10 \
    python3-pip \
    ffmpeg \
    git \
    make \
    && rm -rf /var/lib/apt/lists/* \
    && apt-get clean \
    && ln -sf /usr/bin/python3 /usr/bin/python \
    && ln -sf /usr/share/zoneinfo/Asia/Tokyo /etc/localtime

RUN groupadd -g ${GID} ${GROUP_NAME} \
    && useradd -ms /bin/bash --uid ${UID} -g ${GID} -G sudo ${USER_NAME} \
    && echo '%sudo ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers

USER ${USER_NAME}

COPY --chown=${UID}:${GID} ./ ${APPLICATION_DIRECTORY}
WORKDIR ${APPLICATION_DIRECTORY}

RUN sudo pip install pyrealsense2

CMD ["/bin/bash"]
```

**docker-compose.yaml**
```
version: '3'

services:
  depth_camera:
    build:
      context: .
      dockerfile: ./Dockerfile
      args:
        - BASE_IMAGE=nvidia/cuda:11.3.1-cudnn8-runtime-ubuntu20.04
        - PROJECT_NAME=depth_docker
    volumes:
      - ${HOME}/.Xauthority:/home/docker/.Xauthority:rw
    environment:
      - DISPLAY=${DISPLAY}
    tty: true
    stdin_open: true
    privileged: true
    network_mode: host
    runtime: nvidia
    profiles: []
```

@MartyG-RealSense 
Has proper error bundling been implemented for pyrealsense2 behavior when a GPU is not present? (Currently it seems like the error occurs when the GPU is not recognized)

Also, I get the following error if I don't have root privileges, how can I run it as a normal user?
**error**
```
NvRmMemInitNvmap failed with Permission denied
549: Memory Manager Not supported

****NvRmMemInit failed**** error type: 196626

*** NvRmMemInit failed NvRmMemConstructor

```

MartyG-RealSense, 2024-08-06:
Hi @tkitagawa1999  It's no trouble at all.  It's excellent to hear that you achieved a solution for your particular problem.

When installing librealsense on a Jetson from Debian packages, the special Jetson version of the packages (-utils and -dev) should be used, as they include CUDA support whilst the PC version of the packages do not.  

When building librealsense on a Jetson from source code with CMake, CUDA support is enabled by including the flag **-DBUILD_WITH_CUDA=ON** in the Cmake build instruction.

When CUDA support is enabled it will use the built-in Nvidia GPU on a Jetson board to automatically offload processing for pointclouds, depth-color alignment and YUY-to-RGB color conversion from the CPU onto the GPU.  CUDA only works with Nvidia GPUs like the ones integrated on Jetson boards.

Because the GPU is built into the Jetson board, there should not be a situation where a GPU is detected as absent.

If the flag **-DBUILD_GRAPHICAL_EXAMPLES=FALSE** is used then the SDK's graphics support will not be installed, which is helpful for 'headless' setups without a display screen where the inclusion of graphics support in the SDK can cause an error, or systems where only text-based programs will be used.

***********

My understanding is that if sudo admin permissions are not being used then permission is provided for normal users by having the librealsense **udev** device handling rules correctly installed.

tkitagawa1999, 2024-08-06:
@MartyG-RealSense 
If I install pyrealsense2 by pip instead of building librealsense, can I start it only with a container with cuda?
(Is the only way to pip install pyrealsense2 without cuda?)

I have resolved to root privileges, and I have confirmed that it works as a regular user by adding the following to the Dockerfile. Sorry for the trouble.

`RUN sudo usermod -aG video ${USER_NAME}`

MartyG-RealSense, 2024-08-06:
Regarding the question "can I start it only with a container with cuda", my Docker knowledge is insufficient to answer that query, unfortunately.

If you have a situation where Docker is not detecting the Nvidia GPU then the advice at the link below may be helpful.

https://stackoverflow.com/questions/73851446/gpu-not-available-in-docker

Whether librealsense's support for CUDA is enabled or not is determined by the method by which librealsense is installed.  It is not affected by how pyrealsense2 is installed, so using **pip install pyrealsense2** will not alter the CUDA support's on / off status.

I'm pleased to hear that you found a solution for your permissions issue.

tkitagawa1999, 2024-08-06:
If you install Realsense with pip, does it work regardless of whether or not the GPU is present or recognized on the device to which Realsense is connected? (Is it possible that it is a bug that it does not work when the base image is Ubuntu only, considering this case?)

tkitagawa1999, 2024-08-06:
I have a raspberrypi5 (ubuntu os) which is different from this case, but when I try to run the same method that worked this time in a container, I get the same error and it doesn't work well.

FastCharge-ADAS, 2024-08-06:
Thanks to @tkitagawa1999 I found the solution to my problem, I really appreciate it, thanks a lot.

This is working for me:
```
##############################################
# Dockerfile for depth-camera-hal microservice
##############################################

FROM nvcr.io/nvidia/l4t-base:35.4.1
ENV DEBIAN_FRONTEND=noninteractive

# Disable the non-working nvidia repo in the Docker image
RUN sed -i 's/^deb https:\/\/repo.download.nvidia.com\/jetson\/common r35.4.1 main/#&/' /etc/apt/sources.list

RUN apt-get update && apt-get install --no-install-recommends -y \
    libgl1-mesa-glx \
    libglib2.0-0 \ 
    libusb-1.0 \
    sudo \
    usbutils \
    v4l-utils \
    python3-pip
    
RUN apt -y autoremove --purge \
    python2 python2.7 python-is-python2
    
ARG PROJECT_NAME=depth-camera-hal
ARG USER_NAME=user
ARG GROUP_NAME=docker
ARG UID=1001
ARG GID=1001
ARG APPLICATION_DIRECTORY=/home/${USER_NAME}/${PROJECT_NAME}/
    
RUN groupadd -g ${GID} ${GROUP_NAME} \
    && useradd -ms /bin/bash --uid ${UID} -g ${GID} -G sudo ${USER_NAME} \
    && echo '%sudo ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers
    
USER ${USER_NAME}

WORKDIR ${APPLICATION_DIRECTORY}

RUN sudo usermod -aG video ${USER_NAME}

# Create the settings directory
RUN sudo mkdir -p ./settings

# Add all the local files in src, settings, and utils to the code directory
COPY --chown=${UID}:${GID} src ./src
COPY --chown=${UID}:${GID} settings ./settings
COPY --chown=${UID}:${GID} utils ./utils

COPY --chown=${UID}:${GID} settings/customSettings2.json ./settings/

# Install the requirements
COPY requirements.txt .

RUN sudo python3.8 -m pip install -r requirements.txt

# Run the Python script
CMD ["python3", "src/main.py", "-s", "settings/customSettings2.json"]
```

```
#!/bin/bash

bash build_image.sh

if [[ $(uname) == "Darwin" ]]; then
    docker run -it --rm --privileged --device=/dev \
    --runtime=nvidia \
    as-depth-camera-hal
else
    sudo docker run -it --rm --privileged --device=/dev \
    --runtime=nvidia \
    as-depth-camera-hal
fi
```

MartyG-RealSense, 2024-08-06:
Hi @FastCharge-ADAS  It's excellent news to hear that the comments of @tkitagawa1999 helped you to find a solution in time for your FSAE competition!

MartyG-RealSense, 2024-08-13:
/hi @tkitagawa1999  Do you require further assistance with this case, please?  Thanks!

closed 2024-08-19 by MartyG-RealSense