realsenseai/librealsense #13206: Dealing with Noise and Edge Artifacts in Point Clouds from Intel RealSense D455 Cameras

KonradBk, 2024-07-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455f | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |  RealSense Viewer 2.55.1 }                          | 
| Language                            |  python 3.11 | 
| Segment			|  Object scanning   | 

I'm currently working on a project using the multicamera box dimensioning application to capture point clouds that accurately represent scanned objects, particularly focusing on capturing the top and side walls of packages. My goal is to obtain point clouds that are as detailed and accurate as possible, highlighting the shape and features of the scanned objects.

I've reached a point where I can extract a point cloud from the application that clearly shows the side and top walls of the package. However, I'm encountering a significant issue: the point cloud contains a lot of noise, including unwanted and non-existent points that do not represent any real-world features.

[image]

To address this, I first tried tweaking the camera settings using the Intel RealSense Viewer, hoping to find a configuration that would accurately capture the object while minimizing the noise. Unfortunately, I couldn't achieve the desired clarity and reduction in noise.

Next, I attempted to clean up the point cloud in post-processing. I used Open3D's outlier removal functions (https://www.open3d.org/docs/0.9.0/tutorial/Advanced/pointcloud_outlier_removal.html), which did help to some extent, but I'm still not entirely satisfied with the results. I've reached a point where the outlier removal process, when made more stringent, begins to remove the side walls of the object instead of the noise present in the point cloud.

[image]

Currently, the biggest issue I'm facing is the strange extensions of the side edges at the base of the object (highlighted in the image below). I haven't been able to adjust the camera settings in a way that eliminates these artifacts without significantly compromising the quality of the object's side walls.

[image]

My question is: Are there specific camera settings that can help reduce this noise while preserving accurate details of the object's side walls? Is my approach to solving this issue appropriate, or might there be a method I'm overlooking? Any guidance or suggestions on how to improve the quality of the point clouds or different strategies to consider would be greatly appreciated. Thank you!

Camera configuration: [TestConfiguration.json](https://github.com/user-attachments/files/16409730/TestConfiguration.json)

MartyG-RealSense, 2024-07-29:
Hi @KonradBk  The **DefaultPreset_D455** json file provides clean edges and reduced pointloud 'spraying'.

https://raw.githubusercontent.com/wiki/IntelRealSense/librealsense/d400_presets/DefaultPreset_D455.json

KonradBk, 2024-07-29:
Hi @MartyG-RealSense. Unfortunately, the camera settings you suggested do not resolve the issue. The entire ground surface appears much more wavy, the side wall is full of holes, and the noise along the side edges is still visible. Additionally, the auto-exposure option should be disabled in my case because I'm working with moving objects, and at higher speeds (e.g., 1m/s), the depth image becomes stretched and misaligned with its RGB counterpart.

[image]

In the point cloud obtained after outlier removal, the same problem persists: the side and top walls begin to be removed, while the noise behind the edges remains clearly visible. 

[image]

Is there another way to address this issue? Any further advice would be greatly appreciated.

MartyG-RealSense, 2024-07-29:
What quality of results do you get if you try the Python / Open3D pointcloud script at https://github.com/IntelRealSense/librealsense/issues/11912#issuecomment-1683811415

closed 2024-08-17 by MartyG-RealSense