realsenseai/librealsense #13207: Query on connecting multiple D415

girishsmurthyIntel, 2024-07-29:
Dear Team,
         I have a query where I need to connect multiple D415 cameras to an intel NUC. How could I best connect these such that I am able to stream 720p 30fps RGB only data?

Rgds,
Girish

MartyG-RealSense, 2024-07-29:
Hi @girishsmurthyIntel  You could connect multiple D415 to a NUC if you attach the cameras to a USB hub device, and in turn connect the hub to one of the built-in USB ports on the NUC.  The USB standard permits 5 hubs to be linked together in a 'daisy chain' arrangement.  

However, each active camera will consume computer resources.  So there is a limit on how many **simultaneously** active cameras can be on the same computer.  

For up to three cameras a computer with a processor equivalent to an Intel Core i5 should be sufficient, whilst for four or more cameras a Core i7 or i9 is recommended.

girishsmurthyIntel, 2024-07-30:
@MartyG-RealSense Thank you for the kind reply. Should we also make sure that the USB should be 3.0 and both the cables (camera to hub and hub to NUC) are of USB 3.0?
Currently I am using the  Intel(R) Core(TM) i7-8559U CPU @ 2.70GHz
So >=4 cameras, could be supported?

MartyG-RealSense, 2024-07-30:
That processor should be able to support 4 cameras, yes.  The 8 in 8559U indicates that it is an 8th generation Intel Core chip, a technology which is around 7 years old at the time of writing this.  So 4 simultaneously active cameras may be the most that it can handle without a significant performance drop if further cameras are added.

Yes, the hub and cables should both be USB 3 (preferably USB 3.2).  Yes, you can buy a 4-port hub.  Amazon's own-brand **AmazonBasics** hub works well with RealSense cameras and is very reliable.

5Gbps should be sufficient to handle the data transfer of four cameras.  In 2018 Intel performed bandwidth usage tests with four cameras in a range of streaming configurations using an AmazonBasics hub.

[image]

girishsmurthyIntel, 2024-08-01:
@MartyG-RealSense Thank you for the kind reply. Using a USB3.0 port, I was able to connect to three D415 cameras streaming at 720p@30fps. The question I have further is about how I could launch (Say for each camera, ros2 launch realsense2_camera rs_launch.py camera_name:=camera_top1 serial_no:="'xxxxxx'" rgb_camera.profile:=1280x720x30 enable_depth:=false)  all three cameras in a single script?

MartyG-RealSense, 2024-08-01:
I cannot see any problems in the multiple camera launch instruction in the comment above.  It should work.  It is recommended though that you open a separate terminal for each camera and input the launch instruction for a particular camera rather than launch all cameras in the same terminal.

It is possible to launch multiple cameras from the same launch file.  The ROS wrapper provides such a launch file, **rs_multi_camera_launch.py**.  It is only configured to handle two cameras by default rather than three though.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/realsense2_camera/launch/rs_multi_camera_launch.py

girishsmurthyIntel, 2024-08-01:
Thank you @MartyG-RealSense for the kind references and support.

closed 2024-08-01 by girishsmurthyIntel

MartyG-RealSense, 2024-08-01:
You are very welcome.  I'm pleased that I could help  :)