realsenseai/librealsense #13209: Recording using D405 without RealSense Viewer

kmkomori, 2024-07-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  N/A                                     | 
| Platform                                 | PC |
| SDK Version                            |  2.55.1                         | 
| Language                            | MATLAB                          | 
| Segment			|  not sure                          | 

### Issue Description

Hello! I'm working with the D405 to record depth and color data, and was looking for a way to record without using the Intel RealSense Viewer. Specifically, I'm looking for a way to manage (as in start/stop/etc.) the recording through MATLAB scripts or the terminal window. Looking through the Github page I was hoping to use the '[rs-record](https://github.com/IntelRealSense/librealsense/tree/master/tools/recorder)' function, but it doesn't seem to be included in the SDK (or I am just not sure how to access and use it).

I want to do this so I can automate the recording process (i.e. run a script and have it automatically start and save the recording). Some things I am looking for functionality wise is that it is easy to run (similar to how you can just run the 'rs-convert' function in the terminal window with the file path and conversion settings), and if possible choose the amount of time the recording goes for (which is why I liked the 'rs-record' function). If any of this is not possible, alternatives are also greatly appreciated. Thank you for your time!

MartyG-RealSense, 2024-07-30:
Hi @kmkomori  There are few references available regarding recording to file with the MATLAB wrapper, unfortunately.  Below is a MATLAB recording script that my research found.

```
pipe = realsense.pipeline();
colorizer = realsense.colorizer();
pipe.start();
fs = pipe.wait_for_frames();
depth = fs.get_depth_frame();
depth_color = colorizer.colorize(depth);
rgb = fs.get_color_frame();
pipe.stop();
% create a new pipeline
cfg = realsense.config();
pipe = realsense.pipeline();
cfg.enable_record_to_file('FILEPATH');
pipe.start(cfg);
device = pipe.get_active_profile().get_device();
pipe.stop();
```

kmkomori, 2024-07-30:
Are there other methods of recording without using the RealSense Viewer, such as another tool or in a different programming language? If available as an .exe file, I could potentially run it within my code.

MartyG-RealSense, 2024-07-31:
The rs-record link is working correctly for me.

The rs-record tool is in the **librealsense > tools > recorder** folder.

[image]

If you have installed the full RealSense SDK on Windows using the **Intel.RealSense.SDK-WIN10** installer file then there is a pre-built .exe of a record and playback example program called 'rs-record-playback' at the location below:

**C:\Program Files (x86)\Intel RealSense SDK 2.0\tools**

As well as the C++ rs-record program, there is an example of a Python bag recording script at https://github.com/IntelRealSense/librealsense/issues/3671#issuecomment-482057297

At https://github.com/IntelRealSense/librealsense/issues/4801 the pausing and unpausing of recording in Python is discussed.

You can also save a single frame to a bag with the **save_single_frameset** instruction.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/single_frameset.py

closed 2024-08-17 by MartyG-RealSense