realsenseai/librealsense #13210: D435 depth oscillation

Nimaro76, 2024-07-29:
Hello, 
I hope you are well. I have a question regarding D435if depth oscillation. I have observed that even in realsense-viewer the depth of 3D points oscillates. I have activated several of the filters but the oscillation is still there. are there any other approaches that can lead to a steady pointcloud ? 

kind regards

MartyG-RealSense, 2024-07-30:
Hi @Nimaro76  If depth values are oscillating then setting the **Filter Smooth Alpha** value of the Temporal post-processing filter to '0.1' instead of its default of '0.4' can stabilize fluctuations.

Nimaro76, 2024-07-30:
thank you , and in the code, can I do that as well in c++ ?

MartyG-RealSense, 2024-07-30:
There is not a specific example for setting the Temporal filter's alpha with C++ but the link below has a C++ script that suggests how to do it.

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#using-filters-in-application-code

The code would likely look something like this:

// Streaming initialization
rs2::pipeline pipe;
 ...
// Declare filters

rs2::temporal_filter temp_filter

// Configure filter parameters
temp_filter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.1f);

// Main Loop
while (true) {
rs2::frameset data = pipe.wait_for_frames();
rs2::frame depth_frame = data.get_depth_frame();
...
rs2::frame filtered = depth_frame;

filtered = temp_filter.process(filtered);

closed 2024-08-17 by MartyG-RealSense