realsenseai/librealsense #13213: Couldn't find an option to flip image on RS viewer

omer-re, 2024-07-30:
I have mounted my RS cam on my car and therefore it was captured upside-down.
Is there a way to use the viewer so it'll flip it?
couldn't find it on the menus or looking it online.

I believe it can be recreated flipped using python, but I am worried about data lost.

MartyG-RealSense, 2024-07-31:
Hi @omer-re  The RealSense Viewer tool does not have a flip function.  

Images can be flipped using program code.  However, all calibration data and other parameters would need to be updated, as advised at https://github.com/IntelRealSense/librealsense/issues/6023

MartyG-RealSense, 2024-08-09:
Hi @omer-re Do you require further assistance with this case, please?  Thanks!

omer-re, 2024-08-11:
I'd be happy if you can push it to the roadmap as a feature request 🤷‍♂️
@MartyG-RealSense

MartyG-RealSense, 2024-08-11:
There is already an existing feature request for rotating the depth frame.  The plan is to do so using a post-processing filter.  It is being considered for implementation into an SDK release next year (2025).

MartyG-RealSense, 2024-08-17:
Hi @omer-re  Bearing in mind the information in the comment above, do you require further assistance with this case please?  Thanks!