realsenseai/librealsense #13214: error

Chenhaining8, 2024-07-31:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_create_context_ex(api_version:25501, json_settings:nullptr):
    failed to convert special folder: errno=42

```

MartyG-RealSense, 2024-07-31:
Hi @Chenhaining8  Advice about the **errno=42** error can be found at https://github.com/IntelRealSense/librealsense/issues/13037

Chenhaining8, 2024-07-31:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在上学，无法亲自回复您的邮件。我将在上学结束后，尽快给您回复。

Chenhaining8, 2024-07-31:
Thank you,I have solved this problem.
 

永恒之命运之刻
***@***.***

MartyG-RealSense, 2024-07-31:
That's excellent news.  Thanks very much for the update!

closed 2024-08-09 by MartyG-RealSense