realsenseai/librealsense #13222: install error

anthony930713, 2024-08-01:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_create_context_ex(api_version:25501, json_settings:{):
    failed to convert special folder: errno=42

```

MartyG-RealSense, 2024-08-01:
Hi @anthony930713  Advice about the **errno=42** error can be found at https://github.com/IntelRealSense/librealsense/issues/13037

closed 2024-08-19 by MartyG-RealSense

MartyG-RealSense, 2024-12-04:
My Intel RealSense colleagues have provided some further advice about the errno=42 error.

Given that the problem seems to be related to certain charcters being used in a Windows user account name, I asked whether creating a secondary Windows user account with a name that has no special characters might help.  This was their response. 

**********

You can try to run it as Administrator instead of switching the account.  On Windows 11, it also has a feature to run the app as a different user.  That option can be enabled with the instructions at the link below:
 
https://winaero.com/how-to-run-apps-as-different-user-in-windows-11/