realsenseai/librealsense #13225: What is this

yumjunstar, 2024-08-01:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2024-08-01:
Hi @yumjunstar  This error sometimes occurs when trying to access the RGB camera from the **Add Source** option at the top of the RealSense Viewer's options side-panel, like in https://github.com/IntelRealSense/librealsense/issues/11338

Are you using the Add Source button, please?

yumjunstar, 2024-08-01:
Oh how did you know that? yeah you're right. I did

MartyG-RealSense, 2024-08-01:
This error almost always happens when using Add Source.  This option is designed for accessing non-RealSense cameras such as USB RGB webcams and built-in RGB cameras in laptops.  It does not work with all webcams though and there is no fix if your USB webcam or built-in webcam cannot be opened with it, unfortunately.

If you have a RealSense camera that can be detected then it should be automatically listed in the side-panel and it is not necessary to use Add Source to access it.  If you do have a RealSense camera, which model is it please?

closed 2024-08-19 by MartyG-RealSense