realsenseai/librealsense #13226: imu data stream is invalid

nit-xhg, 2024-08-01:
The device used is L515 and the OS is ubuntu20.04. When using the imu data of the sensor, the data in the imu data stream is invalid. The frequency for accel is 100 and the frequency for gyro is 200. Here is the data stream for rostopic echo /camera/imu. The question is why, does imu need to be calibrated? Thank you very much for answering my confusion!

orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -1.1669913530349731
  y: -9.767423629760742
  z: -0.3628460466861725
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]

MartyG-RealSense, 2024-08-01:
Hi @nit-xhg  Looking at the Z-acceleration value gives a good indication of how well a camera is calibrated.  The ideal Z-accel value is -9.80 and yours is -9.76, which indicates that your current IMU calibration is very good and a re-calibration is likely not necessary.  

If you see a message in the ROS launch log suggesting that the IMU is not calibrated and to perform a calibration then you can ignore it.

If you are concerned about your IMU values then you could try adding **enable_motion_correction:=true** to your ROS launch instruction to ensure that the raw IMU data has corrections performed on it to increase its accuracy, resulting in 'fixed' IMU values.

nit-xhg, 2024-08-02:
The IMU drive or sensor itself does not calculate or output orientation information properly.
orientation:
x: 0.0
y: 0.0
z: 0.0
w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

nit-xhg, 2024-08-02:
> Hi @nit-xhg Looking at the Z-acceleration value gives a good indication of how well a camera is calibrated. The ideal Z-accel value is -9.80 and yours is -9.76, which indicates that your current IMU calibration is very good and that performing a new calibration is likely not necessary.
> 
> If you see a message in the ROS launch log suggesting that the IMU is not calibrated and to perform a calibration then you can ignore it.
> 
> If you are concerned about your IMU values then you could try adding **enable_motion_correction:=true** to your ROS launch instruction to ensure that the raw IMU data has corrections performed on it to increase its accuracy, resulting in 'fixed' IMU values.

The IMU drive or sensor itself does not calculate or output orientation information properly.
orientation:
x: 0.0
y: 0.0
z: 0.0
w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

MartyG-RealSense, 2024-08-02:
Can you confirm that you have enabled the publishing of the gyro and accel topics in your roslaunch instruction by adding the parameters below to it, please?

**ROS1**
`enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation`

**ROS2**
`enable_gyro:=true enable_accel:=true unite_imu_method:=2`

nit-xhg, 2024-08-02:
> Can you confirm that you have enabled the publishing of the gyro and accel topics in your roslaunch instruction by adding the parameters below to it, please?
>
> **ROS1** `enable_gyro:=true [...]

I set it using rs_camera.launch, acceleration and angular velocity are normal, but orientation and orientation_covariance are all 0, no matter how I move. Here are my setup parameters
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="100"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>
   <arg name="unite_imu_method"          default="linear_interpolation"/>

nit-xhg, 2024-08-02:
> I set it using rs_camera.launch, acceleration and angular velocity are normal, but orientation and orientation_covariance are all 0, no matter how I move. Here are my setup parameters

  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="100"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>
   <arg name="unite_imu_method"          default="linear_interpolation"/>

nit-xhg, 2024-08-02:
> I set it using rs_camera.launch, acceleration and angular velocity are normal, but orientation and orientation_covariance are all 0, no matter how I move. Here are my setup parameters

[image]

MartyG-RealSense, 2024-08-02:
That is unusual, you should be getting gyro orientation readings.  Does it change if you set unite_imu_method to **copy** instead of linear_interpolation

nit-xhg, 2024-08-02:
> That is unusual, you should be getting gyro orientation readings. Does it change if you set unite_imu_method to **copy** instead of linear_interpolation

It starts as a parameter to copy, but the result is the same

MartyG-RealSense, 2024-08-02:
What happens if you increase the Accel and Gyro FPS to their supported maximums on the L515 camera (400 Accel and 400 Gyro).

[image]

[image]

nit-xhg, 2024-08-02:
> What happens if you increase the Accel and Gyro FPS to their supported maximums on the L515 camera (400 Accel and 400 Gyro). [...]

It's no different, several different release frequencies have tried it.

MartyG-RealSense, 2024-08-02:
Could you try unplugging the micro-sized end of the USB cable from the base of the L515, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

Does the gyro topic have values if you leave off the **unite_imu_method** parameter from the launch instruction and echo the topic **/camera/gyro/sample**

Are you able to access the **realsense-viewer** tool and test whether the gyro works correctly in that when the Motion Module is enabled?

nit-xhg, 2024-08-02:
This is the data flow of the /camera/imu and /camera/gyro/sample topics, both of which are invalid, and it is also done as you said.

header:
seq: 10985
stamp:
secs: 1722602160
nsecs: 783682823
frame_id: "camera_imu_optical_frame"
orientation:
x: 0.0
y: 0.0
z: 0.0
w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
x: 0.0
y: 0.0
z: -0.001745329238474369
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration:
x: -1.2402251645438847
y: -9.75761604309082
z: 0.10787314921617508
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
header:
seq: 23693
stamp:
secs: 1722603321
nsecs: 742941618
frame_id: "camera_gyro_optical_frame"
orientation:
x: 0.0
y: 0.0
z: 0.0
w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
x: 0.001745329238474369
y: 0.001745329238474369
z: -0.001745329238474369
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration:
x: 0.0
y: 0.0
z: 0.0
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]

> Could you try unplugging the micro-sized end of the USB cable from the base of the L515, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way [...]

This is the data flow of the /camera/imu and /camera/gyro/sample topics, both of which are invalid, and it is also done as you said.

header:
seq: 10985
stamp:
secs: 1722602160
nsecs: 783682823
frame_id: "camera_imu_optical_frame"
orientation:
x: 0.0
y: 0.0
z: 0.0
w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
x: 0.0
y: 0.0
z: -0.001745329238474369
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration:
x: -1.2402251645438847
y: -9.75761604309082
z: 0.10787314921617508
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
header:
seq: 23693
stamp:
secs: 1722603321
nsecs: 742941618
frame_id: "camera_gyro_optical_frame"
orientation:
x: 0.0
y: 0.0
z: 0.0
w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
x: 0.001745329238474369
y: 0.001745329238474369
z: -0.001745329238474369
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration:
x: 0.0
y: 0.0
z: 0.0
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]

MartyG-RealSense, 2024-08-02:
If you are able to use Python on your computer then you could attempt to use the **rs-imu-calibration.py** IMU calibration script.  Whilst this tool no longer supports L515 in the current versions of the RealSense SDK, I have retrieved an older version of the script that still has L515 support and attached the script file to this comment below.

[rs-imu-calibration.py.txt](https://github.com/user-attachments/files/16470965/rs-imu-calibration.py.txt)

If you are able to run this calibration tool and the gyro values are still not responding then it is unfortunately probable that there is a hardware fault in your L515's IMU component.  If you purchased the camera less than a year ago then you can try contacting the retailer that you purchased it from to enquire about an **RMA return** of the L515.

nit-xhg, 2024-08-02:
> If you are able to use Python on your computer then you could attempt to use the **rs-imu-calibration.py** IMU calibration script. Whilst this tool no longer supports L515 in the current versions of [...]

Can you provide a calibration tool instructions? Under normal circumstances, imu does not need to be calibrated. In other words, we can confirm that if there is no value of orientation and orientation_covariance, there may be problems with the imu hardware. But I wonder whether L515 can directly provide orientation and orientation_covariance, or whether I need to calculate it by the values of acceleration and angular velocity.

MartyG-RealSense, 2024-08-02:
There is a PDF guide to the calibration tool at the link below.

https://www.intelrealsense.com/wp-content/uploads/2020/07/IMU_Calibration_Tool_for_Intel_RealSense-Depth_Cameras_Whitepaper.pdf

The RealSense SDK / Viewer can directly provide a gyro stream that contains values reflecting the tilt and rotation angles of the camera.  There is an example of such a script for Python at https://github.com/IntelRealSense/librealsense/issues/6896

**orientation_covariance** is accessible in ROS, as you showed at the beginning of this discussion, but is not usually used in librealsense SDK program scripts.

MartyG-RealSense, 2024-08-09:
Hi @nit-xhg  Do you require further assistance with this case, please?  Thanks!

nit-xhg, 2024-08-09:
> Hi @nit-xhg Do you require further assistance with this case, please? Thanks!
No,thanks

closed 2024-08-09 by nit-xhg