realsenseai/librealsense #13227: Unhandled exception escaping from a worker thread!

wenchenwan, 2024-08-01:
| | |
|---|---|
|**librealsense**|2.54.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2024-08-01:
Hi @wenchenwan  This error sometimes occurs when trying to access the RGB camera from the **Add Source** option at the top of the RealSense Viewer's options side-panel, like in https://github.com/IntelRealSense/librealsense/issues/11338

wenchenwan, 2024-08-02:
Yes, I chose to use the official cable, and my device is t265，But I can't connect
[image]

MartyG-RealSense, 2024-08-02:
Support for T265 was removed in version 2.54.1 as it is a retired camera model.  

I recommend installing the **2.51.1** version of the SDK (Intel.RealSense.Sdk-WIN10) or the RealSense Viewer (Intel.RealSense.Viewer.exe)

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

[image]

closed 2024-08-19 by MartyG-RealSense