realsenseai/librealsense #13228: Error: Failed to open usbinterface and failed to set power state

JaCoHSU, 2024-08-02:
I am using the Realsense D415 on the Jetson Xavier NX with ROS in Ubuntu 18.04.  The ROS Realsense version is 2.3.2 running with LibRealSense v.2.50.0.

I am facing the strange behavior that every 5-10 boots or so when I start "roslaunch realsense2_camera rs_camera.launch" the script does not start. It stops with the error:
- failed to open usbinterface 0, error: RS2_USB_STATUS_ID
- aquire_power failed: failed to set power state

Like I said it only happens like every 5-10 boots. I have tested it with different Realsense firmware versions, different cables (also with the original cable) and also different Realsenses. 

Does anyone can help with this problem?

MartyG-RealSense, 2024-08-02:
Hi @JaCoHSU  Does the frequency of the problem reduce if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch, please?

JaCoHSU, 2024-08-02:
Hey @MartyG-RealSense,
thank you for the fast response.

Actually I tried to add initial_reset:=true to the launch script. I did this because **sometimes** the Realsense got detected as an USB2 devices and I thought this might be the problem. After adding initial_reset:=true to the script the realsense got detected as an USB3 device after the reset. 
However, this does not solve my problem since it looks like that the error occurs **before** the initial reset is triggered.

I added a screenshot of the console when the error occurs. (sorry for the bad quaility)
[image]

MartyG-RealSense, 2024-08-02:
A **Failed to set power state** error suggests that the camera cannot be accessed, perhaps because it is in a 'busy' state.  This might occur because it was not properly closed down from a previous session of use.  If the camera could not be detected at all then the error would likely be 'No RealSense camera detected'.

Are all of your cameras using the **5.13.0.50** camera firmware driver that is recommended for use with SDK 2.50.0 please?

JaCoHSU, 2024-08-02:
Okay, but after a new boot the camera should not be in busy state, right?

Also, i tried plugin out and plugin in the camera but I guess that the launch script does not detect that since the script does not continue. It stays at the error message.
However, I noticed that when I get the error, close the terminal and start a new terminal with the roslaunch realsense script the realsense gets detected and works as expected.

I tried the newest firmware that I can upload with the realsense viewer and an older version (5.12.7)

JaCoHSU, 2024-08-02:
Yes, that's what I read, therefore if used an older version of the firmware.

I tried both, soft reboot and hard boot. The problem occurs in both situations. 

Do you have any idea what I can do to solve or at least to debug the problem even more?

MartyG-RealSense, 2024-08-02:
You could eliminate that possibility that the rs_camera.launch file is responsible by performing an **RGBD** launch instead.

1.  Install support for an RGBD launch on Melodic with the command below.

`sudo apt-get install ros-melodic-rgbd-launch`

2.  Run the rs_rgbd.launch file.

`roslaunch realsense2_camera rs_rgbd.launch`

JaCoHSU, 2024-08-07:
Hey @MartyG-RealSense,

I did update the firmware of the Realsense to 5.13.0.50 and did plenty of hard reboots. In the beginning, I thought that the problem was gone. However, yesterday it occured again 2-3 times.
I did not try to use the rs_rgbd script yet because I need the color and the depth image.

Do you think that the problem is a Realsense problem or might it come from the Jetson Xavier NX as well?
Can you explain the error message to me? What does failed to open usb interface and failed to set power state mean? Maybe that might help me to debug the problem further or to find a workaround. I found out that when I restart the launch script that the problem does not occur. Maybe it would be a workaround to detect the error and restart the script automatically?!

MartyG-RealSense, 2024-08-07:
rs_rgbd does publish both depth and color like the rs_camera launch does, as depth and color are enabled by default in rs_rgbd.launch.

My understanding is that the **RS2_USB_STATUS_IO** error (not _ID) is related to a failure to get data from the USB port (IO stands for Input-Output - basically, the communications between the camera and the USB port of the computer).

As this error occurs first in the log, the subsequent 'failed to open usb interface' and 'failed to set power state' errors could be what I would call 'cascade errors' - errors that are triggered by the preceding error and would not have happened if that previous error had not first occurred.  I would therefore recommend focusing on the RS2_USB_STATUS_IO error.

The 'failed to open usb interface' and 'failed to set power state' errors typically occur when the camera is unable to be accessed.  The "usb interface" error is linked to **RS2_USB_STATUS_ACCESS**, advice about which can be found at https://github.com/IntelRealSense/librealsense/issues/7521#issuecomment-705056917

MartyG-RealSense, 2024-08-13:
Hi @JaCoHSU  Do you require further assistance with this case, please?  Thanks!

JaCoHSU, 2024-08-13:
Hey @MartyG-RealSense,

I am still trying to debug the issue, which is hard because it only occurs after about every 10th-20th boot. 
I have shortly tried it with the rs_rgbd.launch, but I got a warning: "uvc streamer watchdog triggered on endpoint" and the fps seemed to be very bad.
However, currently I am using a different Jetson to rule out that the Jetson is the problem.

I will come back to you as soon as I get new results or have more questions!

MartyG-RealSense, 2024-08-13:
Thanks very much for the update.  I look forward to your next report.  Good luck!

JaCoHSU, 2024-08-23:
Hey @MartyG-RealSense,

good news are that I have not had the failed to set power state error since I changed the jetson! I will continue testing and keep you updated.

However, while testing yesterday I had the problem, that I had a rgb image, but sometimes I did not receive any depth frame at all after starting the rs_camera script.  Do you have an idea why this might be?

Furthermore, I observed that sometimes I do not get any images (rgb and depth) for a few seconds and when the images are coming back they start with an auto-exposure. Do you have any idea if the auto-exposure is the reason for the stop or if a reconnect of the realsense is the reason for the auto-exposure?

MartyG-RealSense, 2024-08-23:
If there is a situation where the RealSense camera is facing directly towards strong light (such as the sun) then the camera sensors can become saturated with light.  If auto-exposure is enabled then the image can automatically recover if the strong light is no longer directly facing the camera, or the camera has moved and is no longer directly facing the light.

If it is not likely that your camera is being exposed to a strong light such as the sun then you could try setting initial_reset to true again now that your **Failed to set power state** issue is resolved to see whether it makes a difference to the problem.

If the camera is being used in a location with consistent lighting levels such as an office then you could also try disabling auto-exposure and use a fixed manual exposure value in order to confirm whether auto-exposure is causing your image problems.

JaCoHSU, 2024-08-30:
Hey @MartyG-RealSense,

last week I have not had any problems. However, the error occurred again a few times this week.
I still do not know why. 
The Jetson does start the rs_camera.launch script automatically 30 seconds after booting. When I get the "failed to open usbinterface....failed to set power state error", I need to restart the Jetson. I do this by plugin out and in the battery. Most of the times the error does occur several times in a row. So I need to restart the Jetson 2-3 times until the script does start correctly. Thats why I do not think that it is a hardware problem.
The only difference to last week where I did not had the errors is that the temperature was not as high as this week. Might a high temperature can have anything to do with this error?

MartyG-RealSense, 2024-08-30:
There have been a small number of past cases where a camera needed to be restarted several times in order for it to be accessed correctly.  https://github.com/IntelRealSense/librealsense/issues/8694 is an example of this and is a case where rs_camera.launch was also being used.

It is unlikely that temperature is a factor in this problem.  Typically with these rare 'boot several times' issues, the reason for it occurring is not found.

JaCoHSU, 2024-08-30:
@MartyG-RealSense 

Hmm... If I do not find the reason for the error I might need to find a workaround. 
Would it be possible to detect the error and restart the rs_camera.launch script?

MartyG-RealSense, 2024-08-31:
One option might be to modify the **base_realsense.node.cpp** source code of the ROS1 wrapper to insert a hardware reset command in the error "catch" section of the wrapper where it registers when an error has occurred.  You would then build the customized wrapper from source code.

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/src/base_realsense_node.cpp#L166-L170

Below is an example of C++ hardware reset code.

```
rs2::context ctx;
rs2::device dev = ctx.query_devices().front(); // Reset the first device
dev.hardware_reset();
rs2::device_hub hub(ctx);
dev = hub.wait_for_device();
```

JaCoHSU, 2024-09-02:
@MartyG-RealSense 

thank you! I will try that out. I built the realsense-ros by source and adapted the base_realsense_node.cpp as you suggested. Since Librealsense version is v2.55.1 now I also updated the realsense to FW version 5.16.0.1

I will come back to you as soon as I get some test results.

MartyG-RealSense, 2024-09-02:
You are very welcome, @JaCoHSU - I look forward to your next report.  Good luck!

JaCoHSU, 2024-09-23:
Hey @MartyG-RealSense,

I just wanted to come back to you and tell you that I have not had the error again since I updated the librealsense and the FW version of the realsense. 

Thank you very much for your help!!!

closed 2024-10-03 by MartyG-RealSense